sinara-hw/sinara #2: dsp/gateware features

jordens, 2016-09-26:
- [ ] Absolute and relative phase modes for all frequency and phase interpolators independently configurable through RTIO
- [ ] f,p,a,u modulation by RTIO (eDRTIO) for all interpolators
- [ ]  DMA m-labs/artiq#553
- [ ] Parallelizing CIC interpolator/AA filter before parallelized DDS core
- [ ] Parallelized FIR filter after CIC, research and implement what’s possible given DSP resources
- [ ] Coherent phase mode (implemented in kernel API and runtime) for frequency/phase updates of the three eDRTIO channels

dhslichter, 2016-09-27:
For confirmation: will DMA as listed above include the streaming of precompiled DAC samples from memory onto the DACs vias JESD204B?  This would allow for sequences that may not be able to be created efficiently by the gateware CORDIC+interpolators (e.g. 20 ns duration Gaussian pulse at 1 GSPS).  Will DMA be able to seamlessly take over streaming of DAC samples from CORDIC+interpolators, and then seamlessly return it?  Use case would be to output a fixed dc level ("off" to a mixer IF port), then briefly stream in a complex/high-speed pulse such as above, then return to fixed dc level (again, "off" to a mixer IF port).

jordens, 2016-09-27:
Not currently. And DMA is functionally orthogonal to sample-based operation.
If somebody is interested in this (or other additional features), please open new issues.

jbqubit, 2017-02-01:
@jordens please remove for-contract label since this is funded by ARL

jordens, 2017-02-01:
@jbqubit not all of it.

jbqubit, 2017-02-01:
@jordens Please annotate which parts are not funded.

jordens, 2017-02-01:
@jbqubit that's not wise and would be redundant. the individual tracking issues should also track their funding status.

closed 2017-08-10 by hartytp

jbqubit, 2017-08-11:
Closed since  gateware Issues migrated to ARTIQ #236