sinara-hw/sinara #4: clocking

jordens, 2016-09-26:
* `ref_clk` distribution
  * rf backplane
  * amc backplane
* `dac_clk` generation
* `sysref` generation
  * by FPGA and retimed by HMC7043/7044 to `ref_clk` and `dac_clk`
  * by clock-gen and mesured w.r.t. `drtio_clk` by FPGA

jordens, 2016-10-07:
Mailing list thread with the current proposal:
https://ssl.serverraum.org/lists-archive/artiq/2016-September/001007.html

@jbqubit , Tom Harty, could you sign off on this?

sbourdeauducq, 2016-10-07:
Can someone double-check that 8 KU transceivers can be clocked without restrictions (other than pin placement) by a single reference clock pin?

sbourdeauducq, 2016-10-19:
The transceiver XO/synthesizers may not be necessary as the Si5324 has a freerun mode with hitless switching. See https://forums.xilinx.com/t5/Xilinx-Boards-and-Kits/VC707-stm1-clk-via-Si570-through-Si5324-PPM-spec/td-p/647852:
>  You only need to use the Si5324 in free run mode with the 114.285 MHz crystal to generate the initial 155.52 MHz for the GTHs to operate, lock and generate a recovered clock from the received data.  See page 73, section 6.5 of the Si53xxReferenceManual for more information on the free run mode. 

> Once you have the GTH locked to the incoming data stream, the RXRECCLK can then be driven from the FPGA to the Si5324 using the REC_CLOCK_C_P|N pins and it will switch over and locked to the clock providing a 0 PPM offset.
 
> The Si5324 data sheet has the necessary register information to enable the FREE_RUN mode and to set the priority for CKIN1 (recovered clock) vs CKIN2 (crystal).  Note:CKIN1 should have priority over CKIN2, so that the Si5324 will automatically switch from the crystal to the receovered clock.

sbourdeauducq, 2016-10-19:
> Can someone double-check that 16 KU transceivers can be clocked without restrictions (other than pin placement) by a single reference clock pin?

And this is fine, those 16 DAC transceivers should just be placed in adjacent quads and the clock pin's quad preferably should be centered.

jbqubit, 2016-10-24:
#27 impacts this.

closed 2016-11-14 by jordens