sinara-hw/sinara #5: license

jordens, 2016-09-26:
* TAPR OHL
* CERN OHL
* GPLv3+

jordens, 2016-09-26:
I don't like the notification requirement in TAPR OHL and share the concerns about the applicability of copyright/left to hardware designs. I'd go for CERN OHL.

sbourdeauducq, 2016-09-26:
CERN OHL sounds good.

jbqubit, 2016-10-04:
CERN OHL sounds good to me

jordens, 2016-10-05:
@jmizrahi: ping

jmizrahi, 2016-10-05:
@jordens Can you summarize what you think the implications of the CERN OHL are for this project? It's unclear to me how the copyleft concept is applicable to hardware.

I have no objection to using the CERN OHL; it just seems like it's hard to come up with a modification which would be both substantive enough to be relevant, but also small enough so as not to be arguably a new piece of hardware to which the license doesn't apply.

jordens, 2016-10-05:
Sure. To me it seems that the most important points are:

* Hardware isn't generally copyrightable.  And distribution of hardware  is generally decoupled from the distribution of software/schematics/board layouts. 
  > "Copyright (on which most licenses are based) doesn't apply to hardware itself, only to the design files for it and, then, only to the elements which constitute original works of authorship (in U.S. law) and not the underlying functionality or ideas." https://opensource.com/law/15/2/intro-open-hardware-licensing
  > "The effect of this is that the license restricts the publication of a schematic or a laser-cutting plan, rather than the use of the schematic to make devices that are sold. You can control how people sell the plans with your license, but you can't keep them from commercially manufacturing and selling the devices with your non-commercial license, unless you have a patent." http://web.archive.org/web/20120114234704/http://wiki.openhardware.org/Hardware_Isn't_Generally_Copyrightable 
  > "The end result: while the design files of your open source hardware project can probably be protected by a license, the hardware itself probably cannot be. Unless you have a patent, someone else can manufacture the hardware without your permission, assuming they do so without infringing your copyright on the design files." http://www.oshwa.org/faq/

* Using the GPL (for example) for the schematics and the board layout would not imply much in terms of your rights (to get access to modified/improved/altered schematics and board layouts) as the user of a product (purchaser of the actual manufactured board).

* Tagging schematics and board layouts with e.g. the CERN OHL results in the requirement that recipients of products get access to the schematics and board layouts.

Now apply this:

* You take the Sayma schematics and board layout, and modify them by e.g. replacing the AD9154 DAC with one or several AD9164 (http://www.analog.com/media/en/technical-documentation/data-sheets/AD9164.pdf).
* Then your modified schematics and board layout would clearly be a derivative work.
* If you were to distribute the layouts and schematics you'd have to do so under the terms of the license (little difference between GPL and OHL here). 
* If you manufacture and distribute/sell hardware based on those modified designs, the GPL would not mandate what documentation the recipient/customer/purchaser of said product receives while e.g. the CERN OHL would.

jordens, 2016-10-05:
Obviously IANAL and I always enjoy to be proven wrong and learn something.

jmizrahi, 2016-10-06:
If someone modified a section of the board, it seems like they could compartmentalize that section as a black box, which communicates with the rest of the board through input and output traces. The rest of the board remains open, but that section is closed. It's not obvious to me how this is prevented by the CERN OHL, and since any modification could be framed that way, it seems like it would have to really be something that could not be compartmentalized, e.g. moving a trace to reduce crosstalk, in order to be unambiguously covered. Clearly embedding OHL hardware in larger hardware has no ramifications for that larger hardware, and the physical coupling does not really change as you go from proprietary PCB section, to proprietary daughter PCB, to proprietary distant PCB connected by cables.

jordens, 2016-10-06:
It is not prevented. In that sense the OHLs are more like the LGPL in the sense that you can "link" the work with proprietary stuff without inheriting the license.
It's about the schematics and board layouts. If in the process of adding the black box, you modify schematics/board layouts because you need to wire it up with power supplies, you will have to distribute the modified stuff with the product.
You don't have to distribute information about the inside of the black box.
And even providing access to (purchase) the black box is not required.
If the black box is actually embedded as individual parts in the modified PCB layout, you'd have to distribute that layout as well. If it is a castellated module (like the power supplies, the LCD PCB, or the JTAG module on the KC705), you wouldn't have to distribute those "internal" schematics/layouts (in my reading).

jmizrahi, 2016-10-06:
OK, thanks for the explanations. We can move forward with the CERN OHL.

closed 2016-10-06 by jordens