sinara-hw/sinara #7: rf connector choice and panel layout

jordens, 2016-09-28:
* which connector for DACs and ADCs
* which arrangement (straddling connectors...)

@jbqubit, @jmizrahi, @gkasprow

gkasprow, 2016-09-29:
I did some mechanical estimation with connectors I have in my library:
standard RA SMA: HUBER-SUHNER_85_SMA-50-0-101/111_NE
high performance edge mounted SMA: JOHNSON_142-0761-821
SMP: MOLEX_73415-3320

gkasprow, 2016-10-01:
I found ones with better performance and panel fixing 
http://pl.mouser.com/ProductDetail/Johnson-Cinch-Connectivity-Solutions/142-0701-871/?qs=iKljIZI%252bxs6VFAALEPTgCg%3D%3D

jordens, 2016-10-05:
The choice of SMA, panel fixing, and the staggered arrangement look good to me given the high density.
The one thing we have to look at very closely is the actual insertion and mounting process of the mezzanines. Since they are now mechanically constrained in the same degree of freedom at several points (carrier connectors constrain in-plane, screws out--of-plane, plus now the SMAs also constrain in-plane) the insertion could become tricky.

gkasprow, 2016-10-05:
I think we can make them mountable without removal of the front panel.

For this reason I placed all the connectors on the other side of the mezzanine than SMAs.

dhslichter, 2016-10-07:
>I found ones with better performance and panel fixing 
http://uk.rs-online.com/web/p/sma-connectors/7123254/
http://uk.rs-online.com/web/p/sma-connectors/8006949/

Are these candidates to replace the Johnson SMAs, or both kinds?  How do you get the stagger with these  connectors?

gkasprow, 2016-10-07:
I know them, I will try to fit on the mezzanine but could be hard.
With these two we can replace all connectors

dhslichter, 2016-10-10:
Note that Tom Harty explicitly requested SMAs for the panel where the SMA shield is not grounded by the panel.  He would like to be able to break low-frequency (~50/60 Hz) ground loops by capacitively coupling the SMA shield to the mezzanine board ground on the mezzanine boards.

gkasprow, 2016-10-10:
For this purpose we will use plastic washers

http://www.telegartner.de/en/telegartner-coaxial/spare-parts-connectors/insulating-washers

http://www.rf-microwave.com/datasheets/603_Huber-Suhner_77-Z-0-0-8_01.pdf

http://www.ebay.com/itm/3x-Bulkhead-SMA-Coupler-insulation-washer-Female-Jack-Gold-Plate-Adaptor-RF-EJ14-/360456131684

dhslichter, 2016-10-10:
Have you checked mechanical drawings to ensure that these plastic washers (and the correspondingly larger panel holes) will have enough clearance between neighboring SMAs?

Also, with such washers we will need to use a nut on the back side of the panel as well as the front side, with enough extra thread for the rear nut to allow for insulating washers to be installed or not installed without changing the distance between the edge of the mezzanine board and the inside surface of the panel.

gkasprow, 2016-10-10:
I'll check once I receive all mechanical drawings of the SMAs.
The producer does not provide step models.
Indeed, this could be potential problem with these washers.

jbqubit, 2016-10-11:
McMaster has insulating sleeve washers I've used for similar purpose.
http://www.mcmaster.com/#standard-washers/=14joufm

gkasprow, 2016-10-11:
Below are 3D files of the mezzanines and the RTM module with all mezzanines.
I placed all connectors that will be used.

[RTM_DACv2.pdf](https://github.com/m-labs/sayma/files/522859/RTM_DACv2.pdf)
[mezzanine4.zip](https://github.com/m-labs/sayma/files/522862/mezzanine4.zip)
[RTM_DAC.zip](https://github.com/m-labs/sayma/files/522863/RTM_DAC.zip)

dhslichter, 2016-10-11:
Looks excellent!  Two question:

- how much height clearance is there on the ADC side of the board for shields?  We need at least .175" as the absolute minimum (.165" interior + .010" shield thickness), .180" or .185" would be better.  We could make the mezzanine boards slightly thinner if need be (e.g. .047" instead of .062" nominal)
- when providing 100 MHz clock to the mezzanines (e.g. for integrated VCO/PLL chip on mezzanines), is it coming via cable or via Sayma?  Where/what will connectors be?

gkasprow, 2016-10-11:
We can fit Wurth shields that are 3mm high for ADC stuff. We can make the boards 1mm to gain additional 0.6mm. This would give .140".
The main limiting factor is minimal SMP stacking height which is 9mm including the bullet.
The AMC specification says that max component height is 14.41mm from PCB centre. Including board thickness it gives about 13.6mm
Here is drawing from MTCA spec
[image]

So if we substract 9mm from 13.6, then pcb thickness we will get:
13.6-9-1.6=3.0mm
But, the distance between AMC slots is 20.32mm. So we have 1.36mm in addition - it is reserved for EMC spring or foam on side of the panel. The shield is not very delicate component so we can slightly violate the specification and still have the board that can be easily inserted.
In this way using 1mm thick mezzanine and taking some space reserved for EMC spring we will gain another 0.6+1mm + 3mm =4.6mm which gives 0.181"

dhslichter, 2016-10-11:
My math is different: looks like max envelope is 14.01 mm from board center, or 13.2 mm from face of Sayma board.  Minus 9 mm SMP height gives 4.2 mm total, including mezzanine board thickness.  There's no way to get the necessary clearance in there.  I think we will just have to have shorter shields and thinner components on the ADC side.

The principal offenders for height are baluns; if we restrict ourselves to some subsets of available baluns, we can live with a height of 0.11" (2.8 mm) under shields instead of .165".  Thus if we make the mezzanines a little thinner, and/or are willing to go to the full 14.4 mm instead of 14.0 mm, we can make it work with this thickness (assuming .25 mm shield material thickness).

gkasprow, 2016-10-11:
all depends if you want to fit below the panel or not:)
I confirmed it with measurement on 3D model

dhslichter, 2016-10-11:
Modified #8 lead thread to reflect new specification.  I recommend closing this thread and continuing discussion on #8 unless there is objection.

dhslichter, 2016-10-28:
Reopening issue to reflect new ideas for compact connectors.

dhslichter, 2016-10-28:
See #8.

dtcallcock, 2016-11-02:
I got quotes for the Carlisle connectors.  Prices seem reasonable.  

[Core HC](http://tmlaunch.carlisleit.com/core-hc/)

CHB-S-04-VL-30
CORE HC 4 POSITION VERTICAL PCB TO CABLE
$56.82 (Qty 100-249)

CHC-S-04-2-36-12-C
CORE HC CABLE ASSEMBLY 4 POSITION
$123.80 (Qty 100-249)

[Core GD](http://tmlaunch.carlisleit.com/core-gd/)

CGB-4P-S-04-VL-00-30
SSMP MALE,4 POSITION, PCB VERTICAL MOUNT
$37.87 (Qty 100-249)

CGC-4R-04-2-36-12-C
Cable Assembly, SSMP FEMALE TO SMA MALE, 4 POS.
$145.07 (Qty 100-249)

However, as you can see, they only have the vertical mount PCB connector in production right now.  I phoned them today and they say the Edge mount Core GD connector is about 6-8 weeks from production.  They have test data and board layouts available that they could release to us much sooner if we want to go with this option.

[image]

gkasprow, 2016-11-03:
Do they have 3d step models for these connectors?

dtcallcock, 2016-11-03:
The step files for the vertical mount and the cable assembly are on the website.  I have asked if I can get preliminary ones for the edge mount.

jbqubit, 2016-11-04:
@gkasprow @dtcallcock In conference call today we discussed use of SMA vs SSMP. Consensus was that for first prototype round we will use SMA on analog mezzanine boards. Strong motivators of this choice a) need to lock-in hardware design so prototype can be completed quickly b) long lead time for Edge mount Core GD (relative to 1st prototype schedule). 

@dtcallcock Are you interested/willing to evaluate SSMP for consideration in second prototype round?

dtcallcock, 2016-11-04:
@jbqubit  Using SMA for the prototype is clearly sensible.  I will evaluate the SSMP connectors and open a new issue when I have all the details in a few months.

dtcallcock, 2016-11-04:
Preliminary STEP files forwarded to @gkasprow

gkasprow, 2016-11-04:
It's not possible to fit 4-way connectors on front panel.
8-way connectors will fit.

hartytp, 2017-11-29:
Reopening this issue: as anticipated, having this many SMAs on the front panel is a pain in the neck, so it would be good to find a better solution for v2...

gkasprow, 2017-11-29:
On some devkits there are multi-port coaxial connectors like these
[image]

gkasprow, 2017-11-29:
but they are only [vertical](http://suddendocs.samtec.com/ebrochures/bullseye_flyer_40_ghz.pdf)
Another interesting from [Molex](https://www.avnet.com/wps/portal/abacus/products/manufacturers/molex/products/mprf-solutions/), [Samtec](https://www.samtec.com/rf/components/original/ganged), another [Samtec](https://www.samtec.com/connectors/high-speed-board-to-board/high-isolation-rf/isorate)

jordens, 2017-11-29:
Oh yes. I remember seeing that stuff on some OHWR/CERN boards as well.

What are the single coax footprints? Just coax testpoints?

gkasprow, 2017-11-29:
@jordens you probably saw the GBTX devkit :) Unfortunately they are only vertical ones. The footprint is just SMT pad.

hartytp, 2017-11-29:
I'm coming to the conclusion that this is the best way to go. 

- Probably makes most sense if we make the AFEs take 4 DAC channels.
- Find the cheapest 8-pin RF connector with decent specifications (including isolation) and put that on Sayma.
- Then ship Sayma with a short (say 30cm) cable that breaks the multi-pin connector out into SMAs. IMO, that's really not much less convenient than just putting the SMAs on the PCB -- well, it's more convenient once one takes into account that the SMAs are almost impossible to fit atm ;)

This isn't a priority for right now, but at some point we should make a list of potential connectors, with info like: price and availability/MOQ both for the connector and breakout cable; return loss + insertion loss + isolation (say at 1GHz). Then we can make a decision about this.

dhslichter, 2017-11-29:
From what I can see the performance of all the "cheap" sorts of cables is not that hot, see e.g. http://suddendocs.samtec.com/testreports/rfc-report_ij5h_web.pdf . 

This connector issue is a mezzanine-specific consideration -- for example, if one wants an upconverting/downconverting AFE mezzanine for Ca+ or Yb+, then these connectors will definitely not cut the mustard.  You'll want a Rogers board and SMP on the front end.  We should just make a large enough hole in the front panel of the RTM card that one can put through a variety of possible connectors.  One could consider having a couple of screw holes on the front panel such that one could screw a custom plate over the end for increased shielding if desired.

gkasprow, 2017-11-29:
@dhslichter we have plastic insulating front panel anyway, so we can simply modify it and use bigger rectangular cut-outs. Then everybody can 3D-print its own panel based on template, using free tools.

dhslichter, 2017-11-29:
I was going to suggest machining something flat but I guess 3D printing works too :)

dhslichter, 2017-11-29:
I thought the front panel was metal and we had plastic standoff washers.  Not true?

gkasprow, 2017-11-29:
Not. Plastic washers don't fit. they collide with neighbouring ones. So I designed and machined plastic panel.
[image]

hartytp, 2017-11-29:
> @dhslichter we have plastic insulating front panel anyway, so we can simply modify it and use bigger rectangular cut-outs. Then everybody can 3D-print its own panel based on template, using free tools.

I was going to suggest something like this: let's do the same thing they do for FMCs and have the Sayma FP have a large rectangular cutout for each AFE mezzanie. That way the mezzanines can have their own front panel suited to the type of connector they use.

The upconverting mezzanines need good microwave connectors, however the channel density is halved (2 DAC channels per output) so the SMAs fit nicely.

The baseaband mezzanine Allaki doesn't need to go to such high frequencies so can potentially use a higher density connector.

hartytp, 2017-11-29:
> From what I can see the performance of all the "cheap" sorts of cables is not that hot, see e.g. http://suddendocs.samtec.com/testreports/rfc-report_ij5h_web.pdf .

That's a shame. Is there no reasonably priced alternative that goes to a few GHz with a RL of something like 20dB?

hartytp, 2017-11-29:
@gkasprow If we do go for one of these connectors on Allaki then we should make sure that we pick one that's suitably rugged -- remember this is an external connector for lab use!

gkasprow, 2017-11-29:
@hartytp I know. Sometimes I have an impression that "for lab use" means compatible with wild animals.

hartytp, 2017-11-30:
> wild animals

We prefer to call them "students"

dhslichter, 2017-11-30:
I think @hartytp and I have a mind meld here -- do something like the FMCs and have customized front panels for each AFE that can go on the AFE and screw into or otherwise electrically contact the RTM metal front panel (e.g. with gaskets).  

We should design the AFE boards to mount with 9 mm standoffs to the RTM, with screws, and then the front panels don't have to be screwed in, just slid in and use gaskets.  This also helps with the issue of mechanical mounting as there is substantial wiggle room at the front panel during the installation.

hartytp, 2017-11-30:
@dhslichter Exactly!

dhslichter, 2017-11-30:
It is a shame about the cable performance at a few GHz for the ganged cables, but AFAICT this is what you get from "cheap" varieties.  If we go to SMP for the front panel connectors and use SMP-SMA patch cables, you can achieve higher densities than SMA with still excellent electrical performance.

hartytp, 2017-11-30:
> It is a shame about the cable performance at a few GHz for the ganged cables, but AFAICT this is what you get from "cheap" varieties. If we go to SMP for the front panel connectors and use SMP-SMA patch cables, you can achieve higher densities than SMA with still excellent electrical performance.

I'd be interested to see what the cost difference between these two alternatives is.

I'm basically fine with either approach. Since this only applies to the baseband mezzanine (the unpconverting mezzanine has space for SMAs) the frequency is a bit lower. IMHO the performance of the cables you found will be fine for Allaki. (But maybe @gkasprow knows of one with slightly better performance and still a reasonable cost?)

dhslichter, 2017-11-30:
Honestly, for Allaki I would be inclined to do something cheap and super-standard like MMCX on the front panel, which then lets people choose their own cables to break out to SMA.  You can get better performance out of standard MMCX (max VSWR dc-6 GHz is 1.3), and it lets people choose the particular type of coax they are using (if, for example, phase noise is important to them).  You could use the Samtec GRF series, which is MMCX-compatible, but those have less-than-stellar crosstalk (~60 dB, even at only 200 MHz). The 8-channel GRF jack connector is $42, 4-channel jack is $24.  Compare to MMCX, which is $2-$5 each for on-board connectors -- it's basically a wash.  Similar with the pricing on breakout cables: 1 m (!!!) GRF breakout cables to SMA for 4 channels are $75, similar length one-off MMCX-SMA cables are $20-$25 each.  We would likely want shorter cables anyway.

dhslichter, 2017-11-30:
https://www.mouser.com/ProductDetail/Johnson-Cinch-Connectivity-Solutions/415-0071-006/?qs=sGAEpiMZZMufBZYvsU%2fbewR0O3E3GfwIVaspFd536l4%3d for example, cheap at scale.

hartytp, 2017-11-30:
@dhslichter My main concern with MMCX/SMP would be mechanical ruggedness. I don't have much experience with those connectors, but are they really rugged enough for external/front panel use in a lab?

> Samtec GRF series, which is MMCX-compatible

GRF look better mechanically, and MMCX compatibility is a big plus.

60dB cross-talk at 200MHz is pretty poor, but probably okay for most applications. IIRC, the DAC cross-talk is something like 77dBc at 180MHz, and the cross-talk on Allaki itself hasn't been measured yet. So, it's not clear that this cable will degrade the overall cross-talk by *that* much...

dhslichter, 2017-11-30:
SMP connectors ~$3-5 each at scale:
https://www.digikey.com/product-detail/en/molex-llc/0734153591/WM9363-ND/3468860

SMP/SMA cables are ~$30 each at scale, so again maybe a $5-$10/channel bump relative to the cheaper MMCX stuff, but with better crosstalk and higher frequency capability -- one could use the same cables and connectors for either Allaki or a Sayma upconverting AFE, so that is a win for people changing AFEs without having to scrap all their cable arrangements.

dhslichter, 2017-11-30:
> are they really rugged enough for external/front panel use in a lab?

If we are breaking out SMP to an SMA panel, I think there should be no problem at all.  If not, as long as you use full detent connectors on the PCB, you should also be fine.  In general, I think doing limited detent and breaking out to a nearby SMA panel is going to be perfectly fine from a physical standpoint.

dhslichter, 2017-11-30:
Either SMP or MMCX are very standard and you can find a lot of different parts available from many vendors if you want to make/buy your own custom cabling for whatever reason.  Wrangling with Samtec to get them to use a different kind of coax in their GRF connectors could be a major PITA.

dhslichter, 2017-11-30:
> So, it's not clear that this cable will degrade the overall cross-talk by that much...

Sure, but every bit counts, and I'd rather not have cable choices be the weak link in the chain.

hartytp, 2017-11-30:
@dhslichter ACK. If they are sufficiently rugged then MMCX for Allaki sounds like a good choice. As you say, we can bulk-buy some short MMCX to SMA pigtails and ship them with Allaki if people want.

gkasprow, 2017-11-30:
@dhslichter what about microBNC? They are used for 10G professional video equipment. Moreover the footprint is compatible with SMA.

hartytp, 2017-11-30:
If the cost and availability are okay then that would be a nice choice. Probably overkill for Allaki though, as that's unlikely to be used much above 1GHz (for higher frequencies Nanonga is probably a better choice!).

dhslichter, 2017-11-30:
A cursory glance suggests the pricing is similar to (or slightly higher than) SMP, similar performance as well.  You'd have more idiot-proof locking with the bayonet but it's not clear that it's a big win over SMP otherwise.  Availability is a bit tricky because HD-BNC is generally 75 ohm for video applications, so finding 50 ohm parts is a bit tougher.  Recommended pitch is 8 mm, slightly higher than SMP.

jbqubit, 2017-11-30:
Samtec IP5 are also available as [90-deg connectors](http://suddendocs.samtec.com/ebrochures/iso_rate_eb.pdf). And there's a mating assembly called [IJ5H](https://www.samtec.com/products/ij5h) for breakout to coax. For example, 8-position to SMA (IJ5H-08-1000-S-2-01SP1) is < $200/ea qty 10 octopart. 

[image]

hartytp, 2017-11-30:
@jbqubit True, but that's not cheaper than SMP, and certainly less convenient IMHO. Might be mechanically better though...

dhslichter, 2017-11-30:
The RF performance is not very good unfortunately - http://suddendocs.samtec.com/testreports/rfc-report_ij5h_web.pdf

If we are not realizing dramatic cost savings vs SMP, I think there isn't a very strong case, since we are planning to break out the SMP to an SMA panel and thus relieve the mechanical issues associated with multiple vs single ganged connectors at the RTM front panel.  I think there is a lot of value in using a standard connector such that if people decide (for example) that they want to use semi-rigid or hand-flex for reduced phase noise, they can buy the necessary parts commercially from a variety of vendors.  I think that if it's user-facing, all other things being equal we should go for the option that is most commonly available/user-tweakable.

[6 more replies]

closed 2017-12-01 by hartytp