sinara-hw/sinara #8: analog mezzanines

jordens, 2016-09-28:
* shape
* rf connectors (type, mechanics, numbers)
* headers for misc signals

@jbqubit, @jmizrahi, @gkasprow

gkasprow, 2016-09-29:
I did some estimation.
We can easily fit four 32x100mm DAC/ADC  mezzanines on RTM. We can even make them longer.
On each of them we will fit four Wurth 30x30mm shields placed on both sides.
In case of RF connectors I consider using:
2 high performance SMA, edge mount for DAC output
2 lower performance, right angled for ADC input.
8 SMP connectors for ADC/DAC. THT version with SMD central pad would give best mechanical properties. The distance between mezzanine and RTM is 9mm which is minimal value for SMP+bullet.
For digital IO and control I'd like to use Samtec hermaphroditic, 40 pins connectors: LSS-120-01-L-DV-A and LSS-120-02-L-DV-A  which combined give exactly 9mm. I use them in other projects and they work well with frequent mating.
Mechanical mounting would be ensured by 4 x 2.6mm holes, 4 threaded spacers and 8 M2.5 screws.
There is still place on front panel for SMA clock input 
I placed typical bulky SMA connectors to check the clearance between them. It is not that bad. Then are accessible with the wrench when the RTM is outside of the crate.
Attached are PDF3D files with mezzanine and RTM.
[mezzanine.pdf](https://github.com/m-labs/sayma/files/501839/mezzanine.pdf)
[RTM_DAC.pdf](https://github.com/m-labs/sayma/files/501840/RTM_DAC.pdf)
We will leave the space below mezzanines to enable the air flow. Some cut-outs in RTM may be necessary as well.

gkasprow, 2016-10-01:
The clock module dimension is 43x80mm
We can make it slightly longer when necessary
here is the RTM with clock module
[RTM_DAC_clk.pdf](https://github.com/m-labs/sayma/files/504373/RTM_DAC_clk.pdf)

jordens, 2016-10-01:
Thanks! Impressive.

* Is this still with the SMA firmly screwed into the front panel? How would the mezzanines mounted in that case?
* Did we want no common LO clock (for whatever upconversion needs there may be) to the mezzanines?
* I assume that there is still space for the dacs, adcs, power supplies, and clock fanout?

sbourdeauducq, 2016-10-01:
I cannot open any of those PDFs, either with mupdf or Evince.
```
$ mupdf mezzanine.pdf 
error: expected trailer marker
error: cannot parse trailer
error: cannot read xref (ofs=617742)
error: cannot read xref at offset 617742
```

whitequark, 2016-10-01:
@sbourdeauducq Yes. As far as I know PDF3D can be opened only with proprietary Adobe Reader for Linux.

sbourdeauducq, 2016-10-01:
Adobe Reader freezes when trying to open those documents.

sbourdeauducq, 2016-10-01:
Is there another format we can use for this?

gkasprow, 2016-10-01:
we can use step

gkasprow, 2016-10-01:
To view step files I use free Design Spark Mechanic.

whitequark, 2016-10-01:
FreeCAD is very bad but it should suffice to view STEP files.

gkasprow, 2016-10-01:
Below are step files. Due to 10MB limit I zipped them

gkasprow, 2016-10-01:
Due to 10MB limit I had to zip them
[CLK module.zip](https://github.com/m-labs/sayma/files/504635/CLK.module.zip)
[mezzanine2.zip](https://github.com/m-labs/sayma/files/504636/mezzanine2.zip)
[RTM_DAC_clk.zip](https://github.com/m-labs/sayma/files/504637/RTM_DAC_clk.zip)

gkasprow, 2016-10-01:
In case it pdf 3D files you have to use recent Adobe Acrobat Reder

gkasprow, 2016-10-01:
> Is this still with the SMA firmly screwed into the front panel? How would the mezzanines mounted in >that case?

the mezzanines are mounted to the RTM board using 4 M1.5 screws 

>    Did we want no common LO clock (for whatever upconversion needs there may be) to the >mezzanines?
I'll add another SMP connector
   > I assume that there is still space for the dacs, adcs, power supplies, and clock fanout?
sure, some of them can be plalced below the mezzanines

gkasprow, 2016-10-01:
>    Is this still with the SMA firmly screwed into the front panel?

There are versions with long thread both right angle and straddle mounted.
But their performance need to be verified since they are low cost.
http://www.amphenolrf.com/132289.html
http://www.rfstreet.com/ch_contentd.asp?id=2043
on the other side the assembly of the modules changed - they will be firmly attached to the rigid RTM board by 4 screws so it may be not necessary to screw them to the front panel

sbourdeauducq, 2016-10-02:
Thanks. The STEP files work fine with FreeCAD. FYI, the Acrobat Android app and the Chrome PDF viewer also choke on the PDF3Ds.

sbourdeauducq, 2016-10-02:
What is the idea behind having more SMPs than SMAs on the mezzanines? Since this also results in more SMPs than DAC/ADC channels on the RTM carrier, how will they be connected?

gkasprow, 2016-10-02:
Did you enable the 3D view in AR?

It needs a while while opening the file

Greg

gkasprow, 2016-10-02:
The signals of the ADCs and DACs are differential, that's why we need the SMPs doubled.
The mezzanines inputs and outputs are single ended.
The idea is to place baluns on the mezzanines because the way one feeds the ADC or couples to DAC depends on application.

sbourdeauducq, 2016-10-02:
Aren't SMPs designed for single-ended unbalanced signals?

gkasprow, 2016-10-02:
Well, in this way you can say that traces on PCB are designed for single
ended signals.
If you make differential traces that are loosely coupled, you can treat
them as single ended. When you make traces tightly coupled, then they
interfere with each other and if you split them then their impedance will
increase.
The same applies to SMAs - on every devkit you can connect gigabit
transceivers using single-ended connectors.
To transfer signals using SMPs you have to make impedance conversion from
tightly coupled diff line to loosely coupled diff line and then transfer
via connectors.
So to make 100Ohm diff line you can use 2 traces wit 50Ohm but separated
far away or use two 70Ohm traces coupled in such way that trace field see
each other. In first case you can transfer them wia SMP, SMA or whatever,
in second case you have to transfer the line to the first case.

jordens, 2016-10-03:
This works perfectly fine. Convince yourself using the method of image charges and symmetries.

dhslichter, 2016-10-03:
It's very hard to find true differential signal connectors where the channel-to-channel crosstalk is anywhere near as low as if one uses two coaxial connectors (e.g. SMP) for a differential signal.  Sending the ADC/DAC differential signals over pairs of SMP connectors is the best compromise we could come up with given crosstalk/space/price/bandwidth/mating limitations and requirements.

jbqubit, 2016-10-04:
Impressive drawings! 
> Wurth 30x30mm shields placed on both sides
- Should these also cover the back side where the SMP penetrate? 
- A single shield on each side would give more freedom for component placement. 
- What's the maximum vertical height for components? Is this sufficient for bulky components like baluns, transformers?

dhslichter, 2016-10-04:
Several points:

- if board will have components on both sides, should use Rogers 4350B on both sides.  OK to make it thinner than .020" -- 50 ohm CPWG on .020" Rogers 4350B is 35 mil trace with 11 mil gaps, based on measurements at NIST.
- shields should be multi-cavity (if a single part) to reduce crosstalk between channels on the same daughtercard.  Something along the lines of https://leadertechinc.com/product/multi-cavity-slot-lok/ would be good.
- interior height under shield lids needs to be .165" minimum to accommodate baluns
- crosstalk reduction would be improved if shields extend all the way to the end of the board, such that SMP connectors are connected through holes in the shields.  Unshielded lengths should be short to the SMA connectors.  Notches need to be made in the shields to avoid impedance discontinuities where CPWG traces exit from under the shields to the SMA connectors.

gkasprow, 2016-10-04:
The Wurth shields were only placed just to reserve place on PCB. There are
smaller ones like these
http://www.digikey.com/product-search/en/rf-if-and-rfid/rf-shields/3539677?k=&pkeyword=&pv1217=1&FV=fff40036%2Cfff802dd%2C13040001%2C13040009&mnonly=0&newproducts=0&ColumnSort=0&page=1&quantity=0&ptm=0&fid=0&pageSize=25

dhslichter, 2016-10-07:
Again, the shields you quote are not tall enough inside, nor would they extend the full size of the board.  I understand the Wurth shields are placeholders, but I think we will probably want something custom/multicavity that covers as much of the board as possible while separating the various channels from each other.

gkasprow, 2016-10-07:
In this case we can use custom Harwin shields (EZ-shield) where they can be
freely adjusted with 5mm step.
You just install clips on the PCB and then manually form the cover.
http://www.mouser.com/new/harwin/harwin-ez-emi-rfi-shield-cans/

jordens, 2016-10-07:
OK. @dhslichter and @gkasprow if you feel that you have arrived at a workable solution, please close the issue. AFAICT the rest has weighed in.

dhslichter, 2016-10-07:
@gkasprow can you clarify on which SMA connectors you want to use on the front panel, and their layout?  Let's try to put the whole spec into a single comment at the end of the thread before we close.

jbqubit, 2016-10-10:
Here's my summary of current design choices for mezzanine interface (and geometry) for each of 4 analog boards. @jordens @gkasprow @sbourdeauducq @dhslichter please review. @weidazh please forward to Harty

**geometry and mechanical**
- what is mezzanine geometry?
- the mezzanines are mounted to RTM motherboard using 4 of M1.5 screws

**RF shield**
- interior height under shield lids needs to be .165" minimum to accommodate baluns

**PCB material/layers**
- 4-layer PCB, top and bottom layers .020" Rogers 4350B
    - 50 ohm CPWG on .020" Rogers 4350B is 35 mil trace with 11 mil gaps
- single ground for analog and digital

**connector choices**
- power and IO
  - Samtec hermaphroditic, 40-pins connectors: LSS-120-01-L-DV-A and LSS-120-02-L-DV-A
  - 9 mm stand off between motherboard and mezzanine 
  - what is maximum current?
- RF to/from motherboard
    - analog output: 4 SMP (2 differential pairs)
    - analog input: 4 SMP (2 differential pairs) 
    - clock: 2 SMP (1 differential pair) generated on sayma_clk mezzanine 
    - what is part number for SMP's? Discussed in #7 but not yet finalized? 
- RF to outside world
    - SMA choice discussed in #7 but not yet finalized? 

**power/IO header routing**
- digital
     - 18 digital lines total 
     - no pull-up resistors
     - +3.3V LVCMOS single-ended to/from FPGA
     - recommended use:
         - pin names in schematic should include names indicative of recommended use case
         - I2C [I2C_SDA, I2C_SCL] 
         - SPI [SPI_SCLK, SPI_MOSI, SPI_MISO, SPI_CS0, SPI_CS1, SPI_CS2, SPI_CS3, ...]

- analog power, low noise (generated on AMC)
    - +12V at 200 mA

Does github permit making an Issue comment as "sticky" so it rises to top? Seems like that's where the summary should lie. 
    - -12V at 200 mA
    - +7V at 1 A
    - -7V at 1 A
    - +3.3V at 1 A
    - other potentials generated on-mezzanine using low-dropout regulators (LDR) -- no switching regulators on mezzanine
    - schematic notation to distinguish between analog and digital power
- digital power
    - +3.3 V at 1 A

**ADCs**
- 4 channels

jbqubit, 2016-10-10:
@jordens Does github permit making an Issue comment as "sticky" so it rises to top? Seems like that's where the summary should lie.

gkasprow, 2016-10-10:
During our last discussion we decided:
+6V 1.5A. This minimizez LDO losses. We don't need -7 for RF stuff.
+12V 200mA
-12V 50mA
no need for other power lines.
I'd add local I2C line for module identification - the same mechanism as is on the FMCs.
In this way we will quickly know which modules are installed in certain mezzanines. With all system enclosed in the rack it's good to know what is inserted in each RTM.
To save pins, the I2C lines can be connected to the I2C muxes and later on to the Artix FPGA and in parallel to the IPMI link. To supply EEPROM and some digital stuff 3.3V is needed with low current capability.
Now the connector looks like this;
[image]

dhslichter, 2016-10-10:
Here is the final summary, based on comments from @gkasprow  and @jbqubit in this thread.  Items labeled **_ACTION:_** are not finalized yet; I will edit this comment with the correct information once they are finalized.  @jordens @gkasprow @sbourdeauducq @jbqubit @weidazh please provide feedback.

## Mezzanine geometry/stackup
- mezzanine board 32 x 100 mm, .062" nominal thickness, 4 layer boards
    - top and bottom layers to be Rogers 4350B, .020" thickness (cap construction for PCBs)
    - 50 ohm CPWG on .020" Rogers 4350B is .035" trace with .011" gaps
    - need ground stitch vias every ~.050" or less along both sides of CPWG
- mezzanines are mounted to RTM motherboard using 4x M1.5 screws
- 9 mm stand off between motherboard and mezzanine
- RF shields on both sides, separate cavities for each DAC channel and ADC channel (4 total per mezzanine)
    - interior height under shield lids needs to be .170" minimum (allowing for fabrication tolerances) to accommodate baluns

## Connectors

- power/IO:  Samtec hermaphroditic, 40-pins connectors: LSS-120-01-L-DV-A and LSS-120-02-L-DV-A
- RF to/from motherboard
    - DACs: 4 SMP (2 differential pairs)
    - ADCs: 4 SMP (2 differential pairs)
    - clock: 2 SMP (1 differential pair) generated on sayma_clk mezzanine
    - **_ACTION:_** need to finalize SMP part number
- panel RF connection
    - option to have SMA shield isolated from Sayma ground (i.e. not grounded to panel)
    - **_ACTION:_** need to finalize SMA part numbers

## Power rails

- +12VDC @ 200 mA, low noise 
- +6VDC @ 1.5 A, low noise
- -12VDC @ 50 mA, low noise
- **_ACTION:_** need to specify noise requirement for above rails
- desired supply voltages to be generated from these rails on mezzanine with dedicated LDOs
- no switching supplies on mezzanines
- **_ACTION:_** +3.3VDC @ 1 A for digital-only devices on mezzanine?
- single shared ground for analog and digital, connected to all SMP shields and all ground pins of power/IO connector

## Power/IO header routing

- as shown in image below from @gkasprow 
- digital lines IO0-IO15 3.3V LVCMOS
- **_ACTION:_** need to choose I2C SDA/SCL voltage, and whether pullups are on Sayma or mezzanine (recommend Sayma).  Easy/compact level shifting on mezzanines (if needed) using discrete MOSFETs: http://www.nxp.com/documents/application_note/AN10441.pdf
- digital IO are general purpose in hardware, but for SPI preferred pinout is:
    - SCLK = IO0
    - MOSI = IO1
    - MISO = IO2
    - CS = IO3
    - additional chip selects starting with IO4 and up
- power rails as shown (7V rail is actually 6V)
- 4x connections to slow ADC for drift/temp/etc monitoring. **_ACTION:_** confirm ADC to live on Sayma board, not on mezzanine.

[image]

gkasprow, 2016-10-10:
I2C voltage is 3.3V.
The calibration ADC is on the RTM board, controlled by Artix FPGA>

gkasprow, 2016-10-10:
SMP part no is: 
MOLEX 85305-0232 for DAC signals
Amphenol SMP-MSLD-PCT-10 for ADC signals 
With such scenario we will keep DAC signals on same layer and ADC signals on opposite layer.
SMA part no is:
Telegartner J01151A0451 for DAC signals. It is the only choice for desired  performance.
Telegartner J01151A0921 for ADC signals. This selection is forced by type of SMAs used for DACs. No other type of connector will fit.

dhslichter, 2016-10-10:
There are still 4 unused pins on the power/IO header.  Would it make sense to use these to make the ADC input lines into differential pairs?  Thus there would be 8 pins total going to the slow ADC, which could be either used as 4 differential pairs or 8 single-ended lines with a chip like http://www.analog.com/en/products/analog-to-digital-converters/ad-converters/ad7173-8.html

gkasprow, 2016-10-10:
> converters/ad-converters/ad7173-8.html

I'd reserve them for future use.
For sure we will find some application for them

dhslichter, 2016-10-10:
@gkasprow especially on the +12V and +6V rails, we need to make sure that the power line filter components are carefully chosen.  0201 package 1 nF capacitor will likely have very substantial DC bias dependence to capacitance (esp. if caps are only 16V rated, I would use larger margin for +12V rail), for example, and we should check choke impedance at 1.5A load for +6V rail.

dhslichter, 2016-10-10:
Temperature monitoring can also be performed on mezzanines by using [ADT7420](http://www.analog.com/en/products/analog-to-digital-converters/integrated-special-purpose-converters/integrated-temperature-sensors/adt7420.html), which contains conversion and I2C hardware, thus freeing slow ADC lines for e.g. monitoring of I/Q modulator drift or amplifier gain drift.

gkasprow, 2016-10-10:
we can use 0402 NP0 instead

gkasprow, 2016-10-10:
The chokes are rated for 3.5A BLM41PG181SN1L

On 11 October 2016 at 00:23, Grzegorz Kasprowicz <
G.Kasprowicz@elka.pw.edu.pl> wrote:

>
> we can use 0402 NP0 instead
>

dhslichter, 2016-10-10:
I vote 0402 NP0 for the 1 nF, and X7R with higher voltage rating on the 10 uF if possible.  Chokes look good.

sbourdeauducq, 2016-10-11:
> thus freeing slow ADC lines for e.g. monitoring of I/Q modulator drift or amplifier gain drift.

We also have an ADC inside the XC7A15T.

jordens, 2016-10-11:
@jbqubit rises to the top of what? What you perceive as important might not be what is currently relevant to others. If you are certain this particular issue is more important than the other "important" issues, give it the aproriate label. You have lots of options to view issues in the way you prefer.
* make yourself a [project dashboard](https://github.com/m-labs/sayma/projects) where you can organize issues for yourself
* add [labels](https://github.com/m-labs/sayma/labels) and 
* create permanent [search links](https://help.github.com/articles/searching-issues/) with special [sorting](https://help.github.com/articles/sorting-issues-and-pull-requests/) to [stuff with certain labels](https://github.com/m-labs/sayma/issues?q=is%3Aissue+is%3Aopen+label%3Aprio%3A2-important) and
* bookmark them or add them to a [wiki page](https://github.com/m-labs/sayma/wiki) or [share them](https://help.github.com/articles/sharing-filters/) by other means.

@dhslichter looks good to me (apart from my usual concerns about layout density)
* Pi-filter (add caps after the chokes) so that the loads don't look into the high impedance chokes?
* I'd either keep editing the top post so that the status quo is immediately visible, or - if history is desired - edit a wiki page or text file in the repository for this, or split this into smaller issues.

dhslichter, 2016-10-11:
@sbourdeauducq I am less concerned about the number of available ADC channels, was more thinking about the number of ADC signal lines coming out of the mezzanine (set at 4 right now).  For a typical IQ modulator setup, we might be monitoring voltages (via lowpass filters) at the I port, Q port, RF port, and after the output amp, thus would use all the ADC lines in the I/O header for those and not have any for temp monitoring; the I2C temp monitor helps with that.

dhslichter, 2016-10-11:
@jordens ack.

dhslichter, 2016-10-12:
@gkasprow where will the SMP connectors providing a 100 MHz clock to the mezzanines go?

gkasprow, 2016-10-12:
Do you mean RF_LO signal?
The idea is to provide RF clock for up/down converters on the mezzanines.
So I want to place SMP connectors routed to the clock module which may in the future provide such signal
In case of the EXT_100M_RF_IN, it is routed directly to the SMA on RTM front panel.

dhslichter, 2016-10-12:
I am thinking of a reference clock (e.g. 100 MHz) that will allow one to place a VCO/PLL on the mezzanine for generation of an LO to be mixed with the I/Q signals from the DACs to give an upconverted waveform.  We had discussed the option of this architecture in addition to having the LO generated off-card and sent to the mezzanines via front-panel SMA.  

So yes, one would want to have an SMP pair routed to the clock module from the Sayma board, or else via an SMA cable from the clock mezzanine.  I believe Tom Harty had some notions on this laid out.

hartytp, 2016-10-13:
@dhslichter @gkasprow I've attached a sketch showing the way I assumed we'd generate + distribute the DAC clock + LO. I've missed some of the recent discussions, so apologies if this is hopelessly out of data/no longer applicable!

Comments:

1.**I'd prefer not to generate the LO on the main clock mezzanine**: since the range of DAC clocks we need is fairly narrow (say, 500MHz - 2.4GHz), we can make a single DAC clock PLL design, with VCO + loop-filer population options to set the frequency. In contrast, the range of LOs spans ~1GHz to >10GHz. The PLL one uses to generate a 12GHz LO for Ytterbium is likely to look very different to, say, a 3GHz LO for calcium. e.g. the Ytterbium PLL might be a 2-loop design, with a 100MHz=>1GHz loop, followed by a 1GHz=>12GHz loop, while the calcium can probably be done as a population option for the DAC clock board. Separating these jobs onto different mezzanines allows one to use the same DAC clock board, regardless of what LO one needs, which seems nice.

2.**We should route the 100MHz reference clock to the analog mezzanine**: for LO frequencies up to a few GHz, one can buy IQ modulators with integrated PLL + VCO to generate their own LO from the 100MHz reference clock (e.g. HMC1197). These provide a very convenient, low-noise method of upconversion.

3.**We should have the option of installing a LO generation mezzanine (separate to the DAC clock mezzanine)**: one can put an ultra-low noise PLL on this board to generate the LO from the reference clock. If this is done on the output mezzanines then it has to be duplicated 4 times per Sayma board, which wastes space + money. This is particularly important for LO frequencies above a few GHz, where one would want the option of a using a fancier 2-loop PLL to generate the LO. (note also that the HMC1197 with PLL + VCO is 7mmx7mm & $39 from Digi-Key in singles, while HMC1097 without PLL + VCO is 4mmx4mm & $18).

4.**We can minimise routing complexity/SMP count by using clock fan-out + mux ICs**: high-quality mux/buffers like the ADCLK948 are extremely convenient and will not degrade the phase noise/stability of the DAC clock or LO.

5.**It's nice to have the option of routing the 100MHz reference directly to the HMC7044**: that way, we can use the HMC7044's integrated PLL + VCO to generate the DAC clock, without requiring a clock mezzanine or external clock source. This is a cheap, convenient option which will provide adequate performance for many applications, such as driving AOMs.

6.**I've assumed that the 100MHz reference reaching the RTM meets the phase noise spec up to >~100kHz**: if not, we'd probably want to put a 100MHz clean-up PLL on the DAC clock mezzanine, and route that to the LO mezzanine etc.

7.**The clock distribution ICs will not work for 10GHz LOs**: the fastest clock distribution ICs I'm aware of don't reach 10GHz. The ones with muxes are generally limited to <5GHz. My suggestion is that we should put mux + fan-out buffer ICs (e.g. ADCLK948) on the Sayma board to cover all LOs up to ~5GHz. Beyond this, I would suggest running coax from the LO Mezzanine to the analog mezzanine (that's about the only practical method at this frequency).

Thoughts?

[DAC clk + LO.pdf](https://github.com/m-labs/sayma/files/527897/DAC.clk.LO.pdf)

dhslichter, 2016-10-13:
Thanks for the thoughts, Tom.  I would argue that the point of the LO generation on the Sayma (versus using an external LO source piped in on SMAs to the analog mezzanines) is to allow standalone operation/reduce clutter for low-precision applications.  No matter how careful we are, I think an onboard LO generation system will never be as high performance as one would desire for e.g. high fidelity operations on hyperfine clock qubits, and that people will probably prefer to use a phase-locked DRO or other source to provide LO to the crate for those sorts of more demanding applications.  

With that in mind, simply providing 100 MHz to the analog mezzanines and leaving it up to the mezzanine designer to implement their desired LO on the mezzanine frees us from a great deal of work and hassle on the Sayma, and allows the mezzanine designer to set things up in just their desired manner.  Remember there is no restriction on double/triple/quad-width analog mezzanines, so for example it would be possible to design a larger mezzanine card with a multi-stage ultra-low-noise PLL system for ultra-low-noise >10 GHz LO generation (if that is desired by the user), which could be distributed to multiple I/Q modulators on the analog mezzanine card itself.   If mezzanine real estate is an issue, one could reduce the channel count slightly to free room.  

When talking about clock routing, I was thinking about an application with a really simple mezzanine containing an [HMC1197](http://www.analog.com/en/products/rf-microwave/iq-modulators-demodulators/iq-modulators/hmc1197.html) or [ADF5355](http://www.analog.com/en/products/rf-microwave/pll-synth/fractional-n-plls/adf5355.html) + IQ modulator, which would need a 100 MHz reference and could then provide reasonable-quality agile microwave signals for low-precision or medium-precision applications.  

Basically what I had thought about would be sending a cleaned 100 MHz LVPECL clock (or equivalent) to each of the mezzanine boards via an additional SMP pair.  This would then let the mezzanine designer make whatever necessary design choices to achieve the required LO performance.  

@hartytp @jbqubit @gkasprow @jordens thoughts/comments?

hartytp, 2016-10-13:
@dhslichter I agree that we should route the 100MHz analog clock to the analog mezzanines. 

You're right that one if one wants to generate the LO on the Sayma board, then the PLL could go on the (possibly quad-width) analog mezzanines(s). However, for the sake of modularity/reusability of hardware, I'd still rather have the PLL on a separate mezzanine. That way, for example, we could have two different versions of the LO PLL mezzanine -- a fancy one and a cheap one (e.g. your ADF5355) -- both of which could be used with the same (up-conversion) analog mezzanines. If we put the PLL/synth on the analog mezzanine, then we'd have to remake a lot more hardware if we ever wanted to alter the PLL.

In general, my philosophy is that I'd prefer to have more mezzanines per Sayma board, but with fewer varieties of each mezzanine to design/test/document/stock/maintain.

I do agree that we should have the option of generating the LO externally. My preferred solution for that would be to have a "passive" version of the LO mezzanine, which takes an external LO from a front-panel SMA, and distributes it to the analog mezzanines (via LVPECL buffers/multiplexers for LOs below ~5GHz, as in my diagram,  and via a passive splitter and internal SMAs for higher frequencies). This is essentially the same approach we're taking for the DAC clock. Again, this approach allows us to use the same analog mezzanine (IQ modulators + output stage) boards, regardless of whether the LO is generated internally or externally.

FWIW, having run the numbers for gate error versus phase noise, and looked at the phase noise of available microwave VOCs, I'm pretty optimistic that we can make a LO mezzanine with good enough phase noise for even our highest fidelity gates on 43Ca+ hyperfine clock qubits. Obviously, it gets a lot harder for higher frequency qubits.

jordens, 2016-10-13:
* Why do you want a clock MUX in front of the HMC7044? The latter seems to be able to do enough clock muxing already by itself.
* I don't know whether there is enough space for a sixth mezzanine. And I would lean towards making less allowances for arbitrary LO feeding and generation setups. I would either just leave the LO generation to the ADC/DAC analog mezzanine(s) and only feed them 100 MHz or expect the sampling PLL mezzanine to do it.

dhslichter, 2016-10-14:
I agree with @jordens here; adding an LO mezzanine, along with LO distribution/fanout, makes the Sayma cards much more complicated and requires us to overengineer at least some parts of the LO distribution system for future-proofing (which we still might screw up).  While the modularity argument is nice for motivating an LO mezzanine, I think it's better left to people designing the analog mezzanine cards, with the current front panel SMAs used for providing an external LO for simple upconverting mezzanines, which will work for most simple cases in the near term while people sort out what exactly they might want an analog mezzanine with integrated LO to do.

hartytp, 2016-10-16:
@jordens My point was really "we should have a way of bypassing the DAC clock mezzanine and feeding the 100MHz reference directly into the HMC7044". I'm not trying to prescribe exactly how this muxing should be done, and have no objection to using a max inside the HMC7044.

hartytp, 2016-10-16:
@jordens @dhslichter My concern with putting the LO on the analog mezzanine is that I don't want to have to re-make my carefully-designed LO PLL if I want to make a slight change to part of my input/output stage, such as a noise eater, (or, have to re-make the input/output stage if I want to alter my PLL).

> adding an LO mezzanine, along with LO distribution/fanout, makes the Sayma cards much more complicated and requires us to overengineer at least some parts of the LO distribution system for future-proofing (which we still might screw up)

I don't think the additional complexity is **that** bad: we're only talking about going from 5 to 6 mezzanines, and adding one extra fanout/mux IC for LO distribution. Still, I agree that if there isn't room for an extra LO mezzanine, then it's not an option.

> I would either just leave the LO generation to the ADC/DAC analog mezzanine(s) and only feed them 100 MHz or expect the sampling PLL mezzanine to do it.

If we don't have a separate LO mezzanine, then putting the LO on the DAC clock/sampling PLL mezzanine would be my preferred option for LOs at <~3GHz. We'd still want a mux to allow the option of feeding the 100MHz to the analog mezzanines, but we can incorporate that into the mezzanine if you'd prefer. We'd also need to make sure that the clock mezzanine is large enough to accommodate both PLLs.

gkasprow, 2016-10-16:
Take into account that some day we may want to use RF backplane which can
not only distribute 100MHz but also 2 RF signals in GHz range. That's why I
wanted to place the LO generation in some other place.
But in such case we may always re-engineer the Sayma RTM to fit that
purpose.
In case of 100MHz mux, in order to not deteriorate the signal, we can use
just capacitor or resistor jumpers.

hartytp, 2016-10-16:
> Take into account that some day we may want to use RF backplane which can
> not only distribute 100MHz but also 2 RF signals in GHz range. That's why I
> wanted to place the LO generation in some other place.

Okay, that makes sense. FWIW, I'm generally in favour of generating frequencies locally where conveniently possible for maximum flexibility. e.g. I can imagine a few cases where it would be desirable to have more than a single LO frequency in the same rack. 

> In case of 100MHz mux, in order to not deteriorate the signal, we can use
> just capacitor or resistor jumpers.

That could work, but it's much more convenient to have a software-programmable switch! For frequencies below ~5GHz, these are readily available with extremely low noise, that's unlikely to degrade even a high-quality LO. If we leave the muxing to the clock board, then users with LOs < 5GHz can build a clock board with an active switch, and users with higher-frequency LOs can go for a different approach.

gkasprow, 2016-10-16:
The signal degradation caused by such switch is one thing, another is signal leaking from second, not selected input.

This can seriously  affect our reference signal.

And this is a reason where we use passive jumpers to select clock source.

If you know good LVPECL or RF swtiches with high isolation, let me know, at least we will try them. In case of troubles we can always remove serial caps.

[40 more replies]

closed 2016-10-28 by dhslichter