sinara-hw/sinara #11: thermal design

sbourdeauducq, 2016-10-02:
Do the DACs need to be actively cooled or even temperature-stabilized? If so, how?

gkasprow, 2016-10-02:
We had a discussion in Warsaw about it.
Certainly the DACs will be cooled by MTCA fans and some crate-wide temperature regulation is possible.
The idea is to place additional 24bit ADC connected to the DAC (or mezzanine) outputs so on-line compensation will be performed. We will compensate for gain and offset changes.
The ADC will be connected to dedicated pins of mezzanine connector so the DAC driver offset can be compensated as well.
We will use  AD7194 for this purpose with external ppm reference.

jbqubit, 2016-10-04:
I think it's reasonable to assume temperature of air supplied to microTCA chassis is stable to +/- 2 C. 

Also from discussion in Warsaw... I'll close #13. 
- [ ] is it possible to achieve 1 LSB stability for both ADC and DAC? 
    - [ ] is 1 LSB accuracy needed board-to-board? 
- [ ] regarding on-board DC calibration
    - use heavy low-pass; sigma-delta DAC on motherboard, 16 single-ended inputs
    - SPI interface
    - [ ] how many channels are needed? (temperature, voltage measurements)
    - [ ]  allocate several pins (how many?) on 40-pin header #40 for routing to calibration ADC @gkasprow 
- [ ] analog component temperature coefficients (DACs, ADCs, amplifiers, attenuators) 
- [ ] are voltage references needed? 
- [ ] power supply noise rejection of components -- should this go on another Issue?

jordens, 2016-10-04:
@jbqubit please don't be sloppy and keep the top comment up to date. Move text if necessary.

jbqubit, 2016-10-04:
OK. I didn't know that was OK. Edited, improved.

-------
Joe Britton
Sensors and Electron Devices
Army Research Lab
2800 Powder Mill Rd
Adelphi, MD 20783
301-394-3130
joseph.w.britton5.civ@mail.mil

jordens, 2016-10-07:
@jbqubit @gkasprow could you then mark those tasks as done or reraise them where approriate?
Also, who should resolve/answer the tasks that need handling?

dhslichter, 2016-10-10:
Temperature monitoring can also be performed on mezzanines by using [ADT7420](http://www.analog.com/en/products/analog-to-digital-converters/integrated-special-purpose-converters/integrated-temperature-sensors/adt7420.html), which contains conversion and I2C hardware, thus freeing slow ADC lines for e.g. monitoring of I/Q modulator drift or amplifier gain drift.

closed 2017-02-16 by jbqubit