sinara-hw/sinara #12: analog mezzanine power and IO

jbqubit, 2016-10-04:
In #8 Greg recommends 40 pin headers.

>For digital IO and control I'd like to use Samtec hermaphroditic, 
>40 pins connectors: LSS-120-01-L-DV-A and LSS-120-02-L-DV-A 
>which combined give exactly 9mm. I use them in other projects 
>and they work well with frequent mating.

From discussion in Warsaw.... 
- digital signal lines
    - pins [1,...,16] are digital I/O
    - if daughter uses SPI
    - pins [1, 2, 3, 4,...,16] are [SCLK, MOSI, MISO, CS0,...,CSn], 50 MHz maximum
    - all of the lines should be the same style... no pull-up resistors
    - all digital I/O are +3.3V to/from FPGA
- power
    - pins [17, 18, 19, 20] are [+12V, - 12V, +5V, +3.3V]
    - pins {21, 22, 23, 24} are power ground

- review likely power supply requirements.... here are ICs identified so far
- HMC1197 :: mixer and VCO :: +3.3V (320 mA) and +5V (48 mA) 
- HMC830 or HMC833:: fractional PLL with VCO :: +3.3V (320 mA) 
- ERA-4XSM+ :: Broadband amplifier, DC-4 GHz, ultra low phase noise :: +5V (85 mA) 
- HMC542BLP4E :: digital attenuator :: +5V (3 mA) 
- CSWA2-63DR+ :: RF switch :: +5V (74 mA) 
- JSW2-63DR+ :: RF switch :: +5V (37 uA)
- ADL5375 IQ :: modulator :: +5V (200 mA)

- questions
    - [ ] What is max current for each power rail? 
    - [ ] What is noise spec for analog power rails?

jordens, 2016-10-05:
* The `IO*` numbering seems to be a bit weird but I'll attribute that to the connector.
* Do we (can we) ship (4+4)*16 digital lines over from the AMC?

gkasprow, 2016-10-05:
It's just rough idea, we can change it.
It will be difficult for all IOs, but we can use extenders for some of them
in case of troubles

dhslichter, 2016-10-06:
- ERA-4XSM amplifiers require a low-noise supply of at least ~7V; the bias current is set with an external high-power resistor.  This supply needs to be as low-noise as possible because supply fluctuations appear directly on the output, with very little supply rejection.  See datasheet, page 3: https://www.minicircuits.com/pdfs/ERA-4XSM+.pdf .  Ideally one would use an ultralow-noise LDO from +12V to generate the supply voltage.  This means we do need to be a bit careful about noise on +12V rail, in particular the frequency content of the noise.
- Typical low-speed (~ few-ish MHz) analog inputs will likely want to use instrumentation amplifiers where the input signal range is +/- 10V, thus supply voltages of +/-12V would be required.  Tom Harty will have strong feelings on this count.  These supply voltages will need to be low noise as well.  

Given these considerations, I think it makes sense to generate a reasonably low-noise +/- 12V on the RTM (heavily filtered switching regulator probably OK) and pass both rails to the daughtercards.  Since we have +/-7 V already, this would only be used for amps requiring the higher voltage.  I worry about putting switching regulators on the analog daughtercards to generate +/-12V as needed from +/- 7V, both in terms of daughtercard real estate and signal integrity due to interference from switching regulators.  

What are the 4 ADC channels on the schematic?  

@jordens the original notion for the digital lines to the analog daughtercards would be to use SERDES chips to convert slow parallel digital busses on AMC and RTM to high-speed differential to send through the AMC/RTM connector.  We did the math and figured it the connector would support it; @jbqubit has the details, I can't find them at present.

dhslichter, 2016-10-06:
If bidirectionality is desired for some DIO pins (this may or may not be the case), we can use additional serdes chip pairs (e.g. http://www.ti.com/product/DS92LV3241) to transmit input data to the AMC card via RTM.  My feeling is that we are unlikely to need bidirectionality for all channels; it might suffice to have e.g. 4 input-capable channels per daughtercard rather than 16.

dhslichter, 2016-10-06:
Also there appear to be 17 DIO on the schematic shown, not 16?

gkasprow, 2016-10-06:
The idea was to have the payload 12V for conversion purposes for users who
want +/-10 or +/-12V.
The +/-7VA can be used directly, after low pass filter or LDO to supply low
noise stuff.
The problem with +/-12V and required current consumption is the power
budget.
Let's assume we want to have +/- 12V at 1A, this already gives 24W,
multiplied by 4 mezzanines + add power conversion efficiency and you will
easily end up with >100W per RTM. This is a lot of power
From our experience with RF stuff, I can say that +/-7 is the most
universal approach.
IF you want +/-12V, just take existing 12V and invert it with capacitive
charge pump that generates very little noise. You can do it on board.

The ADC channels are for slow 24bit ADC for calibration purposes. It was
discussed in another issue.

dhslichter, 2016-10-06:
Where are you getting this 1A number for +/- 12V?  If the mezzanines are only using +/- 12V for special amp stages (and not just as a general power rail to be used to make +5 and +3.3 for the modulators/attenuators/etc), then you're talking about 2x ERA-4XSM @ 85 mA each (for example), or considerably less for instrumentation amplifier input stages.  That's more like 2W per mezzanine (8 W per RTM), much less than the 100W you are talking about.  

My points really are:
- the +12V rail needs to be low noise -- it can't be some garbage switching supply rail
- it would be nice to have a -12V rail supplied by the RTM to reduce the amount of power supply generation required on the daughtercard.  I recognize that one can do a charge pump circuit onboard to invert the voltage, but since we have extra pins on the DC connector, why not do it on the Sayma and simplify/cut noise on the daughtercard?
- the power dissipation argument is irrelevant if we also have the +/- 7VDC rails; if there is a component on the daughtercard that needs X amps at Y volts, that power is going to be dissipated no matter how you supply it, all we want to do is ensure that as little EXTRA power is dissipated by voltage regulators as possible.  As long as you are only running things that really need +12V or -12V off the +/-12V rails, it won't many any difference to your power dissipation.
- neither of these supply rails need to put out much current -- 200 mA per daughtercard would be more than enough for +12V, and the -12V rail could be ~50 mA per daughtercard.  At those ratings we're only talking 12W dissipation total.

Ack on the 24 bit ADC, I forgot about that.

dhslichter, 2016-10-06:
Related to my comment on #9, I think we should pick two pins **in addition** to the 16 pins of GPIO shown here to be dedicated to I2C (SDA/SCL).  I think we should leave the 16 GPIO pins as bidirectional fast (50 MHz) 3.3V LVCMOS lines.  We can make a recommendation for use for which pins to use for SPI, as a convention, if an SPI bus is desired.

gkasprow, 2016-10-07:
This situation would be with initial idea of just +/-12V and 5V supply where low noise 5V or 7V would be generated from +12V. If we implement both +/- 12V (100mA), +/- 7V  and 3.3V this issue then is solved

jbqubit, 2016-10-07:
[image]

dhslichter, 2016-10-07:
Suggested LDO for use on mezzanine cards:
http://www.analog.com/media/en/technical-documentation/data-sheets/ADM7150.pdf

jbqubit, 2016-10-10:
I've summarized #12 at the top of #8. If there's no more discussion about power and pin assignments lets close this.

dhslichter, 2016-10-10:
Closing this issue, all discussion related to specifics of power/IO connectors and the signals they carry to be in #8.

closed 2016-10-10 by dhslichter