sinara-hw/sinara #16: copyright

jordens, 2016-10-06:
How should copyright be handled?

a) leave with the authors of the contributions
b) assign to UMD or ARL or similar
c) assign to M-Labs or WUT/Creotech or similar

b) and c) would need some CLA.

@jbqubit @jmizrahi

jbqubit, 2016-12-12:
This will be easiest if a single entity holds copyright. M-Labs seems like a good choice. Or Greg.

jmizrahi, 2016-12-12:
Shouldn't copyright be held by whoever funded the development? That would be ARL for Metlino, Duke for Sayma.

jbqubit, 2016-12-12:
Works published by Federal employees are in the public domain.

https://en.wikipedia.org/wiki/Copyright_status_of_work_by_the_U.S._government

Many things are easier if a single entity holds copyright. For example,
choosing a new license. I recommend assigning WUT or M-Labs copyright.

-------
Joe Britton
Sensors and Electron Devices
Army Research Lab
2800 Powder Mill Rd
Adelphi, MD 20783
301-394-3130
joseph.w.britton5.civ@mail.mil

sbourdeauducq, 2016-12-13:
> I recommend assigning WUT or M-Labs copyright.

I'd consider Creotech/Technosystems instead of WUT, as I fear there could be red tape associated with WUT. Otherwise M-Labs is fine too.

gkasprow, 2016-12-14:
At the WUT nobody cares what logo is on PCBs, so I can do whatever I feel
is appropriate.
We can put  logos of WUT, M-Labs and ARTIQ

sbourdeauducq, 2016-12-15:
Let's have only the ARTIQ logo. NB: this issue is about copyright which is a different question; logo is discussed in #123 and #15.

hartytp, 2017-02-19:
Is this resolved now?

jordens, 2017-02-19:
At least I don't know what the resolution/decision is/was.

gkasprow, 2017-02-19:
On  boards that went to production I placed Artiq logo and simple statement
Board name rev1, 2017, WUT ISE

hartytp, 2017-02-19:
Does anything else need to be done on this, or can we close the issue?

gkasprow, 2017-02-19:
Just close it.

sbourdeauducq, 2017-02-20:
What is the resolution?

hartytp, 2017-02-21:
Can we go for copyright WUT? 

Decision @jbqubit an other people who paid for this, please...

jasonamini, 2017-02-21:
How does the copyright and CERN OHL interact?  Is the copyright holder the "Licensor"?

jordens, 2017-02-21:
@jasonamini Greg and colleagues, and everyone who contributed are copyright holders right now. They could/can reassign/transfer their copyrights to someone else. The copyright holder can choose the license. This would become relevant if e.g. someone wants to license this work under conditions where they would not have to follow paragraphs 3.3 and 4.1 of the CERN OHL v1.2.

klickverbot, 2017-02-21:
> They could/can reassign/transfer their copyrights to someone else.

I'm sure you are aware of this, but for the benefit of others: Copyright assignment is easy under US law, but rather murky territory in Europe – in Austria and Germany (and possibly others), copyright ("Urheberrecht") can't be transferred to other people except with the passing of the creator.

Of course, creators can still grant others the irrevocable right to "do whatever they want" with their creation, but a blanket copyright assignment wouldn't be effectual. Using an appropriately worded CLA instead avoids such issues, and suitable templates exist.

jordens, 2017-02-21:
Correct. Thanks for elaborating on this.

gkasprow, 2017-02-21:
CERN delivers the license that can be freely used. Yes, here is how OHL specifies it

“Licensee” means any natural or legal person exercising rights under this Licence.

“Licensor” means any natural or legal person that creates or modifies

Documentation and subsequently communicates to the public and/ or distributes

the resulting Documentation under the terms and conditions of this Licence.

gkasprow, 2017-02-21:
It’s the same in Poland. The creator holds the copyright forever. It cannot transfer it to someone else but can grant the right to use their creation.

hartytp, 2017-02-21:
So, ... what needs to be done to resolve this issue?

gkasprow, 2017-02-21:
IMHO just leave as it is. We place ARTIQ logo, we can also add some link as
well and short info about institution who builds stuff. I don't have any
objections of putting founding institutions logo providing that I get it in
vector form:)
<https://github.com/

hartytp, 2017-02-21:
@gkasprow Sounds good to me.

If anyone wants a logo or text on the board/schematics/Wiki then post an issue.

closed 2017-02-21 by hartytp