sinara-hw/sinara #17: sayma development roadmap for 2017

jbqubit, 2016-10-06:
The Sayma gateware/software development for 2017 needs to be sorted. Here's a breakdown of what's in the ARL and Oxford procurement pipelines. If there are additions or modifications to Task Groups A or C, Britton needs to know by 10/10. 

# Task Group A
Tasks are needed for support of many-qubit applications including LogiQ. ARL is funding these items. 
* **17A1**	High-resolution (nanosecond-level) ARTIQ TTLs for Kintex Ultrascale type IOSERDES
* **17A2**	Port single-tone non-interpolating RTIO parallelized DDS (fDATA) core to Kintex Ultrascale
* **17A3**	Absolute and relative phase modes for all frequency and phase interpolators independently configurable through RTIO
* **17A4**	f,p,a,u modulation by RTIO (eDRTIO) for all interpolators
* **17A5**	DMA support (redirect RTIO events emitted in named DMA context to memory pool, then (re-)play by name)
* **17A6**	Parallelizing CIC interpolator/AA filter before parallelized DDS core
* **17A7**	Parallelized FIR filter after CIC, research and implement what’s possible given DSP resources
* **17A8**	Coherent phase mode (implemented in kernel API and runtime) for frequency/phase updates of the three eDRTIO channels
* **17A9** Distributed DMA (for local DRAM on Sayma)
* **17A10** Continued project management for Sayma and Metlino testing and integration
* **17A11** Gate-able counter TTL input RTIO channel

# Task Group B
Full implementation of DRTIO is needed for communication with Sayma hardware both in uTCA chassis and over optical fiber. Oxford is funding these items. 
* **17B1** Implement DRTIO: extend the demo implementation (https://github.com/m-labs/drtio_transceiver_test) into a complete solution, including output event injection / input event monitoring. The plan is for this functionality to be demonstrated on the Metlino / Sayma hardware.
* **17B2** Implement pulse-sequence DMA from the master system. 

# Task Group C
Develop debug tools, support external modulation and increase Sayma dynamic range (in time domain). 
* **17C1**	Time stretcher for interpolators to extend dynamic range beyond spline knot durations of 2**16 cycles
* **17C1**	Saturating summing junctions
* **17C2** 	f,p,a,u modulation by external data stream (ePID) for all interpolators
* **17C3**	Summing junction clipping detectors and their remote readout
* **17C4**	Configurable, saturating output limiters
	
* **17C5**	Monitoring support for all spline interpolators and total DAC output
* **17C6**	Injection/override support for all spline interpolators
* **17C7**	Distributed hardware logic analyzer support for all interpolators and DAC output

# Task Group D
Task Group D contains items not yet precisely specified and need hashing out as stand-alone github Issues. A subset of these tasks could be accomplished in 2017. Should there be other things on this list? Scope of some of these extends beyond Sayma.
* Inagural ARTIQ users group meeting
* Hardware development supporting time-to-digital conversion (TDC)
with 10 ps resolution. Focus is extending ARTIQ support to solid state qubits. 
* Gateware and software development supporting 17D2. Aim to support in-loop feedback on timescales of << 100 ns. 
## related to ADC
* JESD204B support for AD9656
* burst acquisition: sample for 10 us; data handling: a) send data to core device, b) send data to Master by RPC (DMA?)
* spectral analysis: sample for an interval, compute FFT; data handling: a) send to core device real/imag at K specific frequencies, b) send to core device K highest amplitude peaks (real/imag) 
* ePID: drawing from ideas in D. Leibrandt http://arxiv.org/abs/1508.06319. Sort sub-features by a) inter-dependency b) priority 
* floating point support for soft core uP
* extending ARTIQ documentation: tutorial on modifying/extending ARTIO using DRTIO

jordens, 2016-10-07:
@jbqubit  Could you triage this issue (labels, milestones, somebody assigned, state what you would like to see done right now right here)?
Aren't large parts of this redundant because already written down in #1 #3 #2 #6?

jordens, 2016-10-07:
And obviously feel free to reassign.

jbqubit, 2016-10-10:
I'm working on writing up Task Groups A and C. I'll revisit Task Group D in coming weeks.

jordens, 2016-10-11:
Don't use the issue tracker for note-taking, planning, and brainstorming. Unless you have something that can actually be answered/done/resolved/fixed, this is not the right place.

closed 2017-02-21 by jbqubit