sinara-hw/sinara #18: fpga speed grade

jordens, 2016-10-07:
Kintex Ultrascale -1 or -2 choice for price.

The only relevant difference that I can see right now is: (http://www.xilinx.com/support/documentation/data_sheets/ds892-kintex-ultrascale-data-sheet.pdf)

* CPLL on -1 is 8.5 G max. That's not a problem for the 16 DAC TX GTH, but for the 4 SFP/SATA/AMC GTH TX they would need to be driven from the same QPLL if we want to run them faster than 8.5 G.

@sbourdeauducq ok to choose -1 speed grade for price?

sbourdeauducq, 2016-10-09:
Ok.

closed 2016-10-09 by jordens

jbqubit, 2016-10-10:
@gkasprow please update when you know the price from Xilinx.

jordens, 2016-10-10:
@jbqubit update what?
AFAICS This is decided.

jbqubit, 2016-10-10:
Agreed that this is decided to use -1 speed grade. @gkasprow is checking on price per chip which I hope he will post here.

jordens, 2016-10-14:
XCKU040-1FFVA1156C