sinara-hw/sinara #19: github handles

jbqubit, 2016-10-07:
Who is involved in Sayma development? 
* Name, Email, GitHub ID

# m-labs
* Robert Jordens, jordens@gmail.com, jordens
* Sebastien Bourdeauducq, sebastien.bourdeauducq@gmail.com, sbourdeauducq

# NIST Boulder
* Daniel Slichter, dslichter@gmail.com, dhslichter

# UMD
* Jason Amini, amini@umd.edu, 
* Johnathan Mizrahi, 
* Joe Britton, joe.britton.arl@gmail.com, jbqubit

# Warsaw
* team lead is Grzegorz Kasprowicz, kasprowg@gmail.com, gkasprow
* analog/RF lead is Samer BouHabib, samerbouhabib@gmail.com, ___________
* FPGA lead is ___, ___, ____
* project graduate student is Michał Gąska, michalgaska2@gmail.com, _______

jbqubit, 2016-10-07:
@jordens How do I add users to the project? Jason Amini's handle is jasonamini.

jordens, 2016-10-07:
We'do that. But the others need to be contacted. Do you do that?

jbqubit, 2016-10-10:
@jordens please add cjbe and weidazh and jasonamini

jbqubit, 2016-10-11:
@Jordens please add @hartytp

michallgaska, 2016-10-11:
Hello everyone!

jordens, 2016-10-11:
@jbqubit all(ready) added. It seems redundant to update this issue.
Everyone: you need to accept that invitation link.
Everybody else, just ping me.

jordens, 2016-10-11:
@michallgaska hello, https://github.com/m-labs/sayma/invitations

dtcallcock, 2016-10-14:
@jordens can you invite me please (@dtcallcock)

jordens, 2016-10-14:
@dtcallcock sure. done

jbqubit, 2016-12-02:
@gkasprow fill in blank
- documentation and schematic review, NAME, EMAIL, @filipswit

@jordens invite @filipswit

jordens, 2016-12-04:
@jbqubit people should just fill in their metadata in their profiles instead of duplicating it here where it will rot.

jbqubit, 2016-12-04:
@jordens I'm not aware of a github mechanism that aggregates handles, emails, locale and project roles in a single location. Please tolerate the present use of #19 for purposes of the Sinara project.

jbqubit, 2016-12-04:
@filipswit is now added. Issue closed. 

@filipswit welcome to the project!

gkasprow, 2016-12-17:
@sbourdeauducq please change email address of Pawel Kulik to P.P.Kulik@stu.elka.pw.edu.pl

jbqubit, 2017-02-03:
@gkasprow please add Creotech

gkasprow, 2017-02-13:
@sbourdeauducq please add Creotech guys

sbourdeauducq, 2017-02-16:
done

sbourdeauducq, 2017-02-16:
Also let's try to reduce the abuse of the GitHub issue tracker. https://github.com/m-labs/sinara/wiki/Team

closed 2017-02-16 by sbourdeauducq