sinara-hw/sinara #20: Hardware design tasks

jordens, 2016-10-10:
- [ ] schematics
- [ ] initial PCB placement and mechanical details
- [ ] schematics review
- [ ] simulation of PDN
- [ ] SI simulation of critical links
- [ ] RF simulation of analog chain

- [ ] layout of digital part of Sayma
- [ ] layout of RTM part of Sayma

- [ ] PDN simulation of layout (AC & DC)
- [ ] DDRs simulation of layout
- [ ] SI simulation of critical links
- [ ] EMC simulation of whole board
- [ ] PCB proto delvered

- [ ] DC levels
- [ ] Functionality – configuration of FPGA, access to all resources (I2C, SPI)
- [ ] Clock distribution
- [ ] SDRAM operation with Xilinx reference design
- [ ] JESD links operation with Xilinx reference design

Feel free to 
1) break out into smaller issues and reference them here.
2) reassign smaller issues
3) indicate which items/questions are urgent.

dhslichter, 2016-10-14:
I would break out schematic/layout tasks to include analog mezzanines as well.

jbqubit, 2017-01-26:
Closing since nobody used this Issue.

closed 2017-01-26 by jbqubit