sinara-hw/sinara #21: distribution of prototype hardware

jbqubit, 2016-10-10:
The hardware development contract funds two rounds of prototype hardware. Each prototype round will generate two copies of the prototype hardware consisting of the following parts. The list reflects the new organization of the Sayma analog front end with 4 mezzanine boards, each supporting 2 DAC channels and 2 ADC channels. 

**Metlino-Sayma prototype system**
1 metlino_motherboard
1 metlino_clk
2 sayma_motherboard
2 sayma_clk
8 sayma_analog_mezzanine 

Full Metlino-Sayma prototype hardware will be distributed as follows.

* m-labs in Hong Kong
* UMD
* NIST (A)
* Oxford (A)

(A) The additional cost of two additional development systems will be paid directly (eg credit card) to the firm responsible for printing-stuffing the contract-funded prototype hardware (eg Creotech).

jordens, 2016-10-10:
@jbqubit It's unclear what the "issue" is and who is supposed to act and what should be done. If there is no issue, close this.

jbqubit, 2016-10-10:
Tasks:
* @gkasprow determine cost of full Melino-Sayma prototype system
* when the time comes, Oxford @weidazh @cjbe and NIST need to arrange to pay for their copies of the est tsystem
* @gkasprow identify additional part numbers needed for a full test system
    * microTCA crate with microTCA.4 backplane
    * AMC-side power supply
    * MCH Tongue 0 
    * RTM RF backplane
    * single-slot AMC adapter for stand-alone testing
    * passive RTM and AMC blanks (for power dissipation and RF shielding tests)

dhslichter, 2016-10-10:
Also should specify the types of mezzanine cards to be included, since there will be at least 2 designs (direct input/output and up/downconverting out/input), if not more.

jmizrahi, 2016-10-19:
@gkasprow could you provide a list of all the commercial components necessary to run the full system? I would like to go ahead and order all the necessary microTCA hardware.

gkasprow, 2016-10-19:
- MTCA crate: 11850-026
   http://schroff.pentair.com/wcsstore/AuroraStorefrontAssetStore/User%20Downloads/LiteratureRequests/Content_schroff_ssh_39601850_microtca.4_en.pdf
   So far the RF backplne is currently available only in this crate.
   - Smaller version of RF backplane will be available soon (according to
   my colleagues from WUT) for this crate
   http://schroff.pentair.com/en/schroff/MTCA/3074457345616797165.
   - Supply: 600W 11098-547 or 1000W NAT-PM-AC1000
   - There is version of the PSU without the power management:
   http://schroff.pentair.com/en/schroff/11098-416, but I never tested it.
   -  Filler module for empty AMC/RTM slots 21850-106
   http://schroff.pentair.com/en/na/21850-106
   -  Filler module for empty MCH/PSU slots 21850-107 http://sch but
   roff.pentair.com/en/schroff/21850-107
   - MCH: NAT-MCH -Base12 -CLK-PHYS -FP0D

jmizrahi, 2016-10-20:
@gkasprow Thanks! I assume the 1000W would be necessary for a fully populated full size crate?

jbqubit, 2016-10-24:
Notes from discussion on 10/21 follow. 

When operating Sayma ARM+RTM on benchtop use the following.
- communication with Core Device (MCH) 
    - AMC backplane Port 4 uses transceivers to communicate with Core Device 
    - a SATA connector on the AMC backplane is also connected to Port 4 (passive) 
    - in stand-alone configuration use a to SATA to SFP adapter to connect ARTIQ Master with board
    - no change in gateware is needed
- power with +12V ATX power supply (common in PCs)
    - [ ] @gkasprow is this the 4-pin connector in the upper right-hand corner? [link](http://www.ohwr.org/projects/afck/wiki)

Chassis:
- [ ] @gkasprow RTM RF backplane part number for 6-slot chassis
- [ ] @gkasprow RTM RF backplane part number for 12-slot chassis

gkasprow, 2016-10-25:
>
> When operating Sayma ARM+RTM on benchtop use the following.
>
>    - communication with Core Device (MCH)
>       - AMC backplane Port 4 uses transceivers to communicate with Core
>       Device
>       - a SATA connector on the AMC backplane is also connected to Port 4
>       (passive)
>       - in stand-alone configuration use a to SATA to SFP adapter to
>       connect ARTIQ Master with board
>       - no change in gateware is needed
>    - power with +12V ATX power supply (common in PCs)
>       - @gkasprow <https://github.com/gkasprow> is this the 4-pin
>       connector in the upper right-hand corner? link
>       <http://www.ohwr.org/projects/afck/wiki>
>
> Yes

> Chassis:
>
>    - @gkasprow <https://github.com/gkasprow> RTM RF backplane part number
>    for 6-slot chassis
>
> don't know the pn yet

>
>    - @gkasprow <https://github.com/gkasprow> RTM RF backplane part number
>    for 12-slot chassis
>
> NATIVE-LLRFR9BP
Low-Level-RF-Backplane (Add-On for NATIVE-R9): DESY design providing Zone 1
RF connections (between 12uRTMs and 3 eRTMs) as well as Zone 2 power
connections (between 12uRTMs and 2 rear PMs) and Zone 2 LVDS connections
(between 12uRTMs and NAT-MCH-RTM-FPGA).

Delivered as integrated part of NATIVE-R9.

List price is 1600 Euro.

gkasprow, 2016-10-25:
Let me finish the power estimation.

For 12 slots this may be necessary

jbqubit, 2016-10-26:
http://mtca.desy.de/sites/site_mtca/content/e172206/e205636/e212584/e248086/uRFB_concept_Datasheet_19.12.2014_eng.pdf

jmizrahi, 2016-11-13:
I received [this quote](https://github.com/m-labs/sinara/files/587578/NATquote2016_11_12.pdf) from NAT for the microTCA chassis hardware. Is this the same configuration that everyone else intends to use with these boards? Do the rest of you already have a microTCA chassis with the necessary MCH and RF backplane? They quoted a 6-8 week leadtime.  @jbqubit @jordens @dhslichter @hartytp 

@gkasprow they added NAT-MCH-M4 as an alternative, but this is unnecessary, correct?
Can you confirm that what they quoted me is the correct hardware set and the minimal hardware set (as the RF backplane does not appear to be available for smaller chassis's?)  Given the price tag, I'd like to be 100% sure that I don't accidentally purchase incompatible hardware.

hartytp, 2016-12-03:
@jbqubit Our process for approving new suppliers is a bit torturous, so I'd like to start adding the supplier (Creotech?) for the Sayma/Metliono hardware now. Who is the contact for this?

gkasprow, 2016-12-03:
The production of prototype series will be done by Technosystem (http://technosystem.pl/#)
At the moment it cannot be at Creotech due to formal restriction at the WUT (conflict of interests).

hartytp, 2016-12-03:
@gkasprow When we order hardware, will we place the order with Technosystem or WUT?

gkasprow, 2016-12-03:
You can do it via WUT but then 30% charge will be applied. So it depends on
you:)
WUT will anyway subcontract HW production to Technosystem

hartytp, 2016-12-03:
Okay, noted. Is there anyone in particular I should contact at Technosystem about this?

hartytp, 2016-12-05:
@jbqubit @gkasprow 

- What is the estimated cost of each of the pcbs we're designing? (PCB_metlino, PCB_sayma_AMC, PCB_sayma_RTM, PCB_mezzanine_analog_allaki, PCB_mezzanine_clock_template, PCB_baikal)
- Assuming that there's nothing major wrong with any of the PCBs we design (say, just a few minor noise issues) when do you estimate that a second design revision and/or hardware purchasing round will happen?

These factors will influence how much hardware we buy in this round.

gkasprow, 2016-12-06:
I got initial offers from the assembly house:
Sayma AMC for 2 pieces: 4k$/piece. Metlino price would be very similar.
[image]
Sayma AMC for 10 pieces: 3.3k$/piece
[image]

Sayma RTM for 2 pieces: 3k $/piece
[image]

Sayma RTM for 10 pieces: 2.36k$/piece
[image]

hartytp, 2016-12-09:
Minimal parts list required to get the Metlino/Sayma system working in a uTCA.4 rack, all from NAT:

| NAT PN        | description |comment           |
| ------------- |:-------------:|:-------------:|
| NATIVE-R9     | Rack | This is the only rack which currently supports a RF backplane. RF BPs are planned for more racks in future (work not funded/started yet). |
| NAT-PM-AC1000 | 1kW PSU |   Cheaper, lower power-rating PSUs are also available, with similar specs. However, the price difference isn't that great. |
| NATIVE-LLRF | RF BP  | Needed to power RTMs and optionally used to distribute clocks. |
| NAT-MCH-Base12-GbE | MCH | @gkasprow  why is this needed with Metlino? |
| NAT-MCH-FP0D | Double width front-panel for NAT-MCH-Base12-GbE | |
| NAMC-Filler-FS-FP1D | Double-width AMC filler + front-panel | Needed for proper air flow in the rack. @gkasprow we need 1 of these for the unused MCH slot, right? |
| NAMC-Filler-FS-RP1D | Double-width RTM filler + front-panel (for unused MCH + PMs) | @gkasprow how many do we need? |
| NAMC-Filler-MS-FP1D | Single-width AMC filler + front-panel | 12 needed for an empty rack |
| NAMC-Filler-MS-RP1D | Single-width RTM filler + front-panel | 12 needed for empty rack |

12 NAMC-MS-FP1D come for free with the rack.

This assumes that @gkasprow will make a simple tongue 2 board to replace the NAT-MCH-CLK-PHYS part (which we would otherwise use as a ~1k USD spacer).

NB on @jmizrahi quote above: 

- NAMC-Splitting-Kit is mounting hardware for NAT MCH, which is not compatible with Metlino. Not required.
- NAT-MCH-M4 not needed/wanted in this system

@gkasprow please can you confirm that the above is correct?

hartytp, 2016-12-09:
@jbqubit you beat me to it!

hartytp, 2016-12-15:
@gkasprow Thank you for doing that!

gkasprow, 2016-12-15:
Confirmed that:
- NAMC-Splitting-Kit is mounting hardware for NAT MCH, which is not compatible with Metlino. Not required.
- NAT-MCH-M4 not needed/wanted in this system

jbqubit, 2016-12-15:
Some refinements of @hartytp list.
- WUT provides a front panel for the MCH slot which consists of a stack of the following boards 
    1. ARTIQ_EE/PCB_Metlino  (MCH Tongues 3,4)
    2. ARTIQ_ALTIUM/PCB_MCH_mordovia (MCH Tongue 2)
    3. NAT-MCH-Base12-GbE (MCH Tongue 1)
    - ->> so scratch NAT-MCH-FP0D
- PCB_MCH_mordovia is the name of the simple tongue 2 board that replaces NAT-MCH-CLK-PHYS part

[image]

gkasprow, 2016-12-15:
There is an issue with front panels.

They are manufactured (milling, printing) by Elma who has 8..12 weeks lead time. Schroff has similar lead time. Last time I waited even more because they had some material shortages.

So there is very little chance that they will be delivered with first revision of the PCBs.

I couldn’t design and order them before layout was finished. Just a few days ago additional connectors were added to Sayma.

jbqubit, 2016-12-16:
Lack of front panel isn't a problem for v0.1 testing. Please ship PCB's with appropriately sized standoffs and screws to hold them together.

sbouhabib, 2016-12-18:
@gkasprow i think there was a company that did them in 4 weeks, they are not nice ones with emblems and stuff just the metal, Ill try to check tomorrow

dhslichter, 2016-12-19:
What about Front Panel Express/Schaeffer AG?  They'll turn custom panels around in a week, with holes, legend, etc.

gkasprow, 2016-12-19:
This is not standard front panel cut in a piece of aluminium sheet.
It is formed bezel in which holes and slots are machined.

>

jmizrahi, 2016-12-21:
@gkasprow if blank front panels are a stock part, and the long lead time is for machining and printing, why couldn't we buy the blank ones and send them elsewhere for the machining and printing?

jbqubit, 2017-01-16:
@jasonamini pointed out that there are some additional parts users should buy to prepare for testing/using a Sinara system with DRTIO (SFP fiber-based). I've tested these with ARTIQ DRTIOlite (circa summer 2016). 

- managed Ethernet switch with 8 RJ-45 copper ports and 2 SFP ports: Netgear ProSafe GS110TP 
- 1000BASE-BX-D SFP 1490nm-TX/1310nm-RX 10km DOM Transceiver: ZyXEL SFP-BX1490-10-D; ordered from fs.com #39460, qty 6, $12/ea (*)
- 1000BASE-BX-U SFP 1310nm-TX/1490nm-RX 10km DOM Transceiver: ZyXEL ZyXEL SFP-BX1310-10-D; ordered from fs.com #39464, qty 6, $9/ea (*)
- 9/125 OS2 Duplex Single-Mode Fiber Optic Patch Cable - PVC/LSZH/OFNP;  ordered from fs.com
```
    Fiber Type - G.652D     
    Cable Jacket - PVC(OFNR)     
    Cable Diameter - 0.9mm   
    Polish A - UPC    
    Connector A - LC    
    Polish B - UPC    
    Connector B - LC    
    2m    
    #12189 qty 6, $3.25/ea
```

(*) Unfortunately, fs.com no longer stocks the ZyXEL parts and it looks like they're in short supply from other distributors. DRTIO relies on use of bi-chromatic SFP transceivers with the same constraints as White Rabbit. See ohwr.org for a list of White Rabbit compatible SFP modules. @sbourdeauducq What transceivers are you using for DRTIO testing? 

@jasonamini, @jmizrahi I have plenty of ZyXEL transceivers and patch cables so don't ordey any for UMD.  Do order your own NetGear switch.

sbourdeauducq, 2017-01-17:
I'm currently using a pair of generic 1000BASE-BX SFPs from fs.com (1490nm/1310nm) and they do what they are supposed to do, but they are limited to 1G. We plan to have a higher data rate for the phaser-based designs, so I have also tried similar 10G SFPs but they are acting up for yet unknown reasons. Our next step is to get the DRTIO phaser to work with a passive SFP copper cable (http://www.fs.com/products/36648.html) between two KC705s, then figure out what the heck is wrong with the 10G SFPs.

jbqubit, 2017-01-17:
Thank you for the update @sbourdeauducq. I got an email from fs.com indicating what generic SFP is supposed to be compatible with the ZyXEL SFP.  

```
#37930	Customized 1000BASE-BX BiDi SFP 1310nm-TX/1490nm-RX 10km Transceiver
Compatible Brands:ZyXEL	US$8.82	1294 pcs in stock

#37940	Customized 1000BASE-BX BiDi SFP 1490nm-TX/1310nm-RX 10km Transceiver
Compatible Brands:ZyXEL	US$11.76	814 pcs in stock
```

sbourdeauducq, 2017-01-17:
AFAIK this "compatibility" is mostly EEPROM programming, which doesn't matter in our case since ARTIQ does not check that a SFP of an approved brand is connected and completely ignores that EEPROM.

jbqubit, 2017-02-21:
I'm closing this Issue as this looks resolved.

closed 2017-02-21 by jbqubit