sinara-hw/sinara #22: Sayma RTM Atrix7 local interface

jbqubit, 2016-10-10:
Connections between Atrix7 and parts on Sayma RTM <<< THIS IS A DRAFT >>> 

* Atrix7 FPGA on Sayma RTM provides low-speed local IO for ICs on RTM
* p/n Xilinx XC7A15T
* local connect
   * 2 of AD9154 require
        * 6 single-ended for SPI (CLK/MOSI/MISO)
        * 4 single-ended for SPI chip select
    * 2 of AD9656 require 
        * 6 single-ended for SPI (CLK/MOSI/MISO)
        * 4 single-ended for SPI chip select
    * 1 of Si5324 requires
        * [ ] connected "as usual" (what's this?) 
        * one output to the IBUFDS_GTE2
        * [ ] one output to a clock-capable general-purpose input
    * 4 of analog mezzanine 
        * [ ] link to analog mezzanine interface

sbourdeauducq, 2016-10-10:
It's Artix.

jordens, 2016-10-10:
@jbqubit are these the notes from the conversation that you wanted to send around?

gkasprow, 2016-10-10:
It is XC7A15T-1CSG325

gkasprow, 2016-10-10:
What crystal do you want to have attached to the SI5324 chip? 114,285MHz?
Where inputs of the Si go?

sbourdeauducq, 2016-10-11:
Yes, 114.285MHz sounds good. One input to a FPGA general-purpose I/O.

gkasprow, 2016-10-11:
OK, done

sbourdeauducq, 2016-10-11:
Also, the Si5324 clock input, here and everywhere else, should be CLKIN1 (and not CLKIN2), so we can use its automatic free-run feature.

gkasprow, 2016-10-11:
Done.
I spent almost all FPGA pins - there are just a few left in 1.8V bank
Below is very preliminary version of schematic. We are still working on it, there are some components missing, The supply is not finished. There is still some mess.
The pdf schematic is clickable, it means that once you click on the module it will redirect to corresponding schematic. All the nets are also links to the other end of the net.
[Sayma_RTM.pdf](https://github.com/m-labs/sayma/files/522884/Sayma_RTM.pdf)

jbqubit, 2016-10-14:
What power is needed for this FPGA? Where will it be generated?

gkasprow, 2016-10-14:
It is generated locally, just at the FPGA.
The power requirements are no high, but we need 3.3, 1.8, 1.2 and 1.0V
I want to use single chip from Analog (adp5052) to do that locally. It
provides sequencing and was used in several Artix devkits.

dhslichter, 2016-10-17:
- ADC need buffer amplifiers ADA4927 running from +/- 5V rails, low noise, gain 1 (R_F=R_G=301 ohm resistors, see datasheet Fig 46), output common mode voltage 0.9 V, with differential attenuation between output lines of ADA4927 to ensure that max (railed) ADA4927 output amplitude will give no more than 2.8 Vpp differential at ADC inputs (see AD9656 spec).  For +/- 5V supplies, ADA4927 puts out up to +/- 3.8 V on each line. Thus total differential swing is 3.8*4=15.2 Vpp.  Attenuator circuit should be differential between the two output traces, reduce amplitude from 15.2 Vpp to 2.8 Vpp, accounting for 2.6 kOhm differential input impedance of AD9656.  Ideal load impedance for ADA4927 closer to 1 kohm.
- use ADA4927-1 for reduced crosstalk
- no protection diodes on ADA4927 inputs or outputs
- circuit must be designed for dc coupling of signals to ADC

gkasprow, 2016-10-17:
Why not to put the driver at the mezzanine? Some users may want to use a balun as ADC driver to get the best SFDR.

dhslichter, 2016-10-17:
This is also a possibility, but:

- would then require sourcing more current at the -12V rail (or more likely, a -6V rail) to the mezzanines
- allows us to provide input protection/level shifting to ADC without relying on mezzanine card to provide this -- simplifies task for mezzanine design
- ADA4927 has much lower harmonic distortion than AD9656 up until 300-400 MHz, when they become roughly equivalent (still low, around -70 dBc), so you are not really degrading the signal that much (if at all) by including it.

If we place the ADA4927 on the mezzanine instead, we will need to supply a low noise -6V rail capable of ~100 mA per mezzanine to support 2x ADA4927-1.  I don't really like the idea of having to put a switching charge pump inverter on the mezzanines to create -6V at these relatively high currents, since it will introduce additional power supply noise.

gkasprow, 2016-10-18:
We can apply -6V rail to the mezzanine with limited current capability just to supply the driver chips.

We need to have this voltage anyway.

hartytp, 2016-10-18:
@gkasprow @dhslichter I'm with Daniel on this one: I'd rather keep the ADA4927 on the main Sayma board to simplify the mezzanines.

closed 2016-10-23 by jbqubit