sinara-hw/sinara #23: Sayma clock mezzanine interface

jbqubit, 2016-10-10:
Interface between Sayma clock mezzanine and Sayma RTM motherboard. Same interface will be used for passive clock distribution board by WUT and on-mezzanine clock distribution by Oxford.

@cjbe @weidazh please forward to Harty

**geometry and mechanical**
- [ ] what is minimum clock mezzanine geometry? WUT is presently assuming 43 x 80 x 4.2 mm
- [ ] what IC sets/constrains this geometry
- [ ] use same mounting to RTM motherboard as analog mezzanine interface #8? 
- [ ] if space is tight on RTM motherboard, mezzanine could be moved to AMC motherboard or RF-backplane clock distribution slot

**RF shield**
- [ ] minimum interior height under shield lid

**PCB material/layers**
- [ ] 4-layer PCB, top layer .020" Rogers 4350B
- [ ] single ground for analog and digital

**connector choices**
- power and IO
  - [ ] same 40-pin connector as analog mezzanine interface #8? 
- RF to/from motherboard
    - clk output: 
    - clk input:  

**power/IO header routing**
- digital
    - [ ] does following approach in #8 work?
    - [ ] how many IO are needed? 

- digital and analog power, low noise (generated on AMC)
    - what voltage(s) are needed, current? 

**ADCs**
- are inputs on 24-bit calibration ADC desired ?

hartytp, 2016-10-11:
@jbqubit the biggest factor in the geometry is the number of PLLs we'll need: if the 100MHz reference clock reaching the mezzanine meets the phase noise spec up to ~100kHz then we can lock the microwave VCO directly to it. If not, then we'll need to add an extra 100MHz clean-up oscillator, which makes the board bigger.

For the PLL board:
RF Input: 3.3V PECL square-wave, please (2 x SMP?)
RF Output: as input
Power: we need a clean +3V3 analog rail. +5V would be good too.

Is the "passive board" a mezzanine, which takes the front-panel SMA DAC clock input and connects it to the RTM? If so, can we give it a TCM2-43X+, ADCLK905 (or your favourite equivalent IC) & AC coupling caps to perform sine-square conversion. That way, the RTM only needs to be designed to take in a LVPECL square wave.

dhslichter, 2016-10-14:
Can the mezzanine host LDOs to make its own clean 3.3V and 5V rails from a low noise +6VDC rail?  This would allow sharing of power supplies with the analog mezzanines, which would be nice.  One could use the same power/IO connector with a pin-compatible wiring.

hartytp, 2016-10-14:
@dhslichter I'd be happy with putting LDOs on the clock mezzanine.

gkasprow, 2016-10-14:
This is the right way of clean power generation.
I placed on the schematic the IO connector with same pinout as other
mezzanines

sbouhabib, 2016-10-19:
I agree with greg the last LDO step should be on the mezzanine, and there shoul be a possiblity to get the hight possible voltage for future use (12V mains) even if it is not used at the moment, a lot of VCOs use 10V, 12 V even more, lacking this and even stepping up from 6V could be problematic in the future

jbqubit, 2016-10-31:
@hartytp Do you want access to one of the 24-bit calibration ADCs?

hartytp, 2016-10-31:
@jbqubit If there is a spare one, I think it would be a good idea.

jbqubit, 2016-11-02:
@gkasprow Is this possible? I don't know if all the calibration ADCs are allocated.

gkasprow, 2016-11-02:
At the moment we have all 16 ADC inputs connected to ADC/DAC mezzanines.
Shall I add another calibration ADC? Why would you need it?

hartytp, 2016-11-08:
> Shall I add another calibration ADC? Why would you need it?

Don't bother. I have no immediate plans to use an ADC on the clock mezzanine. My comment was just that if there had been a spare ADC channel, then it might come in use later.

closed 2016-12-12 by jbqubit