sinara-hw/sinara #24: digital servo (PID)

jbqubit, 2016-10-11:
Here's a proposed implementation. The proposal at times draws verbatim from Leibrandt & Heidecker (2015) but is distinct to Sayma. See https://arxiv.org/abs/1508.06319

- Each Sayma implements 8 digital servos (DS).  

- Let the DS processing frequency fDS be the ADC data clock frequency (nominally 125 MHz)

- Name the input and output streams available on each Sayma. 
    - INs are any of the 8 ADCs, 8 spline-modulated-CORDICs (decimated to fDS), 8 digital servo OUTs 
    - OUTs are any of the 8 Sayma DACs, 8 digital servo INs
    - LOs are 8 special-purpose CORDICs with amplitude, frequency and phase set by DRTIO at the outset of an experiment; frequency bounded by fDS
    - DINs are any of 8 digital inputs (set by RTIO)

**IIR filters** are single- and two-pole infinite impulse response filters
∗ use sufficient resolution for coefficients and data to prevent rounding errors (35-bits for 16-bit data)
∗ filters parameters may be configured by RTIO at outset of an experiment 
∗ filters may be enabled or disabled (unity transfer function) by RTIO at outset of an experiment
∗ when IIR filter are disabled their latency is minimized

**DS signals** Each DS processes a pair of inputs (AIN0, AIN1) and generates a single output (AOUT) 
∗ AIN0 is obtained by applying 1st order IIR to any one of INs
∗ AIN1 is obtained by applying 1st order IIR to any one of INs
∗ DIN is one of the digital DINs
* LO0 is one of LOs
* LO1 is one of LOs

# DS Modules
Each DS includes several signal processing modules which can be assembled between its inputs and output as follows. See Fig 1 in Leibrandt and Heidecker (2015). 

**lock-in** The lock-in module takes AIN0 or AIN1, optionally applies a 1st order IIR pre-filter (typically a high-pass filter), multiplies it by LO0, and optionally applies a 1st order IIR post-filter (typically a low-pass filter). The output is LOCKIN. 

**phase detector** The phase detector module takes AIN0 or AIN1, multiplies it by both quadratures of  LO1, low-pass filters the products with 1st order IIR, and then takes the arctangent of the ratio of the results. The phase detector output (PHASEDET) is unwrapped such that it can track phase noise with an amplitude up to ±(2**16)*π radians. Undetected cycle slips are avoided as long as the absolute value of the input phase minus the LO phase changes by less than π radians in each 10 ns FPGA clock cycle. The output is PHASEDET. 

**loop filter** The loop filter module input is any of {AIN0, AIN1, LOCKIN, PHASEDET}. It applies up to four sequentially first- or second-order IIR filters. For example, a proportional-integral-integral-derivative (PIID) filter, which is third-order, can be configured by enabling three sequential first-order IIR filters (PI, PI, PD where PI is a proportional-integral filter and PD is a proportional-derivative filter). The output is LOOP.

**re-lock** The re-lock module input is any of {AIN0, AIN1, LOCKIN, PHASEDET, DIN}. If the input signal is within the software defined “locked” range, the re-lock module output is zero. If the input signal falls outside the “locked” range, the re-lock module holds the output (and internal state) of the loop filter module constant and outputs a triangle wave sweep with an amplitude that doubles every cycle and a constant slew rate. Once the input signal returns to the “locked” range, the loop filter module is re-enabled and the re-lock module output ramps back to zero. The re-lock output is RELOCK.

**sweep** Is a module that generates a triangle wave. The output is SWEEP.

**modulation** Takes AIN0 or AIN1 and outputs MOD.

**sum block** The sum block computes LOOP+SWEEP+RELOCK+LO0+LO1+MOD where inclusion of each term can be toggled on/off to generate AOUT. 

**monitor** Is a module that permits monitoring of the operation of the DS. Output is MON. 
- Let SIGs be {AIN0, AIN1, DIN0, DIN1, DIN2, LOCKIN, PHASEDET, LOOP, SWEEP, MOD, AOUT}
- Let TRIGSIG be one of SIGs. 
- Trigger: if TRIG ==0 and TRIGSIG > THESH then set TRIG=1
- If TRIG transitions from 0 to 1, capture M samples of SIGs for a user-specified interval (decimating as necessary); set TRIG=0 
- Relay MON to Master PC by fire and forget RPC.

# Questions
- As a first approximation is it feasible to create a wrapper for the existing NIST Verilog? This could be an economical approach which is a segue to a later re-implementation in MIGEN.
- What is the order of priority in implementing DS modules? My take: loop filter < re-lock < lock-in < phase detector < modulation < monitor

jordens, 2016-10-11:
That wouldn't be economical.
* It's for Spartan 6. 
* It doesn't meet timing. While that may or may not be a problem for the digital servo as is, it would mask other timing failures ad we'd have no easy way to tell whether the rest of our design is ok.
* It's deeply entangled with the OpalKelly blackboxes. The host side code would need porting as well.
* We wouldn't maintain/test that code.

jordens, 2016-10-11:
This is to entrenched IMHO. I'll write up a (counter-) proposal.

hartytp, 2016-12-16:
@jordens Have you had a chance to write a proposal for this? Could you give me a rough quote a minimal noise-eater type firmware?

jordens, 2016-12-16:
Yes. Thanks for poking me.

jordens, 2016-12-21:
@hartytp I wrote up what I can see being relevant at https://github.com/m-labs/sinara/wiki/Servo

hartytp, 2016-12-21:
@jordens Thanks for doing that.

Can you provide a quote for the subsection of that required for a minimal PI controller for laser intensity servoing using the SAWG amplitude?

jbqubit, 2017-01-26:
@jordens Please update the wiki to group the "Items" section so it's clear which are required for the sort of minimal PI controller @hartytp requested. 

- "monitoring taps within the signal chain for debugging" is a feature of the NIST digital servo that's very helpful at configuring the signal path. Please comment on how this fits into your development plan.

jbqubit, 2017-02-21:
@jordens please comment on my questions from 26 days ago. I'd like to close this Issue.

jordens, 2017-02-23:
Having "monitoring taps within the signal chain for debugging" is an important feature, that's why it's in the plan. We haven't decided how to best implement this.

closed 2017-02-23 by jordens

dhslichter, 2017-09-14:
@jordens @jbqubit what is the current status of this?  We are interested in this at NIST as well and are curious whether this is funded/going to be funded/unfunded?  What is an estimated timescale for developing this given the other ongoing work?

jordens, 2017-09-14:
IIRC this is not yet funded. Right now we are booked for the next months but a solid amount of work should first go into writing the specification the feature set that handles the dominant use case (PI power stabilization).

hartytp, 2017-09-14:
@dhslichter: I discussed this with @jbqubit a bit at NACTI. Our interest at Oxford would initially be in having the NU servo ported to Sayma (i.e. no fancy features like demodulation etc). AFAICT, other than implementing JESD Rx, that shouldn't be a huge amount of work (but, correct me if I'm wrong about that @jordens); the NU-servo is already fairly well specified and some elements of the code have already been written and simulated.

However, as far as I know, JESD Rx is not funded yet -- but @jbqubit is probably the person to ask about future plans. If it doesn't get funded, then my current backup plan is probably to hack NU servo onto Sayma using the FMC + VHDCI_carrier + Novo as a short term solution...

dhslichter, 2017-09-14:
I think our initial use case would be similar to Oxford, a simple NU servo-type situation ported to Sayma, basically allowing us to feed back on the amplitude of a sinusoidal DAC output (e.g. driving an AOM) to noise-eat laser pulses using a photodiode being read by an ADC input.  We'd be interested in both CW noise eating and in pulsed noise eating (i.e. on during a pulse, integrator hold and loop open not during a pulse).  The system in full generality (demodulation, lots of filter options) is not critical for the initial phases.  One feature that would be nice would be to have would be to somehow allow for different laser pulse levels on the same photodiode for the same channel -- you might want to servo an AOM drive to one intensity for one kind of pulse and a different intensity for a different kind of pulse.  In general, setting the PI parameters optimally for one of these pulses will lead to oscillation or too-slow response for the other (depending on whether its amplitude is larger or smaller).  So we'd like a situation where we could have more than one (let's start with 2 or 3, say) simple PI loops which share an input signal, can have different setpoints and PI parameters and integrator holds, and have their outputs multiplexed so you can switch on the fly between which PI loop is feeding back on the DAC output.  

Further comments @dtcallcock @r-srinivas @danielkienzlerNIST @yw-NIST @erickson-nist @dleibrandt?

hartytp, 2017-09-15:
@dhslichter Good, that's all in NU-servo AFAICT, see [the wiki](https://github.com/m-labs/sinara/wiki/NovoUrukulServo).

So, it looks like the main roadblock is just getting JESD Rx funded...

-----

Quoting sections of the wiki relating to your points:

> We'd be interested in both CW noise eating and in pulsed noise eating (i.e. on during a pulse, integrator hold and loop open not during a pulse).

"Integrator hold" logic is: integrator active if profile active AND RF switch has been open for channel-specific delay AND (per-channel) integrator enabled bit set. Integrator enabled bit must be settable in real time. Use case for this is lasers which must be on at the same time, and also share a photodiode. Need to servo one, then the other.

Smart linkage of Urukul RF switches and IIR action: servo for active channel/profile enabled when the RF switch is open. Configurable/fixed (~1-10 µs) delay between switch opening and servo activating to allow for transients, AOM propagation, photodiode settling etc.

> One feature that would be nice would be to have would be to somehow allow for different laser pulse levels on the same photodiode for the same channel -- you might want to servo an AOM drive to one [...]

For each ADC-DDS channel, we have a fixed number (at least 4, up to 16 if "conveniently possible") of "profiles" on Kasli each with a user-programmed frequency, phase and amplitude as well as individual filter gains, set points and accumulators.

Active profile for each channel is set via DRTIO.

Expose active profile on moninj.

Tie the IIRs to the DDS ASF channels and route the ADC matrix to the filter inputs. If the crossbar fabric for the routing from ADCs to the IIRs/DDSs is too costly then do fixed routing, with fixed EEM-IDC and ADC:IIR:DDS assignment.

hartytp, 2017-09-15:
While it's on my mind, a couple of things that need specifying/designing/implementing once the JESD Rx + ADC interfaces are written:

- What speed do we want to run the loop filter at? IIRC, the JESD interfaces limits the loop bandwidth to ~300kHz (without taking into account delays due to AOMs etc), so >=1MSPs should be fine.
- How do we want to do the ADC down sampling? IIRC, the ADC is pretty low noise (particularly with the PGIA front-end), so no need for anything too fancy here.

jordens, 2017-09-15:
To adapt nu-servo to Sayma, we'd need quite a bit of work still:

* JESD RX
* ADC configuration
* ADC downsampling and noise bandwidth limiting (just picking samples would not be wise)
* ADC synchronization scheme
* Specifying SAWG DDS control method either using profiles like nu-servo or injecting on a multiplicative/additive term
* Synchronizing SAWG DDS control
* Since nu-servo is a fixed mode of operating both the ADC and the DDS, this would by default also be the case for Sayma.