sinara-hw/sinara #26: CSWA2-63DR+ replacement

jbqubit, 2016-10-14:
Daniel Slichter reports:
CSWA2-63DR+
Recent tests of this part indicate it has a slow (~10s of us) time constant to the transmitted power after switching on. DO NOT USE. We will specify a new part ASAP after testing alternative options.

dtcallcock, 2016-10-14:
Just to give a bit of context, I was evaluating this part for use as a high-isolation switch for RF-outputs on mezzanine cards.  

I could go into details but I think we should wait until we get onto actually specifying the circuitry to go on the different analog mezzanine configurations.

jordens, 2016-10-17:
@dtcallcock which circuit?
I can see two ways of wiring it where one would expect such a behavior and one way where one wouldn't.

dtcallcock, 2016-10-18:
[Datasheet](https://www.minicircuits.com/pdfs/CSWA2-63DR+.pdf)

Tested on both the minicircuits eval board and a board we designed in-house.

Signal (1.7GHz, 10dBm) applied to RF COMMON.  RF1 terminated externally.  RF2 output measured on RF with a fast detector.  47pF Cblock on all three ports.  Control input was 3.3V logic, 30us pulse duration, 1kHz rep rate.  Settling to within 0.05dB took ~4us.  Switching the role of RF1 and RF2 didn't improve things.

Also tested a Minicircuits ZASW-2-50DR (connectorized switch ubiquitous in many labs) in the same setup as a check and it settled to 0.05dB in ~100ns. 

I have selected two possible replacements and ordered eval boards.  Will report back in a few days.

dtcallcock, 2016-10-18:
Also, I should add that if the switch is held on and a pulsed microwaves signal is applied the problem goes away.  This points to it being something to do with the switching circuitry and not the microwave signal itself.

dtcallcock, 2016-10-22:
Given that we want a switch that works at lower frequencies than 50MHz I have selected two more parts (right most two columns) and ordered eval boards.  

[image]

jbqubit, 2016-10-31:
@dhslichter @dtcallcock are the eval boards on hand?

dhslichter, 2016-10-31:
Not as far as I know, will check.

dtcallcock, 2016-11-02:
Our purchasing is gummed-up at the minute I'm afraid.  I will test and report back as soon as they get here.

dtcallcock, 2016-11-14:
I got the HMC349ALPCE eval board tested.  Results are below.  Let me know if there is anything else you want looking at. 

Acquisition of a HMC849ALP4CE eval board is turning into a nightmare so I will leave it if nobody has a particular problem with HMC349ALPCE, which seems good for our purposes.

[Datasheet](http://www.analog.com/media/en/technical-documentation/data-sheets/hmc349alp4c.pdf)

Tested with 1GHz, 0dBm input to RFC (RF common)
100us pulses @ 1kHz
5V supply, 3.3V logic (9ns rise/fall)
Unused RF output externally terminated

RF2 output, rise
50% control to 10% RF = 130ns
50% control to 90% RF = 200ns
50% control to 0.1dB of final = 360ns
50% control to 0.05dB of final = 500ns

RF2 output, fall
50% control to 90% RF = 30ns
50% control to 10% RF = 50ns

RF1 output, rise
50% control to 10% RF = 120ns
50% control to 90% RF = 190ns
50% control to 0.1dB of final = 310ns
50% control to 0.05dB of final = 550ns

RF1 output, fall
50% control to 90% RF = 40ns
50% control to 10% RF = 60ns

To test the effect of the switching rate I decreased pulse length to 10us and varied rep rate from 1-50kHz.
All rise times increase with switching rate smoothly by 10ns (from 1 to 50kHz)
All fall times increase with switching rate smoothly by 2ns (from 1 to 50kHz)

Changing the frequency between 25MHz (low freq cutoff due to DC-block caps on eval board) and 2GHz changes the rise times by +/- 10ns in a somewhat random but smoothly varying manner.

jbqubit, 2016-11-28:
Thanks for making these measurements @dtcallcock. This informed #79.

dtcallcock, 2016-12-06:
Schematic has the HMC849ALP4CE on it, not the HMC349ALPCE that was decided on.  As it happens the eval board for the HMC849 just turned up so I can test it if this change was intentional and not just a mistake.

jbqubit, 2016-12-06:
@sbouhabib close after updating schematic

jbqubit, 2017-01-26:
HMC349ALP4CE is in the schematic. Closing.

dhslichter, 2017-12-20:
Measure before you get too excited @hartytp .  We were using a CSWA-63DR from Minicircuits, which specs a much faster rise/fall time, but it turned out to have a very unpleasant few percent tail out to tens of microseconds.  We tested the HMC349 ourselves and confirmed it doesn't have this problem.  You'll have to get some eval boards for these other parts and have a look if you want.  

Negative control voltage sounds like a pain in the neck.

jordens, 2017-12-20:
I would take convenient power supply/driving logic and low long-term (µs-ms, phase, amplitude) transients over fast(er) switching.  The HMC232 also seems to be obsolete which is a real issue.

dhslichter, 2017-12-20:
Agreed with @jordens  here.  If you need a faster switch for some lines, put an external switch on.

dhslichter, 2017-12-20:
Return loss for HMC435 in "off" state is not that hot at the relevant frequencies, also.

hartytp, 2017-12-20:
> Measure before you get too excited @hartytp .

Of course; I wouldn't propose a change like this without testing first. Just asking if there is anything obvious I missed that would make it not even worth testing a different switch.

> w long-term (µs-ms, phase, amplitude) transients over fast(er) switching. 

Agreed, this is more important than faster switching. We'd want to test this carefully before deciding to change the switch.

> The HMC232 also seems to be obsolete which is a real issue.

Good catch, sorry for missing that. Okay, let's rule out the 232.

> Return loss for HMC435 in "off" state is not that hot at the relevant frequencies, also.

Good catch, I hadn't noticed that those terminators are AC coupled so it's reflective below a few hundred MHz.

Okay, let's leave this then...

klickverbot, 2017-12-20:
As for the HMC232 being obsolete, there is the newer HMC232A (speced to be a bit slower, but as people have pointed out, t_10-90 alone doesn't mean much anyway).

gkasprow, 2017-12-20:
negative control voltage is not that big issue. Depends how fast it needs to be. We need to generate 0 or -5V from LVTTL signal and this can be done easily  with delay of a few ns. The only drawback is -5V supply, but the current needed is low (~20uA) so charge pump would do the job.

hartytp, 2017-12-20:
@dhslichter Looking again at the HMC349ALP4CE we currently have on Urukul, the terminators are also AC coupled (which is actually on the schematic). Off-state return-loss versus frequency looks to be about the same as the 232. So, if that's a problem for the HMC232 then it's a problem in the current version of Urukul.

Pending the results of testing, I don't see any issues with the HMC232, and would prefer that to the current HMC349 on Urukul.

hartytp, 2017-12-20:
My feeling is that the HMC349 is a little slow for some applications. Settling time to 1% amplitude is about 360ns from @dtcallcock's measurements above. While it's nothing we absolutely can't live with, it will be a bit inconvenient for, say 1us pi-pulses. So, I would be in favour of something faster if "conveniently possible".

So, pending characterisation (but, for now, assuming that faster rise-time to 90% really means faster rise-time to 0.1%) I'd prefer the 232A or 435. A plus of the 232A is that it's the only fast switch which has DC coupled termination resistors.

For BaseMod and MixMod, the 232A seems like the best option (again, pending measurements).
- On those boards we already have a -5V rail, and the level translation is basically trivial. This is in contrast to Urukul where we'd need to add an extra -5V rail, which makes things more complicated.
- On MixMod, it's also nice that this switch goes up to 12GHz.
- IMHO the rise-time is more important for Sayma than Urukul, since Sayma is more likely to be used for high-fidelity microwave quantum logic, where rise-time matters most.

What do people think? If I do the testing and these switches turn out to perform well, would we use them?

dhslichter, 2017-12-21:
@hartytp I am not going to propose we make any changes whatsoever until the settling time of the HMC435 or HMC232A is measured, and the assumption that faster 90-10 time has a correlation with settling time is shown to be false by our mini-circuits experience.  I am totally open to a better switch part if it exists; let's not waste any time or energy discussing possibilities until we know that the switch is actually better.

dhslichter, 2017-12-21:
> Off-state return-loss versus frequency looks to be about the same as the HMC435. So, if that's a problem for the HMC435 then it's a problem in the current version of Urukul.

Good catch, @hartytp , I had missed this previously.  Again, if the HMC435 has equivalent or better settling properties then I am OK with it.

hartytp, 2017-12-21:
> Edit: In sum, we would all be excited to know what the results of your measurements are, and if they are good it seems sensible to change parts. But for now let's table any and all discussion/details until the measurements are completed.

Sure. All I was looking for for now was whether people were going to veto these parts for other reasons, such as needing negative voltages. If those aren't showstoppers for people (personally, I think that shouldn't be for either Urukul or BaseMod/MixMod) then I'll test both parts in the New Year.

dtcallcock, 2017-12-21:
Yeah, I selected the best switch with a 5V bias and single digital control input to keep things simple.  For switches with no bias and two control lines I could imagine you need to use the actual proposed drive circuitry to do a meaningful test.

>IMHO the rise-time is more important for Sayma than Urukul, since Sayma is more likely to be used for high-fidelity microwave quantum logic, where rise-time matters most.

On Sayma, wouldn't you just use the DAC to create very short pulses?  Even with a faster switch there is an on/off asymmetry and propagation delays to calibrate.  I thought the purpose of this switch was to totally shut off DAC/preamp noise when the channel isn't being used - not create pulses.  For Urukul I can imagine the faster rise time being useful as reprogramming the DDSs to create pulses would probably be tedious.

hartytp, 2017-12-21:
> Yeah, I selected the best switch with a 5V bias and single digital control input to keep things simple. For switches with no bias and two control lines I could imagine you need to use the actual proposed drive circuitry to do a meaningful test.

ACK. That's true. That will need testing. 

I'll look into the best way to drive these switches in the New Year (although, if anyone has experience with this, feel free to post).

> On Sayma, wouldn't you just use the DAC to create very short pulses? Even with a faster switch there is an on/off asymmetry and propagation delays to calibrate. I thought the purpose of this switch [...]

For "very short pulses" yes. But, I wouldn't call 1us very short in this context. For that, I'd generally like to have the option of using the switches.

My guess is that the thermal transients involved in switching the DAC/amp right on and off will be quite bad, reducing the fidelity.

If getting faster switches is a massive pain then it's not worth it. But, if we can make a simple change to get faster switches then I'd argue it's worth it.

dhslichter, 2017-12-21:
> My guess is that the thermal transients involved in switching the DAC/amp right on and off will be quite bad, reducing the fidelity.

Really?  The DAC transients shouldn't be a problem, it draws the same amount of current no matter what the output.  Changing the output voltage involves just rebalancing the current between two arms of the differential output.  If your external termination resistors have a reasonable tempco I doubt you'll see anything.  The digital stuff is going to use the same current no matter what the data are.

The amp will have some transient perhaps but it also runs a pretty substantial bias at dc so again, thermally it's probably more stable than you think.  Worth running some tests with an eval board to see whether you get thermal duty cycle effects and at what level.

Extra driving circuitry, and especially extra rails, seems like a pain if the performance of the alternative switch is not demonstrably superior.

hartytp, 2017-12-21:
> and especially extra rails,

Well, BaseMod/MixMod already have the -5V rail, so no extra rails required on there. And if we did make a change to Urukul, we'd probably use the 435, which doesn't need the negative rail. But, point taken.

> Really? The DAC transients shouldn't be a problem, it draws the same amount of current no matter what the output. Changing the output voltage involves just rebalancing the current between two arms [...]

> The amp will have some transient perhaps but it also runs a pretty substantial bias at dc so again, thermally it's probably more stable than you think. Worth running some tests with an eval board to see whether you get thermal duty cycle effects and at what level.

Okay @dtcallcock @dhslichter. All reasonable points. So, the intended use-case for Sayma is that the DAC does the bulk of the switching (e.g. pulse shaping) and the RF switches are just there for extra isolation during "prolonged off-periods".

For Urukul, we're not aiming for absolute top-notch performance, so we can stick with the simpler slower switch.

I'll go along with that. Let's leave the switches alone for now then. If we do find thermal transients in Sayma when switching with the DAC then we can revisit this.

closed 2017-12-21 by hartytp