sinara-hw/sinara #27: RTM backplane clock distribution

jbqubit, 2016-10-14:
The microTCA.4 RF Backplane will be used for clock distribution. 

- input: SMA from front panel
    - 100 MHz, 50 Ohm, +13 dBm
    - low phase noise oscillator is supplied by physicists
- output: RTM backplane
    - routed to RTM slots 4 to 12
    - if CLK is used, 100 Ohm differential 
    - if REF is used, 50 Ohm single-ended
    - path-length equalized 
- clock driver
    - unspecified

- target performance
   - assume 100 MHz master clock is SDI LNFR-100: -158 dBc/Hz at 1 kHz, -177 dBc/Hz at 10 kHz ([ref](https://docs.google.com/spreadsheets/d/1mkg7u-MaYnXeptUxOmuWZalwc4X5r3Jy-z8v8w91Q00/edit?usp=sharing))
   - noise floor of AD9914 (ADC) at 100 MHz: -145 dBc/Hz at 1 kHz, -155 dBc/Hz at 10 kHz ([ref](https://docs.google.com/spreadsheets/d/1mkg7u-MaYnXeptUxOmuWZalwc4X5r3Jy-z8v8w91Q00/edit?usp=sharing))
   - noise floor of AD9154 (DAC) at 100 MHz: -135 dBc/Hz at 1 kHz, -141 dBc/Hz at 10 kHz ([ref](https://docs.google.com/spreadsheets/d/1mkg7u-MaYnXeptUxOmuWZalwc4X5r3Jy-z8v8w91Q00/edit?usp=sharing))
   - goal: don't adversely impact ADC or DAC due to clock distribution
   - side goal: better clock distribution enables more applications where high-frequency LO is generated from the 100 MHz

- [ ] 1:8 clock splitting: passive or active? **ACTION ITEM**
    - active example, boxed solution: Spectra Dynamics [HPDA-100RM
](http://www.spectradynamics.com/products/hpda-100rm-high-performance-distribution-amplifier-80-20mhz/) achives 100 dB isolation, -174 dBc/Hz at 10 kHz, +16 dBm output
    - active example IC: Linear LTC6954-1 has 3 LVPECL outputs (use 4 chips, 9 outputs; has programmable delays) achieves -164 dBc/Hz at 10 kHz, 0.7 ps/C, channel-channel skew +/- 50 ps
    - passive example IC: Mini-Circuits SCPA-8-13-75+ achieves only 20 dB isolation, up to 250 ps channel-to-channel skew, unknown phase noise, unknown temperature sensitivity

[ ] - clock type on backplane: LVPECL or sinusoidal? **ACTION ITEM**
- [ ] impacts choice which RTM backplane lines to use: CLK (differential) or REF (single-ended) 
- [ ] LVPECL clock driver
    - sinusoid to digital conversion: LTC6957-1 achieves -160 dBc/Hz at 10 KHz, chip-to-chip phase skew < 0.2 deg, tolerant of supply noise at 100 MHz 500 mVrms
    - choose: RF amplifier and passive power splitter OR separate driver for each clock line

jbqubit, 2016-10-14:
@gkasprow In phone conversation today it was mentioned that the "RTM backplane" was single-ended. However, the DESY [specification sheet](http://mtca.desy.de/sites/site_mtca/content/e172206/e205636/e212584/e248086/uRFB_concept_Datasheet_19.12.2014_eng.pdf) it looks like both single-ended and differential options are available.

hartytp, 2016-10-16:
All looks good to me.

I'd vote for LVPECL clocks, as it's easier to distribute them without degrading the phase noise. The LTC6957-1 looks like a nice way of squaring up the input square wave. Note that the phase noise degrades as the input power decreases, which suggests that we should avoid passive splitting before squaring if possible.

Is the delay + divide functionality of the LTC6954-1 required for the 100MHz clock? This IC has a propagation delay temp co that's about an order of magnitude larger than a good straight buffer IC. If we want to keep the option open of using this 100MHz clock to generate LOs, we need to be careful over this. For two buffers in series, the (typical) temp co is 2*0.7ps/K = 1.4ps/K = 5deg/K @10GHz, which would be a complete killer!

gkasprow, 2016-10-16:
Is adclk944 OK for you as a clock splitter for mezzanine 100MHz distribution?

hartytp, 2016-10-16:
ADCLK944 would probably be my first choice for this (excellent noise and propagation delay stability). But, there are other good HMC/ADI/LTC/whoever parts as well, which I would be equally happy with.

gkasprow, 2016-10-18:
That’s true, for low frequency ADC clock distribution we used diff lines.

sbouhabib, 2016-10-19:
@jbqubit: the RF back plane does implement differential lines that go out on an ERNI connector, for "Square" clocks i would say up to 100 MHz to keep good performance, the single ended lines are for RF and LO distribution.

sbouhabib, 2016-10-19:
Probably LVPECL differenctial clocks would be the best but there is one issue, the rise and fall times of LVPECS are quite small (usally leading to better ADC/DAC performance) which makes the signal bandwidth wider, thus upon going through at least two connectors and "long distances" the various harmonics have a different flight time and matching (the higher the harmonic, the worse the matching and the higher frequencies are damped more etc...) then when they reach the reciever they will not be so pretty. same thing goes for temeprature drifts(if it is an issue) upon temeprature changes,different harmonics get different delays and the same thing happens. this could be tested to see performance degradation (i can check, but im not really sure LVPECL was really tested by the designers)

jbqubit, 2016-10-24:
Discussed this in google hangout on 10/21. 

@sbouhabib raises an important consideration. Path from RTM clock generation slot to the Sayma RTM analog mezzanine traverses many steps. Many transitions... 
- RTM clock generation module
- connector to RF backplane
- connector to Sayma RTM motherboard
- connector to Sayma RTM clock mezzanine 
@sbouhabib What did you find? 

Narrow-band 100 MHz distribution directly to the Sayma RTM clock mezzanine avoids these difficulties. Low phase noise narrow band distribution amplifiers are available. 
- SDI HPDA-100RM-A operates at 100 MHz, 1:5 $3k/ea ; 1:10 version also available [link](http://www.spectradynamics.com/products/hpda-100rm-high-performance-distribution-amplifier-80-20mhz/)
- Wenzel Associates makes a variant of the LNDA-3 at 100 MHz, 1:3k $538/ea [link](http://www.wenzel.com/model/distribution-amplifier-lnda-3/)

hartytp, 2016-10-24:
@sbouhabib What are you worried about here? Reflections from impedance discontinuities at the transitions?

The reasons I'd prefer to distribute the 100MHz clock via the RTM backplane if possible are:

1. **Convenience** -- don't need a bulky external amplifier + an extra SMA going to each RTM
2. **propagation delay stability** these kinds of distribution amplifiers typically have temperature coefficients of ~1ps/C (1deg/K relative to 3GHz reference freq, 3.6deg/K relative to 10GHz), which is much worse than good PECL clock buffers (<~0.1ps/K).

Even though there are a few transitions in the RTM clock distribution chain, they are all made using high-frequency connectors + layout techniques. Thus, even in the GHz regime, reflections should be small. Moreover, as the whole thing is rigidly bolted together, any phase shifts due to reflections will be stable, and hence should not cause problems.

jbqubit, 2016-10-25:
ping @sbouhabib

sbouhabib, 2016-10-26:
I totally agree that the RF backplane is the way to go, I was proposing this long ago from the first visit that came to Warsaw. 
I was just talking about using LVPECL on the ERNI connectors but for this frequency (100 MHz) there should be no problem, the tests were done for an 81 MHz signal, but without actual temperature drifts measurements, I wasnt actually afraid of the buffers or amplifiers but the ERNI connector itself (not really a good high frequency analog connector, its fine for digital data but) makes a little bit of a poor match at higher frequencies, but for the 100 MHz signal with its harmonics it should be fine. 
and my suggestion too was to keep the fallback option for the DAC clock on the front panel, 
maybe it is also worth not blocking the way of using a (100MHz?) reference signal on the analog RF-backplane connector for generating the clocks as well

hartytp, 2016-10-26:
@sbouhabib What are the ERNI connectors like at a few GHz (the PECL drivers will give fast rising edges with lots of harmonics)?

> maybe it is also worth not blocking the way of using a (100MHz?) reference signal on the analog RF-backplane connector for generating the clocks as well 

Do you mean distributing both a PECL square-wave and a 100MHz sine over the RF backplane and then routing both to the clock board?

sbouhabib, 2016-10-26:
I had some tests made myself for the connectors, and i downloaded the model for the connector from erni, Im not at my work PC at the moment I should send the sims tomorrow, but in general at a few GHz already the ERNI is not so optimal, for narrow-band/single frequency operation it is already on the edge due to mismatches and imbalance in pair length.

If the analog channels on the backplane are not used for anything at the moment, it is a possibility to think about, a single line with a physical jumper is not much of additional work.(just a thought) even though the signal wouldn't be provided at the moment.

I just got this info: the clock distribution was tested using the LVPECL buffer HMC987 @81MHZ which has very fast edges,I got the phase noise results, i will give them a quick analysis as and get back to you

sbouhabib, 2016-10-27:
Unfortunately my concerns were true.
The ERNI Connector performance degrades at high frequencies, the measurements that were done incorporated an ADT2-1T transformer as the receiver which acts as a LPF above a few 100 MHz so the high frequency parameters were "cut-out" and we cannot trust they have no impact, all of the issues I have raised before will be present more or less and will cause problems (bigger/smaller/neglect-able?) 
there were no measurements with a proper LVPECL receiver so this can be tested but it would require designing a small receiver board. The models given by ERNI also show a change in matching getting worse with frequency even though the models do not incorporate most of the additional "disturbances"

In sum:
-  the ERNI connectors themselves will cut of the higher harmonics (increase rise/fall edges) acting as an LFP 
- phase imbalance (not matched lengths within pairs of the connectors,not compensated on RF-backplane) should be taken into account and the difference will show more for higher frequencies
-time of flight of the higher harmonics should be taken into account going through the backplane (again phase imbalance issues)
- mismatch increases with frequency, so one must be careful esp. broadband matching (which the LVPECL can be)
-temperature affects higher harmonics more (drifts)

This all can be tested but would require additional work using the precise drivers and receivers that we would want to use to check "is it good enough".

without precise measurements, we can keep it this way and leave the possibility to use one of the reference signals for clock generation/distribution using jumpers and we have of course the final fall-back option: the SMA

hartytp, 2016-10-28:
@sbouhabib Thanks for the update.

> If the analog channels on the backplane are not used for anything at the moment, it is a possibility to think about, a single line with a physical jumper is not much of additional work.(just a thought) even though the signal wouldn't be provided at the moment.

I'm not up to speed on uTCA.4 terminology, so let me check I understand what you mean here -- apologies if this is (re-)stating obvious things:

1. The RF backplane provides both differential "clock" lines from the clock module (eRTM#15) to the RTMs (2 pairs to each RTM). These use ERNI ERmetZD 973062 connectors & 100Ohm differential traces.
2. The RF backplane additionally provides single-ended "LO", "REF" & "CAL" signals from the clock module to the RTMs. These use RADIALL R694.251.027 (6GHz) connectors and 50Ohm single-ended traces.
3. Your comment is that the ERNI ERmetZD 973062 are not good at high-frequencies (they specify 10GBs data-rates, but that's not a good analogue specification!), which could lead to issues with fast clock rising edges.
4. As a fallback plan, we could distribute our 100MHz reference clock using the REF lines instead. 

Is that all correct? If so, what fanout buffer/distribution amplifier would you use to generate the clocks for the REF lines? And, would they be sine or square-wave? One option might be to use the PECL chips, along with baluns (TCM2-43X+ or similar) at either end.

sbouhabib, 2016-10-28:
@hartytp : all is correct, as for distribution we could do it this way, i think that it might be better for them to be sine waves, although the PECL chips tend to have really good jitter properties, one could use the fanout PECL chips with smaller bandwidth baluns and get the base frequency (with a few non-problematic harmonics). let me think more :) and analyze this for a while and answer after the weekend.
In general I am for your idea esp. that the balun can be substituted easily with one with a smaller BW, but I'd sleep on the thought till next week.

hartytp, 2016-10-28:
@sbouhabib Sounds good.

A couple of thoughts:

- square-sine conversion tends to be one of the hardest parts of clock distribution, with most ICs having lower phase noise/drift with square-wave inputs than sine inputs. So, I'd be keen to keep as many harmonics as possible in the waveform we distribute. 

- if we have any low-/high-pass filters in the clock distribution network (e.g. due to a low-bandwidth balun) then we'll have to think carefully about their propagation delay temperature coefficient.

jbqubit, 2016-10-28:
@sbouhabib Yes, thank you for looking into this. 

ERNI ERmetZD 973062 application note [0] reports eye diagrams for 5 Gb data rate. I can't find any performance specifications for RADIALL R694.251.027. Anybody find details on this? 

[0] http://www.erni.com/fileadmin/medien/downloadcenter/ermetzd/ERmet_ZD-Appl_note.pdf

jbqubit, 2016-10-28:
@sbouhabib "the clock distribution was tested using the LVPECL buffer HMC987 @81MHZ which has very fast edges,I got the phase noise results, i will give them a quick analysis as and get back to you" What are your phase noise results form this study?

jbqubit, 2016-10-31:
Resolved by #40.  

The microTCA.4 RF Backplane will be used for clock distribution. eRTM15 provides clock distribution to RTM slots 4 to 12. 
- input: SMA from front panel of eRTM15
  - 100 MHz sinusoid, single-ended, 50 Ohm, +13 dBm
    - this is a low phase noise oscillator is supplied by physicists
  - Sinusoid -> LVPECL conversion using LTC6957-1 
    - achieves -160 dBc/Hz at 10 KHz, chip-to-chip phase skew < 0.2 deg, tolerant of supply noise at 100 MHz 500 mVrms
  - @hartytp comment from #40 applies. "Another point to consider is that the sine-square conversion is inherently non-linear and thus tends to mix any low-frequency noise onto the clock. So, if one cares about close-in noise and spurs, then breaking ground-loops around the clock input is generally crucial. Again, this is probably best done using a balun on the mezzanine."
- output: RTM backplane distributes 100 MHz LVPECL to 8 RTM cards
  - routed to RTM slots 4 to 12 for use by Sayma
  - HMC7043 for clock distribution 
  - CLK, 100 Ohm differential LVPECL, ERNI ErMetZD connector
  - REF, 50 Ohm single-ended, RADIALL connector

closed 2016-10-31 by jbqubit