sinara-hw/sinara #28: test JESD204B waveform demo gateware

jbqubit, 2016-10-14:
m-labs is nearly complete with JESD204B waveform demo. https://github.com/m-labs/artiq/issues/580

Here are some tests I will do. 

**external clock** supplied to KC705 
- REF: 10 MHz oscillator reference
- use Linear Technology LTC6957-2 to generate low noise differential LVDS clock
- apply to KC705

**phase noise**
    - use Symmetricom 5125A
    - DAC outputs sinusoid at 100 MHz, 500 MHz
    - measure phase noise between pair of DAC channels
    - measure phase noise between REF and a DAC channel
    - this is a duplicate of measurements made by Slichter this spring

**long-time operation**
- look for programming and bit errors
    - set 100 MHz frequency repeatedly in ARTIQ Python program (long loop), compute phase offset to maintain continuous phase
    - between RF and a DAC channel...
        - frequency wander: look for frequency drift over time         
        - glitch: look for missed oscillation over interval 1000 s using counter
    - glitch: program 50% duty cycle square wave 0V, 1V at 10 MHz
        - trigger scope at 1.2 V to look for bad data word

**absolute phase**
    - confirm phase stability; write ARTIQ Python program generating experiment start pulse, with parallel: set phase of two RF channels; schedule program repeatedly on Master; trigger scope on start pulse; measure jitter 
    - above for RF tone
    - above for block-RAM waveform

- documentation & code
    - setup instructions
    - RTIO interface 
    - interface with ARTIQ Python

jbqubit, 2016-11-09:
Measure DAC0 to DAC1 phase noise. 

## Oscillators
OSC10a is free-running 10 MHz Wenzel (501-04609).
OSC100a is 100 MHz Wenzel (501-23588) locked to free-running 10 MHz Wenzel (501-04609). 
SYN2000a is ADF5355 PLL Eval Board is generating 2 GHz with OSC100a as reference 

## Measure
Measure DAC0 to DAC1 phase noise on AD9154 FMC Eval board using KC705 and Syn2000a. Use Symmetricom 5125A with two input ports (Ref and Input).  

[image]
(above, blue) Input: DAC CH0 100 MHz, Ref: DAC CH1 100 MHz
(above, pink) Input: DAC CH0 250 MHz, Ref: DAC CH1 250 MHz

[image]
(above) AD9154 data sheet (for comparison)

The channel-to-channel phase noise is less than the external-clock to channel phase noise. This is as expected. 

TODO: 
- [ ] Measure channel-to-clock phase noise. This requires a 2 GHz source with low phase noise. I've got one on order.

hartytp, 2017-07-16:
@jbqubit Has this been taken care of by the ARTIQ issues you've filed? Do we need to keep it open here?

jbqubit, 2017-07-16:
I've tested most of these -- checkmarks updated. Closing.

closed 2017-07-16 by jbqubit