sinara-hw/sinara #29: AD9154 input protection on Sayma RTM

jbqubit, 2016-10-14:
@dhslichter and @hartytp discussed this in email conversation 7/28/2016. Proposal was as follows. 

ADC protection on Sayma RTM
- input buffer (unity-gain amplifier) ADA4927  
- low-leakage protection diodes onto the input stage (e.g. BAV199) to protect the ADCs 

Questions:
- [ ] @gkasprow is this in the schematic?
- [ ] comments on this

jbqubit, 2016-10-14:
Recalling comment by @dhslichter in private email.

The AD9656 should have a 1.4V reference, common mode voltage of 0.9V, and can thus take 2.8 Vpp differential swings at its input.  This corresponds to VIN+=1.6, VIN-=0.2 and the reverse (i.e. each individual line can swing +/- 0.7 V around the common mode level).  Absolute maximum ratings are -0.3V and +2V for the device.  The ADA4927 will run at a gain of 1, and can swing within about 1V of the rails.  Thus if you want to have the ADA4927 limit things appropriately, it should run off of +/-2.5V supplies, and thus you will be safe.  The problem here is that we are now going to be running signals that come within ~1V of the rail, so you start to get this distortion issue again with high speed signals if you put protection diodes on the rails.  However, we may prefer to use +/- 5V rails, with protection diodes on the amp inputs, and then you have to put some attenuation in between the ADA4927 and the AD9656 to ensure safety.

My feeling here is that we should make the empty 0402 pads between amp and ADC such that one can use them to build an impedance-matched attenuation stage if desired.  However, the best thing would be to enforce maximum output levels at the analog daughtercard in the design process.

jbqubit, 2016-10-14:
@dhslichter "The issue below about grounding and ground loops has been resolved by the insulating washers for the front panel."

hartytp, 2016-10-16:
@jbqubit @dhslichter I'd prefer to put the protection diodes on the analog mezzanine, and leave the main Sayma board without protection (beyond that already offered by the ESD diodes in the amp/adc). That way, it's up to the mezzanine designer to decide on the appropriate way of providing protection for their application. e.g. if the mezzanine has an InAmp front end, then the protection diodes should go before the InAmp, rather than between the InAmp and buffer amp.

Similarly, RC pads for RFI filter, etc should go on the mezzanine -- again, they would want to be before the InAmp if one is used.

dhslichter, 2016-10-21:
- ADC need buffer amplifiers ADA4927 running from +/- 5V rails, low noise, gain 1 (R_F=R_G=301 ohm resistors, see datasheet Fig 46), output common mode voltage 0.9 V, with differential attenuation between output lines of ADA4927 to ensure that max (railed) ADA4927 output amplitude will give no more than 2.8 Vpp differential at ADC inputs (see AD9656 spec).  For +/- 5V supplies, ADA4927 puts out up to +/- 3.8 V on each line. Thus total differential swing is 3.8*4=15.2 Vpp.  Attenuator circuit should be differential between the two output traces, reduce amplitude from 15.2 Vpp to 2.8 Vpp, accounting for 2.6 kOhm differential input impedance of AD9656.  Ideal load impedance for ADA4927 closer to 1 kohm.
- use ADA4927-1 for reduced crosstalk
- no protection diodes on ADA4927 inputs or outputs
- circuit must be designed for dc coupling of signals to ADC

dhslichter, 2016-10-21:
from @gkasprow : 
> Why not to put the driver at the mezzanine? Some users may want to use a balun as ADC driver to get the best SFDR. 

This is also a possibility, but:

- would then require sourcing more current at the -12V rail (or more likely, a -6V rail) to the mezzanines
- allows us to provide input protection/level shifting to ADC without relying on mezzanine card to provide this -- simplifies task for mezzanine design
- ADA4927 has much lower harmonic distortion than AD9656 up until 300-400 MHz, when they become roughly equivalent (still low, around -70 dBc), so you are not really degrading the signal that much (if at all) by including it.  Any input signals above 62.5 MHz are going to be in higher Nyquist zones and need to be reconstructed anyway....so people who really care about direct signals at 300-400 MHz will not be getting a high-performance solution out of this device anyway, and would be better with a different DAC chip if they care about maximizing SNR etc.

If we place the ADA4927 on the mezzanine instead, we will need to supply a low noise -6V rail capable of ~100 mA per mezzanine to support 2x ADA4927-1.  I don't really like the idea of having to put a switching charge pump inverter on the mezzanines to create -6V at these relatively high currents, since it will introduce additional power supply noise.

dhslichter, 2016-10-21:
> @gkasprow says :  We can apply -6V rail to the mezzanine with limited current capability just to supply the driver chips. We need to have this voltage anyway.

> @hartytp says: I'm with Daniel on this one: I'd rather keep the ADA4927 on the main Sayma board to simplify the mezzanines.

dhslichter, 2016-10-21:
Performance considerations for including vs not including ADA4927 driver on Sayma board:

- what is noise added/SNR degradation/SFDR degradation from using ADA4927?
- should we attempt to include impedance-matching components for AD9656 on Sayma board?  Input impedance of switched-capacitor ADCs generally requires external impedance matching, can be difficult to make this broadband, depending on application.  Ideally would match for higher frequencies, at lower frequencies matching is less critical.
- transformer-coupled signals into AD9656 (no ADA4927) may well give better noise/spurious performance, but cannot pass dc
- can we use jumpers to bypass ADA4927 on Sayma if desired for different applications?  Would involve making jumpers like the red lines shown in the drawing below -- ultra-tiny solder job, or preferably with wire bonds (for non-soldered ENIG surfaces only).  Pink lines show chip boundaries.
[image]
- what is the maximum frequency for which one can put all matching/coupling components on mezzanine, with bare ADC only on Sayma, and still achieve good performance?  AD9656 eval board has all matching components as close as possible to ADC inputs, with transformers approx 35-40 mm away from ADC inputs.  
- typical demodulators have IF bandwidths of several hundred MHz (up to 1 GHz for LTC5586), so it would be possible to operate the ADC at or near its maximum analog bandwidth.
- predistortion/examination of AOM drive signals would involve direct digitization of narrowband signals at 80-350 MHz.

gkasprow, 2016-10-22:
Look at the image below.
The space occupied by two ADC drivers on the mezzanine is really tiny - comparable with one SMP connector providing that we mount components on both sides. So maybe the mezzanine real estate is not an issue here
[image]

dhslichter, 2016-10-24:
@gkasprow two questions:

1. What is the distance from the SMP connectors to the ADC inputs on the Sayma board?
2. If the ADA4927 chips are on the mezzanine, on the opposite side from the SMPs, will there be enough room to place a band-defining filter after them before the SMPs?

Unless the answers to these questions are problematic, I am thinking it would be better to keep the bare ADC on the Sayma RTM and move all driver circuitry (e.g. ADA4927) to the mezzanines.  Note that this is a change from what we tentatively agreed on on Friday.

jbqubit, 2016-10-25:
We've not discussed impact of phase noise on ADC SNR. 
- 1st Nyquist is DC to 62.5 MHz
- 3rd Nyquist is 125 Mhz to 187.5 MHz

Assume: ADC differential nonlinearity and input noise is 0. 
Focus: Given an signal frequency and desired resolution what is maximum tolerable clock jitter? 
Note: The actual SNR degradation is assuredly worse than this idealized model. 
```
    fa (MHz)   nbit    fs RMS 
        62.5     12       508
        62.5     15        64
       187.5     12       169
       187.5     15        21
```

Jitter below 200 fs is difficult to achieve in practice. The plot below shows jitter calculated from phase noise of several sources (10 Hz to 200 MHz). 
- wenzel_std_100 is a standard very good phase noise source 
- SDI_LNFR-100 is a superb dual-oscillators system from Spectra Dynamics Inc ($15k)
- HMC7044 shows phase noise under a particular (realistic?) configuration of the PLLs assuming a low-jitter reference
- LTC6957-1 shows added phase noise for a dual-output buffer amplifier, LVPECL outputs

[image]

See repository for sample oscillators and jitter calc.
https://github.com/m-labs/sayma/tree/master/design-calcs

gkasprow, 2016-10-25:
@gkasprow <https://github.com/gkasprow> two questions:
>
>    1. What is the distance from the SMP connectors to the ADC inputs on
>    the Sayma board?
>
> 3..4 centimetres

>
>    1. If the ADA4927 chips are on the mezzanine, on the opposite side
>    from the SMPs, will there be enough room to place a band-defining filter
>    after them before the SMPs?
>
> If it is just a capacitor and 2 resistors then yes.

>
>
> Unless the answers to these questions are problematic, I am thinking it
> would be better to keep the bare ADC on the Sayma RTM and move all driver
> circuitry (e.g. ADA4927) to the mezzanines. Note that this is a change from
> what we tentatively agreed on on Friday.
>
I already placed these drivers on Sayma.
There is another issue related to ADC protection.
Even if we attenuate the driver signal to the ADC levels, there is still a
danger.
The ADA can output +/- 3.8V while ADC can accept -0.3....2V
So with division rate of 2, we are safe from positive side but not from
negative side.

dhslichter, 2016-10-25:
We may be stuck with running Schottky diodes on both inputs.  According to analog devices, low-capacitance Schottky diodes can be used for ADC input protection without doing too much harm to the SFDR or SNR.  One has to choose RF Schottky diodes with the lowest possible capacitance to minimize nonlinearities.  Something like [SMS7621-060](http://www.digikey.com/product-detail/en/skyworks-solutions-inc/SMS7621-060/863-1171-1-ND/2349611) would work well and not take up too much extra space, since they need to be right next to the ADC.

sbourdeauducq, 2016-10-26:
In this case, is there really a reason for placing the drivers on the Sayma side?

dhslichter, 2016-10-26:
I tend to agree with @sbourdeauducq here; after looking at things in detail, and seeing that the ADA4927 chips can fit on the analog mezzanines in a compact way, I think it may make sense to just have low-capacitance Schottkys and some simple impedance-matching components (series R, differential C) on the Sayma board, with all other stuff on the mezzanines.  We will have to run a -6V rail at ~150 mA to the mezzanine.

gkasprow, 2016-10-26:
The -6V rail is already connected to the mezzanine.
What RC values of the filter (bandwidth) do you want?

dhslichter, 2016-10-28:
Action items to #38, closing issue.

closed 2016-10-28 by dhslichter