sinara-hw/sinara #30: analog mezzanine: RF output, 1st Nyquist

jbqubit, 2016-10-14:
Baseline analog mezzanine. 

## output
- [ ] high-level specification
    - 50 Ohm output, 13 dBm
    - [ ] __ bandwidth to 500 MHz
        - [ ] low frequency cut-off 
    - isolation of SMA barrel from ground  -- break ground loops
    - [ ] single-ended or differential output 
- applications
    - driving 200 MHz AOM with +/-20 MHz bandwidth
- components in sequence, 4 parallel identical channels -- DRAFT, not compatible with high-level specification
    - ATC 600S101JT250XT 100pf capacitor, AC coupling 
    - TC1-1-43X+ balun, differential to single-ended
    - MiniCircuits FV1206 filters
    - ERA-4x5M+ amplifier
    - filter block
    - ATC 600S101JT250XT 100 pf capacitor, AC coupling
    - attenuator properties
        - absorptive
        - high speed
        - eg HMC542BLP6 or DAT-31R5A-SP+
    - ATC 600S101JT250XT 100 pf capacitor, AC coupling
    - RF switch properties
        - absorptive
        - high speed -- how fast?
        - eg JSW2-63DR+ or CSWA2-63DR+

[image]

# input **ACTION ITEM** 
- [ ] high-level specification
    - user-selectable switch for intput termination
        - 50 Ohm to ground 
        - 0 Ohm to ground
    - voltage input
    - range +/- 5 V
    - [ ] single-ended or differential input
    - [ ] isolation of SMA barrel from ground  -- break ground loops
    - [ ] 150 MHz low-pass -- target 1st Nyquist zone only?

jmizrahi, 2016-10-16:
One small comment on the output spec -- our application requires at least
200 MHz +/- 30 MHz bandwidth (although I don't expect that to be a
limitation here).

dtcallcock, 2016-10-18:
Comments on the output stage:

**Low cutoff frequency**

I feel that going for 100kHz might mean quite a few compromises in the 100MHz regime.  Would frequencies from DC to a few MHz be better addressed with a different mezzanine that has something like an op-amp output?

**Filters**

The FV1206 footprint parts are quite high frequency.  The lowest f0 is 145MHz for LP, 1445MHz for BP and 440MHz for HP.  The other common Minicircuits footprint is GP731.  This is significantly larger (8.89x8.89x2.54mm vs 3.2x1.6x0.94mm) but the lowest f0 is 56MHz for LP, 75MHz for BP and 44MHz for HP.  Now though you are limited at high frequency as there are few parts of this footprint with f0 >300MHz.  Is there room for both footprints?  I looked around for other suppliers but nobody seems to have the range that Minicircuits does.

[FV1206 Package](http://www.minicircuits.com/case_styles/FV1206.pdf)
[GP731 Package](http://www.minicircuits.com/case_styles/GP731.pdf)

**Balun**

The TC1-1-43X part is 650-4000MHz

[Datasheet](https://www.minicircuits.com/pdfs/TC1-1-43X+.pdf)

There is a pretty equivalent part (TC1-1-13MX) that’s 4.5-3000MHz

[Datasheet](http://www.minicircuits.com/pdfs/TC1-1-13MX+.pdf)

Probably half a dozen others from Minicircuits that’d do the job.

**Power detector**

It would be good to have a power detector for calibrating the output using the slow DACs on the RTM.  I think the best place to put it would be instead of the 50 Ohm termination on the switch's RF1 output.  The only thing that would have to be independently calibrated is the difference in insertion loss of RF1 vs RF2 of the switch as a function of frequency. 

**Switch**

As discussed in #26, the CSWA63DR+ has poor settling times so I am looking for a replacement switch with the following specs:
50 Ohm
Absorptive SPDT
Single logic control pin
Single 3.3-5V positive supply
Fast (~100ns) switching and settling
High (>60dB) isolation

The table below shows what I came up with.

[image]

I might have to rethink if we want to go below 50MHz though.  I also wonder if we need such fast switching and high isolation if we have such tight control of the DAC output (these things are historically important when eg. driving an AOM from a synth)?  Relaxing these constraints to say 1us switching and 40dB isolation would give more options.

sbouhabib, 2016-10-19:
Hi all, 
the balun proposed as well as the 1-1-13 version are both 1:1 Ohm ratio baluns, the are are not differential to single ended and vice versa (just like that), it would be better to use a 1:2 ohm ratio, there are a few on minicircuits in general TC2xxx series( not TC1-1xxx) and others, one could check anaren also the have dielektric baluns and transformers but usualy for smaller bandwidths, the benefit is smaller size (sometimes a drawback) and lower drifts vs temeperature.

sbouhabib, 2016-10-19:
Has anyone simulated the chain? or part of it?

dhslichter, 2016-10-19:
Good point @dtcallcock -- we may not really need a switch for the baseband module given the DAC output noise performance (see datasheet Fig 17).  The amplifier will bring the power up somewhat above the thermal noise floor, but one could in theory switch in a large attenuation with the digital attenuator to get rid of drift over longer "off" periods.  The attenuator is good to dc -- it would be good to test the switches to see if the low frequency cutoff is set by the blocking cap values or by something internal to the switches.

dhslichter, 2016-10-19:
@sbouhabib in my experience with the AD9914 DDS chips, these 1:1 transmission line baluns provide a better impedance match (measured S11 looking in to the DDS output) at frequencies below 1 GHz, 100 ohms differential to 50 ohm single ended, than if I try to do the "correct" thing by making it 50 ohms differential to 50 ohms single ended.  S11 is in the -20 dB range up to ~1 GHz.  

I think the point is that you are right, technically one should use a 1:2 balun, but really we will need to actually test the circuits to see what works best (unless you have very good rf/microwave model for the AD9154 output stage, which I could not find).  

This card really needs to be able to do quasi-DC (~100 kHz) up to 500+ MHz, so we need a broadband balun.

dhslichter, 2016-10-19:
@dtcallcock you are correct, we would probably want the 1-1-13X (my drawing that Joe posted was for a microwave-frequency output, not baseband, thus has the wrong balun on it).

dhslichter, 2016-10-19:
@sbourdeauducq I renamed this "Curly" because "Allaki" in the current spec is an input-only card.  We have now changed the mezzanine spec so that mezzanines host both input and output on the same card.  We can either scrap all previous definitions and reuse the Allaki/Ulvidy/Argayash names, or just scrap the names and make new ones.  For now I want to avoid confusion since this is totally different from the hardware spec.

sbourdeauducq, 2016-10-19:
I would scrap the definitions and reuse the names.

dhslichter, 2016-10-19:
I agree on scrapping the definitions, have always hated the names and found them hard to remember/work with.

jordens, 2016-10-19:
If someone has problems remembering and working with "Allaki", I see no reason why "Curly" would be any different, other than NIH.

hartytp, 2016-10-19:
@jordens I'm not mad about either, but I'd still prefer "Curly" as it has an obvious spelling (e.g. if I tell a grad student the name, they can search for it without having to be told the spelling as well)

dhslichter, 2016-10-19:
See my comment at the bottom of #32 -- widely-known, phonetically spelled names where the first letter carries some information, would be better than random grab-bag names.

hartytp, 2016-10-19:
I'd like to use something along the lines of the output stage proposed here to:

1. Drive AOMs at ~85MHz - ~200MHz
2. Produce microwaves for applications that require large bandwidth, but aren't overly sensitive to phase noise, and can hence use the AD9154 in mixer mode (e.g. state preparation for 43Ca+ at 288G requires 2.6GHz-3.2GHz)

@dtcallcock What sets the high-frequency cutoff for this design? It looks like it should be good from >~10MHz to >~3.6GHz with suitable choice of pin-compatible balun/filters, and so should cover all use-cases that don't require up-conversion. Or, am I missing something?

allcockd, 2016-10-19:
@hartytp Yeah, as I stated in #32, I think only the component selection will limit the high-frequency cutoff.  I don't think choosing components good to 3.6GHz will be a problem.

closed 2016-10-28 by dhslichter

sbouhabib, 2016-11-28:
I tried to upload the schematics for the Mezzanine board for discussion, hope the issue will be resolved now

here are some issues for the discussion:

-is the name final, i.e. "Curly"? or some "name-says-it-all"?
-does the design have to fullfill the "curly" req. only or as @hartytp suggested to add an up-converting possibility?or anything else?
-Are there any monitoring needs (usage for Slow ADC channels, or GPIOs) i.e. temperature measurements etc...
-do I add an "identifier" i.e. eeprom, this would help identifing the type of mezz used and turning on LO signals or such
-The LDO discussion seems to just have ended, any more comments on the matter?
- DO I add custom (DNP or 0R resistors/caps) i pi/T formation for matching inputs and outputs, i think this could help for differnet applications
-DAC channel:
--RF switch choice finalized?
--the aforementioned up-conversion possibility
-ADC channel:
--in general: Input requirements finalized?
--programmable/fixed attenuation?
--filter block?

dhslichter, 2016-11-28:
- name should not be "curly" - can use "allaki"
- the design is only for baseband output, no upconversion
- temperature measurements and power detection circuitry to slow ADCs for power drift calibrations will be necessary
- will want to have an EEPROM containing mezzanine information to be read by AMC FPGA
- want to have footprints for band-defining filters in various locations, likely populated DNP and 0-ohm resistors

hartytp, 2016-11-28:
@gkasprow Can you add the same EEPROM to the clock mezzanine template you provided, please?

gkasprow, 2016-11-28:
It is already there from very beginning: 24AA02E48T-I/OT

hartytp, 2016-11-28:
@gkasprow Of course, sorry I missed that!

jbqubit, 2016-11-28:
This issue is closed. For design review see #79.