sinara-hw/sinara #31: debug connector

sbourdeauducq, 2016-10-18:
I propose a simple 2.54mm 3.3V header that can be connected to a USB/Serial adapter for basic debugging tasks. Just two wires (rx/tx) would be enough.

sbourdeauducq, 2016-10-18:
And should we have a direct a JTAG connector on the board or will this go through standard MicroTCA pins (only)?

jordens, 2016-10-18:
IIRC in Warsaw we wanted to slap a 4 channel FTDI thing on there with the channels doing (a) fpga jtag (b) fpga console (c), (d) doing some MTC stuff. That gets rid of having to find usb serial adapters and jtag adapters, it can be exposed on the panel, and it is a nicer connector.

sbourdeauducq, 2016-10-18:
OK, that would be fine too.

gkasprow, 2016-10-18:
I always place some header with IO pins around FPGA for debugging purposes :)

gkasprow, 2016-10-18:
I  placed standard Xilinx JTAG connector on board for ones who want to use standard Xilinx cable.

jordens, 2016-10-18:
sounds fine. i don't think we will touch that if we have the ftdi interface.

closed 2016-10-22 by jbqubit