sinara-hw/sinara #33: move to public repository

jbqubit, 2016-10-22:
@gkasprow WUT starts using github as your primary repository for layout, routing, modeling, etc. This replaces use of private WUT git repository.

gkasprow, 2016-10-24:
Completed schematics of Sayma RTM are uploaded. Pdf version is here
https://github.com/m-labs/artiq-hardware/blob/master/ARTIQ_EE/Sayma_RTM.pdf

gkasprow, 2016-10-25:
I placed the repo here
https://github.com/m-labs/artiq-hardware
It is common design for Metlino and Sayma

>

jmizrahi, 2016-10-26:
On page 10, the SYNCOUT lines are not connected. Don't these signals need to be sent back to the FPGA as part of the communication protocol, and to enable deterministic latency?
@gkasprow @jordens

gkasprow, 2016-10-26:
I have only 3 free pairs on RTM connector.
The same probably applies to ADC DSYNC
Am I rignt @jordens <https://github.com/jordens> ?

jordens, 2016-10-26:
A bit rambling about using this issue tracker efficiently:

* @jmizrahi this is the wrong place to bring up schematics review items. Please file an issue next time. #34
* This is an issue about putting something in the repository. Either @jbqubit or @gkasprow should have closed it when it was resolved. Is it?
* If the schematics are in the artiq-hardware repository, then issues need to go there. Otherwise it falls apart.
* If it is really impossoble or inconvenient enough to split the metlino and sayma designs into separate repositories (@gkasprow is it?), then we should use this repository (and rename it).
* Are we doing schematics review now? Should we organize that?

gkasprow, 2016-10-26:
The Sayma RTM schematic is ready for review.
The Sayma AMC schematic is still in development, we will finish it in a few
days. It took longer because we had to make additional symbols.
The Mentor Expedition tool very strongly supports design reuse so I keep
all related pieces of hardware in one project.
It is also very easy to copy pieces of verified layout from one design to
another together with schematic blocks.
So from design flow point of view it's easier to have them in common
project.
But I can place dedicated HW versions in right repositories

jmizrahi, 2016-10-27:
@jordens Got it, I will do that in the future.

Perhaps it would make more sense to merge artiq-hardware, sayma, and metlino into a single repository? Is it really structurally more efficient to make these separate repositories, as opposed to being entirely under the artiq-hardware project? (Especially given the overlap that Greg is talking about)

jordens, 2016-10-27:
It is more efficient if they are developed separately. Then they can be tagged and released separately and it would be clear what an issue is actually affecting. It becomes unmanageable if everything is thrown into one repository where neither the submitters, nor the assignees, nor the commenters take even minimal care of tagging, triaging, and handling issues.
IMHO it could even be wise to split the hardware development into "hardware-libraries", "sayma", and "metlino" then the hardware libraries would be used in the two hardware projects. Only if they are developed very closely, I can see this becoming inconvenient.
I see no reason to put the designs into the artiq-hardware repository. That doesn't make any sense.
If the consensus is using one repository for the designs, then use this one, rename it to "sinara" and roll back "artiq-hardware" to be the place where we write the whitepaper.

jordens, 2016-10-27:
@gkasprow *you* need to be the one that closes this issue if it is resolved (and the other issues where you feel they have been resolved).

gkasprow, 2016-10-27:
I’m convinced.

Tagging is powerful feature and I can always merge the two designs to get one.

You can remove all the stuff from artiq-hardware repo.

jordens, 2016-10-27:
@gkasprow ;) convinced of what? I do see the advantage of having the library, Sayma, and Metlino all three in one repo. We can do that. It's just that if we do it, it should be *this* repo (i.e. I would rename `m-labs/sayma` -> `m-labs/sinara`) because the issues are here.

gkasprow, 2016-10-27:
@jordens <https://github.com/jordens>
I see advantage of separate development of boards. I can place designs in
repositories linked to Sayma, Metlino, etc. Such approach makes issue
tracking easier.
But Sayma AMC and Metlino share 80% of blocks so it makes a lot of sense to
keep them in same repo. In this way some bugs fixed in one board will
automatically be transferred to another.
In case of Sayma RTM and Sayma share only 1 block so we can split them and
place in dedicated repositories.
btw, how to roll back changes in github repo? I use tortoise git under
windows and can rollback my local files to any version but don't want to
make mess on gthub. I can roll back changes only to my first commit where
the HW files were placed.
<https://github.com/jordens>

jordens, 2016-10-27:
Then lets' throw all the hardware designs into this (`m-labs/sinara` f.k.a. `m-labs/sayma`) repository.
One would do a properly parametrized `git reset` or `git checkout` depending on the exact goal and then `git push --force`. But don't do that lightly.
I can clean up the `artiq-hardware` repository once you are done there.

jbqubit, 2016-10-28:
@jordens I agree. Keep the schematics in the same repository as all the Issues.

closed 2016-10-28 by jordens