sinara-hw/sinara #34: SYNCOUT

jordens, 2016-10-26:
@jmizrahi from #33:
On page 10, the SYNCOUT lines are not connected. Don't these signals need to be sent back to the FPGA as part of the communication protocol, and to enable deterministic latency?
@gkasprow @jordens

jordens, 2016-10-26:
Yes. SYNCOUT0 needs to go the the AMC FPGA.
@gkasprow If there is pin shortage, let me know. We don't necessarily need a bunch of the others.

gkasprow, 2016-10-26:
Fixed, there is one diff pair left on the RTM connector
What about DSYNC of ADC?

jordens, 2016-10-26:
All four (2xSYNCOUT0 from the DACs and 2xDSYNC to the ADCs) need to go to the AMC FPGA.

gkasprow, 2016-10-26:
But ADC DSYNC can be common to both ADCs?

jordens, 2016-10-26:
I'd rather not do that but let me check.

gkasprow, 2016-10-26:
I can do that but both FPGAs would be connected with 2 GTPs instead of 4.

jordens, 2016-10-27:
@gkasprow Logically DSYNC can be common. But then you need match the impedances with fan-out or 200 Ohm differential lines and 200 Ohm terminations.

jordens, 2016-10-27:
Hint: @gkasprow if you move the hardware designs here, you can just say "closes #34" in the commit message and this issue will [automatically be closed](https://help.github.com/articles/closing-issues-via-commit-messages/).

gkasprow, 2016-10-27:
Fixed, there is one diff pair left on the RTM connector
> What about DSYNC of ADC?
>
@jordens <https://github.com/jordens>, you probably need to somehow keep
the ADCs in sync together. I can connect both DSYNC using single LVDS to
the AMC FPGA or to the RTM FPGA.

jordens, 2016-10-27:
@gkasprow How do you want to do that? Using 200 Ohm terminations near the ADCs and connect them in parallel with two 200 Ohm diff lines or do you want to use an active fan-out? Or, do you want to use the RTM FPGA as a fan-out? This signal is a bit timing critical and its electrical length should not be too different from the JESD links between the ADCs and the AMC FPGA.

gkasprow, 2016-10-27:
I already connected them separately to the AMC FPGA
So there is no issue here anymore.

>

jordens, 2016-10-27:
@gkasprow excellent! Would you like to close this issue then? The commit you did was in the other repository (`artiq-hardware`): there is no issue #34 there.

closed 2016-10-27 by gkasprow