sinara-hw/sinara #36: width/rtm

jordens, 2016-10-28:
_From @jordens on September 29, 2016 16:51_

Usually an MCH is single-width and would thus not be able to connect to "its" RTM.
The root Metlino needs a reference frequency -- 100 MHz -- to seed the DRTIO timing tree. Should we make Metlino double-width and somehow connect it to its RTM (that also drives the rf backplane clocking and thus has the reference frequency -- 100 MHz -- on it)? Or should the root Metlino always get its reference through its AMC-side panel? The latter would have the advantage that root/satellite Metlinos do not already require a double-width rack with RTMs and rf backplane and would also work in other racks with smaller hardware.

@jbqubit @gkasprow @jmizrahi 

_Copied from original issue: m-labs/metlino#2_

jordens, 2016-10-28:
_From @gkasprow on September 29, 2016 17:23_

MTCA.4 crate has double width MCH slot that accepts both single and double width MCH modules
Metlino is double width because we need plenty of space on the panel for VHDCI connectors, SFPs and FMC for extensions. We _may_ use RTM in the future, but not necessarily now.
MCH has simple clock module that delivers 100MHz clock to both Metlino and all AMCs.
The MCH has 2 SMAs connected directly to the clock crosspoint switch so both Metlino and all AMCs can have same clock.
In this scenario we may deliver the clock from the front panel.
In the future we will make Metlino RTM with additional SFPs and whatever is needed there, so we can use the clock delivered via  the RF backplane.

jordens, 2016-10-28:
It's not clear to me what we need that 100 MHz clock module on Metlino and the 100 MHz clock distribution for. Every Sayma can and will get an excellent clock from their RTM. And for all kinds of booting, testing, and stand-alone operation, we'd want a simple oscillator on Sayma anyway.

jordens, 2016-10-28:
_From @sbourdeauducq on October 2, 2016 7:23_

> We may use RTM in the future, but not necessarily now.

As per my email: 
https://ssl.serverraum.org/lists-archive/artiq/2016-September/001007.html
it appears necessary to transmit the same 100MHz clock on both RTM (RF backplane) and AMC sides. The actual 100MHz on AMC side is only necessary for the root Metlino, the other cards recover it from the serial links.

gkasprow, 2016-10-28:
In uRFB-enabled  MTCA.4 crate the MCH is double width and also has RTM that connects to the RF backplane

The MCH-RTM-BM module has 3 LVDS lines routed to the eRTM-15 clock distribution box.

SO the same 100MHz clock that is routed to all the AMC-RTMs can also be routed to the Metlino via RTM backplane. In this way we can route 3 clocks. For applications without uRFB we will provide SMA connector on Metlino front panel.

jbqubit, 2016-10-29:
The 100 MHz on the AMC backplane may be used by AMC cards that don't extend to RTM. This is looking toward future AMC cards not yet developed.

jbqubit, 2016-10-31:
@jordens This looks resolved.

closed 2016-10-31 by jbqubit

jordens, 2016-10-31:
yep