sinara-hw/sinara #38: AD9656 input protection diodes

dhslichter, 2016-10-28:
Use Schottky diodes for input protection on AD9656.  For each input, want diodes both to ground and +1.8V supply rail.  Suggest [HSMS-286K-BLKG](http://www.digikey.com/product-detail/en/broadcom-limited/HSMS-286K-BLKG/516-1938-ND/1233685), which has low forward voltage and tolerates up to 4V reverse bias.  Anodes connected to signal lines, cathodes connected to GND and +1.8V.

jordens, 2016-10-28:
What about ESD protection (i.e. some nice TVS)? Do we leave that to the mezzanine?

hartytp, 2016-10-28:
I'd leave that to the mezzanine -- there are going to be relatively few cases where the input signal is coupled directly into the ADC, so protection will be needed upstream...

jordens, 2016-10-28:
@hartytp ack
@dhslichter the lower diode needs flipping...

dhslichter, 2016-10-28:
TVS is a disaster from a distortion standpoint, so we leave that decision to the mezzanine designers.  @jordens good catch on diode directions, have updated top comment to fix.

dhslichter, 2016-10-28:
Further information on TVS vs Schottky for ADC input protection: http://www.analog.com/media/en/analog-dialogue/volume-49/number-2/articles/rf-samp-adc-input-protection.pdf

dhslichter, 2016-10-28:
@sbouhabib please add any comments/thoughts here.

closed 2016-11-04 by gkasprow