sinara-hw/sinara #41: Sayma AMC clock distribution

gkasprow, 2016-10-29:
@jordens, @sbourdeauducq 
how to generate GTP clock for RTM FPGA? It is routed to Sayma AMC. Shall I split one of the outputs of Si5324.
The same applies to 3 clocks from AMC connector. Are there any plans to use them? Maybe as a reference for Si5324
What about DDR clock? On devkit it is generated from SI5335.
In our case I connected single ended 50MHz oscillator to one of the FPGA global clock inputs. Is it sufficient to drive the DDR controller?

sbourdeauducq, 2016-10-30:
Which GTP? The one for DRTIO/backplane? It is generated by the Si5324, we will use its automatically switched freerun mode. The reference for the Si5324, when it is not in freerun mode, is the recovered GTP clock (see #4). You can add a 200MHz XO on another transceiver clock pin, but in theory that is not necessary. We can have it for development/debug and DNP it later.
A simple XO on a global clock input is enough for the DDR.

gkasprow, 2016-10-30:
@sbourdeauducq please confirm
So actually we don't need a clock routed from AMC FPGA to the RTM FPGA becauses it will be recovered anyway on RTM by Si5324.
The Si5324 on AMC is routed on the same manner as on RTM FPGA - to one of transceiver clock pins and to global FPGA clock / IO pin.
What about clocks from AMC backplane that could be optionally generated by Metlino clock module?

sbourdeauducq, 2016-10-30:
I am confused. Are you talking about the Artix-7 FPGA (but then why do you bring up DDR?) or the Ultrascale FPGA?

sbourdeauducq, 2016-10-30:
The Ultrascale needs to be have clocks as per https://ssl.serverraum.org/lists-archive/artiq/2016-September/001007.html. The Artix-7 synchronizes itself via its transceiver link with the Ultrascale and does not need further sync/clock signals with the rest of the system.

gkasprow, 2016-10-30:
Sayma AMC FPGA = UltraScale
Sayma RTM FPGA = Artix

sbourdeauducq, 2016-10-30:
OK. After reading the above, is there anything which remains unclear to you?

sbourdeauducq, 2016-10-30:
The "3 clocks from AMC connector" are clock pins that are in the MicroTCA standard and connect to the AMC side of Sayma, correct? In theory we won't need them, but you can always connect them to FPGA pins (clock capable general purpose or transceiver reference) if that's easy.

gkasprow, 2016-10-30:
The only unclear thing is utilisation of the AMC clocks. Do we keep them
unconnected or should I route them to the FPGA global clock or SI chip
input.

sbourdeauducq, 2016-10-30:
Route them to the FPGA. It can forward them to the Si chip if needed. Those connections are very low priority as theoretically we will not use them.

jbqubit, 2016-10-31:
Why isn't Sayma AMC FPGA and Sayma RTM FPGA derived from clock mezzanine?

sbourdeauducq, 2016-10-31:
Because then there are two conflicting clocks (transceiver recovered clock and clock direct from the backplane), with clock domain crossings that are difficult to do in a latency-deterministic way.

jordens, 2016-10-31:
@jbqubit in short, and as explained numerous times in detail, that would break absolute timing.

closed 2016-10-31 by jordens

gkasprow, 2016-11-01:
The AMC FPGA is clocked by HMC chip directly (GTX Tx clk, GTX Rx clk, global clock).

The RTM FPGA is also clocked by HMC chip (FPGA_REF_CLK)

sbourdeauducq, 2016-11-02:
> The RTM FPGA is also clocked by HMC chip (FPGA_REF_CLK)

That's not necessary, but if we have spare pins OK.