sinara-hw/sinara #42: JESD204B Subclass 1, plan for Sinara

jbqubit, 2016-10-31:
@jordens @sbourdeauducq Please detail your plan for obtaining synchronization of DAC and ADC using the JESD204B Subclass 1 approach. The last detailed discussion of this was on the mailing list in [September](https://ssl.serverraum.org/lists-archive/artiq/2016-September/001009.html). 

 What is synchronization approach for 
- [ ] DACs on the same Syama card
- [ ] DACs spanning multiple Sayma cards in same chassis
- [ ] ADCs on the same Sayma card
- [ ] ADCs spanning multiple Sayma cards in the same chassis -- is this supported? 

Questions:
- [ ] RFSYNC is required for DAC synchronization. In current version of Sayma_RTM schematic REFSYNCIN is routed to RF backplane ERNI connector. Is the intent to generate and distribute REFSYNC from eRTM15? LVPECL? HMC7043?  
- [ ] Metlino FPGA will ultimately be responsible for synchronization.  Currently FPGA_DAC_SYSREF and FPGA_ADC_SYSREF are routed across RTM connector to Sayma FPGA. Consider also broadcasting FPGA_DAC_SYSREF to Metlino FPGA acrossAMC backplane. Fatpipe Ports 6 and 7 could be used as they're not currently allocated. Do this using a clock driver -- not via Sayma FPGA. This provides for a Subclass 1 synchronization mechanism from a single point (Metlino FPGA) and doesn't rely on availability of DRTIO.

sbourdeauducq, 2016-10-31:
Have you read and understood https://ssl.serverraum.org/lists-archive/artiq/2016-September/001007.html "JESD204 synchronization procedure" ?

jordens, 2016-10-31:
@jbqubit It is really all in that e-mail.

1. RFSYNCIN is not required. Why would it be? The rest of your first question is answered in #40 
2. Why is the Metlino FPGA responsible for sync? What is that FPGA supposed do with those signals (two from each Sayma)? I don't understand what problem you are trying to solve and how that solution is supposed to work. It's also not a question.

jordens, 2016-11-11:
https://github.com/m-labs/sinara/wiki/SinaraClocking

closed 2016-11-11 by jordens

jbqubit, 2016-11-11:
Thank you for posting this Robert.

-------
Joe Britton
Sensors and Electron Devices
Army Research Lab
2800 Powder Mill Rd
Adelphi, MD 20783
301-394-3130
joseph.w.britton5.civ@mail.mil