sinara-hw/sinara #43: clock mezzanine HF design to github

jbqubit, 2016-10-31:
@hartytp @WeiDaZhang
- [ ] add layout to github
- [ ] what is simulated phase noise? 
Be aware of #39.

jbqubit, 2016-11-08:
ping @hartytp @WeiDaZhang

hartytp, 2016-11-08:
@jbqubit sorry, busy few days. Will do in a mo...

hartytp, 2016-11-09:
@jbqubit in terms of checking the size for the clock mezzanine:
- The various PLL ICs we're looking at are all no larger than 7mm x 7mm
- My preferred choice of VCO for the final mezzanine would be the Crystek CVCO55 family. This family has a wide selection of frequency choices with superb phase noise. For 2.4GHz, we'd use CVCO55CC-2400-2400, which is available in singles from Digi-Key @ $25.77000 (which isn't a lot of EUR/GBP any more). VCO size is 12.7mm x 12.7mm, 5.59mm height, making it the largest thing on the board.
- There are a few additional components, such as baluns, regulators & the loop filter.
- We should leave enough room for 2 PLLs (e.g. DAC clock + LO)

gkasprow, 2016-11-22:
I published the clock module template. 
There are: zipped stp file, dxf drawing, 3D pdf and Altium sources. SMP placement facilitates RTM routing but this can be changed if makes any difficulties on mezzanine routing.

hartytp, 2016-11-22:
Thanks for that @gkasprow,

- What is the maximum component height on the board top side (not the side with the SMPs)? Will the 5.6mm VCO height be okay?
- What is the plan for screening enclosures on the clock/AFE mezzanines? Will the 5.6mm VCO fit under the screening enclosure?

gkasprow, 2016-11-23:
The height constraints are identical as on analog mezzanines.
It was discussed in one of the issues.
The absolute maximum ix 4.6mm. So the VCO would have to be installed on
bottom side of the clock mezzanine. Anyway, since SMP connectors are there
it is recommended to put all RF stuff on the same side to eliminate vias.
I don't plan to put any components under the clock mezzanine so you can use
almost all available height which is 9mm

hartytp, 2016-11-23:
@gkasprow The latest I found was on #8:

- mezzanine board 32 x 100 mm, .062" nominal thickness, 4 layer boards
- top and bottom layers to be Rogers 4350B, .020" thickness (cap construction for PCBs)
- 50 ohm CPWG on .020" Rogers 4350B is .035" trace with .011" gaps
- need ground stitch vias every ~.050" or less along both sides of CPWG
- mezzanines are mounted to RTM motherboard using 4x M1.5 screws
- 9 mm stand off between motherboard and mezzanine
- RF shields on both sides, separate cavities for each DAC channel and ADC channel (4 total per mezzanine) interior height under shield lids needs to be .170" minimum (allowing for fabrication tolerances) to accommodate baluns

Is that still correct? Also, has a decsion been made about made about what brand/series of screening to use yet?

9mm should be enough height for a 5.6mm VCO + screening.

gkasprow, 2016-11-23:
One comment - 1.5mm screw are difficult to handle.
Let's use M2.5 screws. They are used with FMC boards.

jbqubit, 2016-11-28:
Phase noise supplied in #75

closed 2016-11-28 by jbqubit