sinara-hw/sinara #44: [schematics] add proper page frames with metadata

jordens, 2016-11-01:
* page numbers
* identifiers (git commit hash and date)
* names for pages
* copyright
* license
* The href boxes (links) in the PDF should be invisible. They are massively cluttering the PDF in all my PDF readers.

gkasprow, 2016-11-01:
@jordens 
where to find in a directory structure "git commit hash" value?
I can make automatic sheet page frame update but need to take this hash value from somewhere, preferably from file.

sbourdeauducq, 2016-11-02:
Run ``git log -1 --format="%h %cd" HEAD``

jordens, 2016-11-02:
@gkasprow use a [filter](https://git-scm.com/docs/gitattributes) that writes e.g.. `git describe --tags --always` into a file.

gkasprow, 2016-11-03:
@jordens, @sbourdeauducq 
There is a little problem with correct git hash value.
I  do commit, then get hash, update schematics with it and do commit again.
But the hash is different and always reflect previous commit.
Wouldn't just the date be enough?

jordens, 2016-11-03:
Yes. You can't do what I think you are trying to do. The result would be undefined.
When doing this properly, the PDFs would also not be in the repository because again, they are generated from the "source" files. I'd leave it to you to choose what you prefer:

a) upload the PDFs as "releases" https://github.com/m-labs/sinara/releases
b) keep them in the repository with the "previous" commit in them.
c) just use an ISO 8601 date-time stamp.

closed 2016-11-14 by jordens