sinara-hw/sinara #45: Sayma RTM/AMC schematics review

jordens, 2016-11-01:
I'll give managing this reivew a shot.
Go to https://github.com/m-labs/sinara/wiki/ReviewSaymaRTM and follow the instructions.
Will close this issue when the review is done.

michallgaska, 2016-11-02:
Missing ground connections at sheet FPGA_XCKU040FFVA1156(FPGA Decoupling)

gkasprow, 2016-11-07:
I committed newest version with reference designators and correct symbols which are paired with footprints. I did not fix bugs. Now project passes ERC, compiles so I will start component placement.

jbqubit, 2016-11-08:
What type of tests are included in ERC?

jbqubit, 2016-11-08:
@Jordens There's not been much activity on this approach to coordinating a design review. By my eye it doesn't provide enough structure or accounting. Please discuss with Greg how to divvy up the job and make use of e.g. Issues to increase activity.

gkasprow, 2016-11-08:
> What type of tests are included in ERC?
>
>
> it depends how one configures it.
- power distribution, local/global nets
- hanging nets
- bus width and bus signals
- missing ports
- signals connected to power ports
- outputs shorted
- undriven inputs
and many, many others
I just passed very basic tests. Once the schematics are frozen, I will do
full diagnostics and cleanup things like aesthetic fonts, designator
placement, line widths..

jordens, 2016-11-08:
@jbqubit That's an unacceptably vague and nebulous comment. By my eye the low activity means either "everybody is lazy or has no time" or "everything is good".
We have had the schematics to do the review since yesterday.
Please also give me a list of people how are reviewing something (other than WUT and M-Labs).

hartytp, 2016-11-09:
@jordens the version of the schematics you link to on the wiki is now out of date.

jordens, 2016-11-09:
@hartytp edited.

jbqubit, 2017-01-30:
Commit ID: 06928c8 is now far in the past. Closing this Issue.

closed 2017-01-30 by jbqubit