sinara-hw/sinara #46: clean up repository

jordens, 2016-11-01:
@gkasprow 

* With more than a GB this is unmanageably large.
* There are are a bunch of `*.raw` files consuming space. I assume they are the raw output of the simulation. Those shouldn't be committed to the repository and `dgitignore`d.
* `*.tmp` `*.old` `si_cache` `*.bak` `BackupDatabase` `ProjectBackup` `Layout_Temp` `Greg` all look like they should be removed from the repository (not your working tree through) and added to `.gitignore` as well. There are probably many more.
* I am pretty sure you don't want to commit `Work` or `vbreport` either. Please commit only actual "source" files, the ones that you edit, not the ones that the tools (re)generate and scatter around.
* The problem is that once committed they stay in the repository's history. You could have gotten rid of them before doing a fresh commit to this repository. Or now you can get rid of them in the history with some creative use if `git filter-branch`. But in any case, please don't commit any changes to those files, i.e. gitignore them and remove them from the repository.

gkasprow, 2016-11-01:
@jordens 
Could you just remove all the files I added to the repo?
I will commit them once again without the history

jordens, 2016-11-01:
Sure. Done. Let me know if you need help. I am on IRC (freenode/#m-labs).

gkasprow, 2016-11-01:
Looks much better now.

closed 2016-11-01 by gkasprow