sinara-hw/sinara #49: Sayma RTM FPGA debugging

sbourdeauducq, 2016-11-02:
I propose that the second UART channel (currently "SoC") of the FTDI chip on AMC goes directly to a small header that can be connected to the RTM FPGA via an additional cable when the AMC/RTM is outside the crate. This can come useful to debug RTM FPGA transceiver problems.

gkasprow, 2016-11-02:
We have 2 spare line in RTM connector. They can be used as UART or reserved
for future needs.

>

sbourdeauducq, 2016-11-03:
Sounds good, in that case connect them to the FPGAs on each side and keep the two UART connections between the Ultrascale and the FTDI-chip. The Ultrascale could trivially connect a UART to the RTM and this gives a bit more flexibility.

gkasprow, 2016-11-03:
OK, I will route them as diff pair.
For UART it doesnt make any difference but can be used for other higher speed purpose.

closed 2016-11-03 by gkasprow