sinara-hw/sinara #50: LDO for ADCs and DACs

sbouhabib, 2016-11-04:
what are the LDOs used for powering the ADCs and DACs (most of the names are not present on the schematics), these should be chosen for best noise performance,
@gkasprow could you check and post, and then maybe we can recommend sth. or leave them be.

dhslichter, 2016-11-04:
I recommend the ADM7154 as giving very low noise at high currents with low dropout voltages; for +5V rails, use the ADM7150.

jbqubit, 2016-11-08:
Has LDO choice been finalized?

-------
Joe Britton
Sensors and Electron Devices
Army Research Lab
2800 Powder Mill Rd
Adelphi, MD 20783
301-394-3130
joseph.w.britton5.civ@mail.mil

gkasprow, 2016-11-08:
nope, I didn't wanted to modyfy the design files while review takes place.
Just add an issue.

WeiDaZhang, 2016-11-09:
Some noise performance comparison among LDOs.
The curves are extracted from each datasheet.
[image]

dhslichter, 2016-11-09:
It looks like the LT3042 is a good choice.  PSRR is also generally a bit higher for the LT3042 than for the ADM7154.  However, there are some additional subtleties to consider -- see the comments below:

- The ADM7154 noise spec here assumes a 1 uF C_byp.  Choosing larger values of C_byp can dramatically reduce the noise in the range between 1 Hz and 1 kHz; choosing 100 uF C_byp, for example, brings the noise level down close to that of the LT3042.  It is possible to get capacitors this large in very small packages (here's an [0603 100 uF 6.3V tantalum](http://www.digikey.com/product-detail/en/avx-corporation/F980J107MMAAXE/478-9836-1-ND/5957637), that one would parallel with a small ceramic to improve high frequency performance).  See below:

[image]

- The SET capacitance on the LT3042 is 22 uF to achieve the noise given above, suggested to be ceramic X7R with high voltage rating.

- LT3042 will require precision, very-low-tempco set resistors.

- Maximum output current for the LT3042 is 200 mA; this means that for some of the rails (e.g. AD9656 AVDD, AD9154 SVDD12) multiple regulators would need to be paralleled.  The datasheet gives instructions on how this can be done.  It may be advantageous to use ADM7154 (can source up to 800 mA) for e.g. SVDD12 rails to reduce footprint/complexity (this rail would require three or four LT3042 per DAC chip, for example), and save the LT3042 for the most demanding rails, noise-wise.  Note that the maximum JESD204B data rate depends on the SVDD12 rail -- to do 1 GHz DAC clock without interpolation we require 1.3V, not 1.2 V -- thus we may prefer an adjustable regulator and not a fixed one like ADM7154.

gkasprow, 2016-11-28:
I used yet another LDOs which satisfy all needs.
LT3045 has same noise characteristics, PSRR, pinout as LT3042 but output current is 500mA.
And it works from 1.8V so I can generate 1.3V from 1.8...2V easily.
With ADM7154 I'd have to use 2.3V at least so it would generate much more losses.

closed 2016-11-28 by gkasprow