sinara-hw/sinara #51: clock synth on Sayma RTM

jbqubit, 2016-11-08:
As discussed in #40. Consider this feature for future version of hardware. 

>@hartytp IMHO, the purpose of putting a synth (PLL + VCO) on the RTM is to allow the commonly required DAC clock frequencies to be produced with respectable phase noise, without a clock mezzanine. At a minimum, this means that the RTM synth should be able to produce: 1GHz (max DAC clock for 1x interpolation); 2 GHz (max for x2 interpolation) and 2.4GHz (max DAC clock frequency). If it doesn't cover these frequencies then it's likely that we'll end up having to build and support a broad-band, medium performance version of the clock mezzanine in addition to the the high-performance, narrow-band mezzanine. In my view, if we have to do this anyway, there isn't much point having a synth on the RTM as well.
>
>I think that a good case has been made for having a broad-band synth on the RTM, but I'm not totally convinced that the 7044 is necessarily the best IC for the job. Why not use a HMC7043 in conjunction with a separate synth IC? It would be nice to decouple the JESD204B functionality from the clock synthesis and, adding an extra IC doesn't add much extra complexity/cost of the design.
>
>A good choice of synth might be the HMC830, as this is a relatively high-performance synth with integrated VCO covering 1.5MHz-3GHz in fundamental mode as well as a wide range of output dividers (and hence capable of generating any frequency we require).

jbqubit, 2016-11-08:
@jordens Please create a new label. Something like area:proto_round_2.

jordens, 2016-11-08:
@jbqubit You can use the milestone I had created a while ago.
Or create your own labels if you are certain it's well defined https://github.com/m-labs/sinara/labels

gkasprow, 2016-11-08:
We can still switch to a version with external PLL - I will update
schematics anyway.

gkasprow, 2016-11-08:
I even have HMC830 in my libraries so can do it quickly.

gkasprow, 2016-11-08:
The question is - how we create REF signal. It is single ended.
Will we use balun to convert LVPECL that comes from the muxes or just apply
external 100MHz from SMA or backplane directly?

gkasprow, 2016-11-08:
HMC7043 has only one clock input so it seems that place for HMC830 is on the clock mezzanine. I can place it on "demo" one

hartytp, 2016-11-08:
@gkasprow 

> HMC7043 has only one clock input so it seems that place for HMC830 is on the clock mezzanine. I can place it on "demo" one

or, put a mux (ADCLK925 or similar) in front of the HMC830 on the RTM. We already have muxes on the RTM for other purposes, so this won't add an extra BOM line.

> The question is - how we create REF signal. It is single ended.
> Will we use balun to convert LVPECL that comes from the muxes or just apply
> external 100MHz from SMA or backplane directly?

The input should come from a PECL mux/fanout-buffer, so use a balun (TCM2-43X+).

> We can still switch to a version with external PLL - I will update
> schematics anyway.

Does this mean no HMC7044 on the RTM?

hartytp, 2016-11-09:
@sbourdeauducq #40 

With the preface that I don't feel too strongly about this, since we can add whatever clock mezzanine we want later on:

> What do you mean by [it being nice to decouple the JESD204B functionality from the clock synthesis] and why?

There are two independent tasks we would like to perform:
1. Synthesising a low-noise, widely-tuneable DAC clock when neither a clock mezzanine or front-panel input are present
2. Generating the signals required to achieve deterministic latency on the JSED204B links.

The HMC704x is clearly a good IC for job 2. However, I'm not convinced that the HMC7044 is the ideal IC for job 1:
- the (guaranteed) tuning range of its VCOs is 2.4-3.2GHz, which is a little restrictive, even with the output dividers. For example, it can't produce 2GHz which, IMHO, is a pretty important frequency, since it's the highest DAC clock frequency that can be used with 2x interpolation. Going to 2.4GHz or 1.6GHz has trade-offs, which may or may not be a problem for a given application. Given the number of different ways this hardware will be used in different labs, it would be nice to be as flexible as possible about this in the long run.
- The HMC7044 is a bit of a behemoth, with a lot of functionality that we don't want (dual-loop PLL architecture, clock detection + automatic switching + hold-off, etc). That means more details to understand/potential for undocumented hardware issues/registers to program/etc.
- The phase noise for the VCO on the HMC830 is slightly better, which may allow more applications to use this instead of an ultra-low noise clock mezzanine.

So, the trade off is between getting to use a (potentiallY) slightly nicer IC to generate the DAC clock, or getting to reduce the number of BOM lines by 1.

sbourdeauducq, 2016-11-09:
OK, HMC7043+HMC830+ADCLK925 on the RTM is alright too if the performance is good.

hartytp, 2016-11-09:
FWIW, the HMC830 (or other IC in this family) is also likely to be a good contender for generating LOs on the clock/analog mezzanines. Thus, there is also a design/code reuse argument for choosing it over the HMC7044.

jordens, 2016-11-09:
To me it is important to get a "sufficient" clocking design working with the first prototype with a minimal amount of moving parts and maximal chances of being able to test everything. That's why I would put the HMC7044 eval board design (I looked at that loop filter in simulation: it's good) on there and not bother much with being unable to cover 2 GHz 2x interpolation. I would put the effort into making the clock mezzanine able to cover everything that's needed (including 2 GHz) with excellent quality.
For the second prototype if it really turns out that HMC830+HMC7043 on the RTM is the preferable option over the tested HMC7044 on the RTM and a HMC830 on a "simple clock mezzanine", I am fine with it.

hartytp, 2016-11-09:
@jordens While I take your point, I'm not convinced that the HMC830 is actually any more complex to set up/likely to go wrong. It also has a good reference design, and Greg already has a known-good package in his library. Moreover, as it's generally a simpler IC, there is less scope for getting things wrong. @gkasprow correct me if I'm wrong here, but the HMC830 is a pretty "classic" synth IC, which is widely used and understood, and hence we should expect to get it working on the first attempt

IMHO, if we think that an HMC830 is likely to be a better choice in the long run, then we should try to use it in the first prototype round -- that way, we're doing bug-fixes in the second round, rather than design changes...

hartytp, 2016-11-09:
Either way, someone (@jbqubit) needs to make an executive decision on this, so that we can finish the design review...

gkasprow, 2016-11-09:
OK, HMC7043+HMC830+ADCLK925 on the RTM is alright too if the performance is good.

ADCLK925 is not a mux.

That’s why I used SI53307

hartytp, 2016-11-09:
@gkasprow You're right, ADCLK925 isn't a mux -- I got the part numbers muddled in my head this morning when I wrote that. Sorry!

However, for the reasons outlined in #55, please don't use SI53307, but something like ADCLK948/HMC6832 instead.

gkasprow, 2016-11-22:
implemented HMC7043+HMC830+ADCLK925  as in in #55,

closed 2016-11-22 by gkasprow