sinara-hw/sinara #52: SMA alternatives for Sayma RTM

jbqubit, 2016-11-08:
SMA alternatives for Sayma RTM RF connectors was discussed in detail in #7. @dtcallcock is ordering SSMP for evaluation and consideration for next prototype round.

sbourdeauducq, 2016-11-11:
Smiths MDHC is expensive, they quote over $2k for a 8-way MDHC to SMA cable.

sbourdeauducq, 2016-11-11:
Smiths MDHC is expensive, $677 for 4-way right angle PCB mount connector, $823 for 8-way.

dtcallcock, 2016-11-15:
I got a Core GD mechanical sample kit to evaluate (see photos below).  This is the vertical mount, not the edge mount obviously.

[Kit Photo](https://cloud.githubusercontent.com/assets/20587606/20321563/20d28f2a-ab33-11e6-93e7-eaa48f703b96.jpg)
[Disassembled Connector](https://cloud.githubusercontent.com/assets/20587606/20321553/144cfae2-ab33-11e6-98f4-194390466d5a.jpg)

Whilst the connector is small it feels sturdy enough for the job.  I also like that the board connector and individual cables are field replaceable.  Let me know if you have any questions about it or if you want me to ship it to you (@jbqubit ?).

Given the reasonable price and improved usability and electrical performance over SMA I think they are worth using.  

There are two drawbacks I can see though.  One is that 4x4way won't fit on the card.  We will have to use 2x8way which implies double width mezzanines (2x10ch is also possible for bringing LOs etc. in).  The other disadvantage is that a common ground is enforced.  For example @hartytp wanted the option to capacitively couple the SMA shield to the mezzanine board ground.  However this could be done at the breakout panel if desired.

gkasprow, 2016-11-15:
It looks cool

Ask the manufacturer for detailed datasheet of 10 way RA connector – I want to make a PCB footprint

It’s better to design the mezzanine with 10 way connector since it affects also  the front panel design and gives option to use remaining 2 connectors as up/down conversion clock

dtcallcock, 2016-12-12:
Pingy McPingface @jbqubit.  

Let me know if you want to look at the sample otherwise I will return it this week.

jbqubit, 2016-12-12:
Oh, yes. Please send it my way. I'll take a look and will return to vendor.
Will send mailing address under separate cover. Thanks for the reminder.

-------
Joe Britton
Sensors and Electron Devices
Army Research Lab
2800 Powder Mill Rd
Adelphi, MD 20783
301-394-3130
joseph.w.britton5.civ@mail.mil

jbqubit, 2017-01-09:
I've had some additional correspondence with the sales rep for Carlisle. 

> Ryan Spencer
> Carlisle Interconnect Technologies
> Mobile: 360-607-7729

I asked about edge-mount version of their Core GD components. 
- SSMP MALE, 10 POSITION, PCB EDGE MOUNT 
- CORE HC CABLE ASSEMBLY 10 POSITION

His response was: 

>Regarding pricing, I don't have a fixed number to report to you, but I think you 
>can use something around $60 - $75 price per mated channel would be a rough estimate.

jbqubit, 2017-02-21:
Alternatives are established. No further discussion is needed at present. Closing.

closed 2017-02-21 by jbqubit

gkasprow, 2017-09-19:
Yet another [RF connectors](https://cdn.harwin.com/Harwin_Coax_Product_Announcement_Aug_17.pdf?utm_source=Hi-Rel+Newsletter+Markets+-+EU&utm_campaign=668c2423f5-High_Rel_Newsletter_EU_Sept_2017&utm_medium=email&utm_term=0_457246cfd8-668c2423f5-104411045) to consider in future designs