sinara-hw/sinara #53: Sayma RTM: RFB reference clock connections

hartytp, 2016-11-09:
@gkasprow uRFB_CLK_IN & uRFB_REF_IN will both be PECL square-waves, and so can be fed into the clock mux without a LTC6957. For uRFB_REF_IN, please use a balun (TCM2-43X+ or similar) for single-ended to differential conversion before the mux.

Bearing in mind that uRFB_CLK_IN & uRFB_REF_IN will contain harmonics in the GHz regime, do you think that selecting between them using DNF capacitors can be done with an acceptably small impedance discontinuity? (I assume the answer is "yes", but wanted to check...)

gkasprow, 2016-11-23:
@hartytp 
In such cases I place 2 caps with shared pad so no additional impedance discontinuity is added.
SMA ground needs to be connected to GND by capacitor or only to to the balun?
In case we connect it to the balun only, the ESD can kill the LTC input, so it's better to connect balun common to GND and SMA ground to GND via capacitor.

hartytp, 2016-11-23:
@gkasprow 

> In such cases I place 2 caps with shared pad so no additional impedance discontinuity is added.

Sounds good.

>In case we connect it to the balun only, the ESD can kill the LTC input, so it's better to connect balun common to GND and SMA ground to GND via capacitor.

Agreed:
- connect the balun common to circuit ground
- connect SMA ground to circuit ground through a small capacitor. Would you be happy with 10pF (16k @ 1MHz, so fine for killing ground loops) for this?
- AC couple balun inputs and outputs with 100nF

gkasprow, 2016-11-23:
@hartytp
The balun will be placed very close to the LTC6957.
Do you prefer termination at the input or at the output of the balun?

hartytp, 2016-11-23:
@gkasprow  no strong preference, so long as the termination is after the AC coupling capacitor.

gkasprow, 2016-11-23:
ok

closed 2016-11-23 by gkasprow