sinara-hw/sinara #55: Sayma RTM: 100MHz clock mux

hartytp, 2016-11-09:
@gkasprow Please consider replacing the Si53307 used to select between the RFB and EXT 100MHz clocks (sheet 6) with an ADCLK925 or similar, because:
- AFAICT, the phase noise is quite a bit better on the ADCLK925 (comparing Si53307 data sheet figure 13 with ADCLK925 figure 8)
- the ADCLK925 propagation delay is significantly lower. Low propagation delay usually corresponds to better propagation delay stability... (The ADCLK925 specifies the propagation delay temp co, but the Si53307 doesn't seem to).
- The ADCLK925 goes up to higher frequencies than the Si53307, which will allow us to use the same IC for DAC clock fan out. Thus, we can remove a line from the BOM.
- If we are keen to keep the BOM as small as possible, we could use an ADCLK948 or similar instead, since we already have one elsewhere on the RTM.

gkasprow, 2016-11-09:
@hartytp 
ADCLK925 is fanout buffer, not mux.
ADCLK948 is cheaper than ADCLK925 so we can use it in this case

hartytp, 2016-11-09:
@gkasprow Yes, you're right! Sorry, misremembered my IC part numbers...

gkasprow, 2016-11-22:
Si53307  replaced by ADCLK948

closed 2016-11-22 by gkasprow