sinara-hw/sinara #56: Sayma RTM: clock diagnostic outputs

hartytp, 2016-11-09:
@gkasprow If there is sufficient space on the RTM, please consider adding RF connectors (SMA/MMCX/whatever) on some of the unused outputs of clock fanout buffers, inc HMC704x (1 pair of connectors per fanout-buffer). These can be DNF by default, and would be useful for characterising/debugging the clock distribution network.

Don't worry if space is too tight though...

sbourdeauducq, 2016-11-09:
I suppose those would not be on the front panel, right?

hartytp, 2016-11-09:
Definitely not!

gkasprow, 2016-11-09:
What abut UFL connectors?
I often mount them as high frequency testpoints
And they take very little space

hartytp, 2016-11-09:
@gkasprow happy to use whatever connectors you prefer for this.

gkasprow, 2016-11-09:
Add it as an issue with example signals specified.

sbourdeauducq, 2016-11-09:
@gkasprow this is already an issue and @hartytp proposed "some of the unused outputs of clock fanout buffers, inc HMC704x".

sbourdeauducq, 2016-11-11:
There are no unused outputs right now. But we can disconnect RTM_MASTER_AUX_CLK_* (aka FPGA_REF_CLK_*, see #63) which is the output that goes to the Artix-7 FPGA and probably is not useful, and put a pair of UFL connectors instead.

closed 2016-11-22 by gkasprow