sinara-hw/sinara #58: electrical compatibility of FMC transceiver pins and regular FPGA IOs

sbourdeauducq, 2016-11-10:
The FMC transceiver pins (FMC_DP...) are connected directly to Ultrascale IOs. Aren't those signals supposed to be CML? http://www.xilinx.com/support/documentation/user_guides/ug571-ultrascale-selectio.pdf does not say that regular IOs support CML. Isn't some sort of converter required?

jordens, 2016-11-10:
I think this is fine. Since those are not GT*, you could only hope to either use them for your own (slower) serial link non-CML or as regular IO. Connecting them directly to support some custom usage is a better choice than level translators or leaving them unconnected IMHO.

dhslichter, 2016-11-10:
We are however breaking the FMC standard if we populate the transceiver pins in the FMC connector with non-transceiver lines.  Not sure the degree to which we care about this, since we will not likely be inserting any 3rd party FMC boards into this slot?

gkasprow, 2016-11-10:
It’s still better than keeping tchem unconnected. One can run i.e. gigabit Ethernet over IOSERDES.

WE don’t have any free GTX. Moreover, I don’t know any useful for our application FMC board that uses only one MGT line.

I can also add option to route 8 MGT lines to GTX ports in favor of RTM connection as it is done in AFCK.

sbourdeauducq, 2016-11-11:
With level translators, and if a GT clock is provided, then the regular IOs can be used like GTs at low speeds. I don't have any particular FMC card that we could use that way though.

sbourdeauducq, 2016-11-11:
Translators we could use are SN65LVDS100 and SN65CML100. I don't have a strong opinion about this, as this is only for supporting a speculative FMC that would make use of a low-speed transceiver link.

sbourdeauducq, 2016-11-11:
The same situation is present on FAT_PIPE2 and half of FAT_PIPE1, but this is again unimportant.

jordens, 2016-11-11:
@sbourdeauducq yes. as i said, it would be custom anyway. i would not put level translators there.
AFAICT at least CML driving LVDS did work fine in previous FPGAs at usual speeds: https://www.xilinx.com/support/documentation/application_notes/xapp756.pdf

jbqubit, 2016-11-11:
> I can also add option to route 8 MGT lines to GTX
> ports in favor of RTM connection as it is done in AFCK.

Seems like a reasonable compromise.

-------
Joe Britton
Sensors and Electron Devices
Army Research Lab
2800 Powder Mill Rd
Adelphi, MD 20783
301-394-3130
joseph.w.britton5.civ@mail.mil

gkasprow, 2016-11-22:
This would impact signal integrity and BERR - I would need additional pair of layers to do such routing. 
So I leave connected to IOSERDES

closed 2016-11-22 by gkasprow