sinara-hw/sinara #62: RTM UART link needs to be bidirectional

sbourdeauducq, 2016-11-11:
As far as I can tell, there is only one pair USR_UART_N/P between the AMC and RTM. We need one pair for TX and one pair for RX - or go single-ended if there aren't enough pins.

sbourdeauducq, 2016-11-11:
This UART link will operate at a low speed (115kbps) so we can have a slow slew rate to reduce EMI/SI problems with single-ended signaling.

sbourdeauducq, 2016-11-11:
RTM_FPGA_GTP_CLK can be removed as we can use the CDR + the Si5324 on the RTM side, this makes space for another UART pair.

gkasprow, 2016-11-12:
This pair is routed as diff signal but can be used as 2 single ended signals for slow UART.
We can also use it as high speed LVDS

sbourdeauducq, 2016-11-12:
I am fine with that solution.

jbqubit, 2016-11-18:
Can this be closed?

-------
Joe Britton
Sensors and Electron Devices
Army Research Lab
2800 Powder Mill Rd
Adelphi, MD 20783
301-394-3130
joseph.w.britton5.civ@mail.mil

sbourdeauducq, 2016-11-18:
I'd rather keep all those open until @gkasprow posts a new version of the schematics that includes the fix.

gkasprow, 2016-11-22:
fixed

closed 2016-11-22 by gkasprow