sinara-hw/sinara #64: clock length matching

jordens, 2016-11-11:
When doing the layout, make sure that each SYSREF is matched to **its** CLK (a.k.a. DEVICECLK) to better than 100 ps.

gkasprow, 2016-11-11:
What other lines should be matched?
DAC - mezzanine connector between channels
ADC - mezzanine connector between channels
JESD204B not necessary

jordens, 2016-11-11:
* differential wire length matching (positive to negative): required
* I/Q wire length matching (the pair of channels to/from each mezzanine): yes, if possible
* matching any two mezzanines: nice to have
* ADC-to-DAC length matching: not necessary
* JESD204B links: only differentially length matched

sbourdeauducq, 2016-11-12:
Matching between the RTM_FPGA_GTP_Rx* and between the RTM_FPGA_GTP_Tx* is also desirable, but not strictly necessary (we can have gateware to compensate, or use only one line to begin with).

jbqubit, 2016-11-29:
@gkasprow  who on your team is responsible for Sayma layout? I'd like to assign them this and other layout related items.

gkasprow, 2016-11-29:
It's me and Michal. Samer will do all SI verification

>
>

gkasprow, 2017-02-17:
Yes.

closed 2017-02-17 by gkasprow