sinara-hw/sinara #67: AMC FPGA <-> RTM FPGA clocking

gkasprow, 2016-11-11:
@sbourdeauducq 
How to organize clocking for interconnectivity between AMC FPGA (Kintex) and RTM FPGA (Artix)?
At the moment the clock for Artix GTP is connected to Kintex IO.
It is not optimal. Artix can recover the clock from data link, but what about Kintex IOSERDES? It needs quite well defined clock to data alignment. Maybe it makes sense to connect Kintex IOSERDES Global CLock line to the SI5324 on the RTM side since it already generates clock for GTP on Artix side.

sbourdeauducq, 2016-11-12:
> It needs quite well defined clock to data alignment.

It doesn't, we'll use a phase detector and adjust IDELAY automatically like in HDMI/DVI receivers. All it needs for this is a well defined clock frequency, which we will get by having the RTM FPGA use the recovered clock for transmission after a trip through the Si5324.

gkasprow, 2016-11-12:
I mean IOSERDES does not have CDR so it is sensitive to the jitter. So what option would you like:
- Kintex IOSERDES clocked by RTM Si5324 which also clocks Artix GTP
- Artix GTP clocked by Kintex IO which also clocks IOSERDES in KIntex
- both options selected by capacitor placement

sbourdeauducq, 2016-11-12:
The speed is relatively low (~1Gbps), I don't think there will be significant high-frequency jitter to cause any trouble. Low-frequency (<kHz, maybe more) jitter/skew can be compensated for by using the phase detector as mentioned. What is the GTX CDR loop bandwidth anyway?

sbourdeauducq, 2016-11-12:
To clarify what I am proposing: both the TX and RX IOSERDES are clocked by the same clock. The RTM transceiver recovers it with its CDR, and uses it to clock its transmission, after cleaning it up with the Si5324. The AMC RX IOSERDES uses a phase detector to automatically adjust to the clock skew. So it's all a synchronous system in the end, with some self-tuning (CDR and phase detector).

gkasprow, 2016-11-12:
They are clocked by the same clock i both cases. The question is if the clock is generated on AMC or on RTM side

sbourdeauducq, 2016-11-12:
You can leave the clock pair if you want, but it is not necessary.

sbourdeauducq, 2016-11-12:
The clock is generated on AMC and transmitted on the same pair as the data.

gkasprow, 2016-11-12:
OK, but the RTM GTP needs some clock as well. I assume it comes from SI chip hooked to the Artix.

sbourdeauducq, 2016-11-12:
> it comes from SI chip hooked to the Artix

Correct.

gkasprow, 2016-11-12:
So in this way we can free a pair of signals on the RTM connector, unless you need a clock for RTM FPGA driven by AMC FPGA chip or AMC FPGA driven by RTM Si5324 chip or Artix FPGA.

sbourdeauducq, 2016-11-12:
Yes. That's what I was proposing here: https://github.com/m-labs/sinara/issues/62#issuecomment-259893569

closed 2016-11-22 by gkasprow