sinara-hw/sinara #68: SYNCOUT{1,2}1_{N,P} not needed

jordens, 2016-11-11:
We don't need to run SYNCOUT1 from the two DACs to the AMC FPGA since we don't expect to unbundle the JESD link to the each DAC.
SYNCOUT{1,2}1_{N,P} are not needed on the AMC-RTM connector. Leave them unconnected or terminated (whatever the datasheet recommends).
This frees up four pins on that connector. I would use two of them to implement #66 instead.

jordens, 2016-11-11:
On the RTM also `GTP{8,9,10,11,12,13,14,15}RX_{N,P)` are unused. I am fine with having AMC GTP_RX connected there. But we should note in the schematics that we don't strictly need them and could use those 16 pins for something else if need be.

gkasprow, 2016-11-12:
The idea was to enable users define their own RTM modules so I don’t want to limit the GTX receive mode.

Maybe someday we will need 16xSFP RTM card or high performance ADC input?

OK, I will mark them accordingly.

gkasprow, 2016-11-22:
done

closed 2016-11-22 by gkasprow