sinara-hw/sinara #71: add an XO on Sayma RTM

sbourdeauducq, 2016-11-12:
On MGTREFCLK0_216 for RTM<->AMC transceiver link debugging. 50MHz.

gkasprow, 2016-11-22:
@sbourdeauducq 
what about using pair of ufl connectors to deliver 100MHz clock from one of ADCLK muxes?
I don't have any 50MHz differential oscillators.
Since it is for debugging only, you can install ufl to ufl jumpers and deliver 100MHz to any ufl you want
So I added pair of ufl for both AMC and RTM FPGA GTP clock inputs

sbourdeauducq, 2016-11-23:
A quick Digikey search turned up a number of XOs with LVDS output. I suggest putting a footprint for one of those and keeping it a placeholder (DNP) for now.

gkasprow, 2016-11-23:
I mean that I don't have in my libraries:)
I will create one.

>
>

sbourdeauducq, 2016-11-23:
OK thanks.

closed 2016-11-23 by gkasprow