sinara-hw/sinara #72: RTM<->AMC transceiver link electrical compatibility

sbourdeauducq, 2016-11-12:
Does the Artix-7 CML play nice with the Ultrascale LVDS, or do we need level converters?

gkasprow, 2016-11-12:
No level translation is needed in AC-coupled mode
http://www.ti.com/lit/an/scaa059c/scaa059c.pdf

sbourdeauducq, 2016-11-12:
OK, what translator chips are you going to use?
Those chips should be on the AMC side, as they are slower (<=2Gbps) than the Artix transceivers but faster than the KU IOSERDES. This way we don't introduce any additional speed limitation, in case we want to plug the RTM to some other card in the future.

gkasprow, 2016-11-12:
Are you sure that translators are needed? Do you want to DC-couple them?

sbourdeauducq, 2016-11-12:
Ah sorry, I misread and thought they were needed. AC coupling is fine.

closed 2016-11-12 by sbourdeauducq