sinara-hw/sinara #74: Metlino RTM

gkasprow, 2016-11-12:
Do we consider future use of Metlino RTM?
One  option is to use existing  utca-rtm-8-sfp-2 design that has 8 SFP+ cages on it.
But in our application for the moment we need to have only the same reference clock as RTM modules have.
The main issue with RTM connector on Metlino is related to MCH tongues - Metlino sits in slots 3 and 4 while RTM connector can be placed on base board that sits in slot 1. So one would need small adapter that holds the RTM connector and board to board connector that passes signals from Metlino slot 3 to slot 1 where RTM is.

The uRFB has 2  diff clock lines per AMC-RTM module and 3 diff lines from eRTM15 to MCH-RTM. One of these lines could be used for Metlino clock reference distribution.
Any idea what else we would like to have on Metlino RTM? It would be an overkill to have such board only to pass a few signals.

jordens, 2016-11-13:
Thanks for bringing this up, Greg. I can only think of FMCs (but probably not enough signals on the connector) and SFP cages.

gkasprow, 2016-11-23:
Can we close this issue?

jordens, 2016-11-23:
@gkasprow just ping whoever needs/should sign off on this:
@jmizrahi @jbqubit ping.

jmizrahi, 2016-11-23:
@gkasprow What is the conclusion here?

gkasprow, 2016-11-23:
The solution I proposed enable 3 possible options: 
1. Get rid of NAT MCH in favor of dedicated Tongue 1 board with simplest possible management or without it, RTM connector and additional SFPs and VHDCI on the Metlino front panel. 
2. Keep NAT MCH and use simple FMC to RTM adapter.
3. install FMC-SFP board directly for stand-alone operation.
In both cases RTM will be used for SFPs, uRFB clock distribution to Metlino and whatever you can imagine.
The conclusion:
We want to build simple FMC-RTM adapter and test it with existing RTM-SFP board (http://www.ohwr.org/projects/utca-rtm-8-sfp/wiki)
Later on we will modify it and add uRFB connectors.

jbqubit, 2016-12-02:
This decision can be made in the future. All these options are forseen.

jbqubit, 2016-12-05:
I've assigned this Milestone 0.2 (prototype 2). Reopened since it remains to make a decision about which path to implement.