sinara-hw/sinara #75: Sayma RTM: HMC830 connections and recovered clock

hartytp, 2016-11-15:
@gkasprow Consensus on connections for the RTM recovered clock + HMC830 connections (this is mainly a summary of issue #66):

- No new AMC <-> RTM connections are required for the for the recovered clock
- Disconnect RTM SI5234's SI_CLK_OUT2 output from the RTM FPGA
- Route RTM SI5234 CLK_OUT2 to a clock mux, whose other input is the backplane/front-panel 100MHz reference (on the current version of the schematic, this corresponds to the ADCLK948 output that routes to OSCIN on the HMC7044)
- The output from this mux goes through a balun (TCM2-43X+) for differential to single-ended conversion and then to the HMC830 reference input.
- The HMC830's output goes through an ADCLK948 or similar clock mux, whose other input is the clock mezzanine output. The output of this mux goes into the HMC7043 input.
- Other HMC830 connections should be as in the reference design. We can pick exact values for loop filter passives later on.

hartytp, 2016-11-16:
With these connections, the hardware can support the following clock configurations (not that I'm suggesting we have to support them all in software!):

**REF_CLK** (100MHz in most cases):
1. **Backplane** -- uses either single-ended or differential ("clock") lines on RF backplane. Selection between these options is by component population choice.
2. **Front-panel** -- source is external oscillator. Requires coax between SMAs on the RTM front-panel and pcb
3. **Internal XO** -- source is RTM SI5324 + crystal, used for debugging when no CDR clock is available
4. **CDR** -- recovered from gigabit serial links, supplied by RTM SI5324.

**DAC_CLK**:
1. **External** -- DAC_CLK supplied from an external oscillator via the front-panel "clock" SMA. Requires a simple clock mezzanine and coax between the front-panel "clock" SMA and mezzanine MMCX. Not compatible with REF_CLK option (2).
2. **Internal** -- DAC_CLK generated from REF_CLK by HMC830 on RTM. Compatible with all sources of REF_CLK.
2. **Mezzanine** -- DAC_CLK generated from REF_CLK by PLL on the clock mezzanine. Compatible with REF_CLK options (1) & (2) (NB SI5324 phase noise is too bad to provide a reference for a low-noise mezzanine anyway).

**Analogue mezzanine REF/LO**:
1. **REF_CLK** -- REF_CLK routed to the analogue mezzanines via SMPs. Used, for example, when the mezzanines have PLLs to generate their LO. Compatible with REF_CLK options (1) & (2), and all DAC_CLK options.
2. **Clock mezzanine/LO** -- LO/RF output from clock mezzanine routed to the analogue mezzanines via SMPs. This is used, for example, when the clock mezzanine supplies a high-quality LO. The source of this LO can either be an external oscillator, routed from the front-panel "clock" SMA to MMCX on the mezzanine by coax, or a PLL on the clock mezzanine. NB the maximum frequency for this signal is limited by the clock muxes used in the distribution network (4.8GHz for the ADCLK948)
3. **Analogue mezzanine** -- LO is supplied by an external oscillator via an analogue mezzanine front-panel SMA. Used, for example, when the mezzanine has IQ mixers, so 2 DAC channels give 1 microwave output, leaving spare SMAs.
4. **Mixer-mode** -- no LO required on the analogue mezzanine, since the DAC mixes its RF with DAC_CLK. Should allow operation to 3.5GHz

I think that should give enough options for everyone to be able to find a configuration that works for them!

NB Names are just suggestions/place-holders and can be changed later...

hartytp, 2016-11-18:
Suggested passive, 2nd order loop-filter for HMC830, optimised to minimise noise when driven by an ultra-low noise 100MHz reference.

[image]

Noise model, when using a high-quality 100MHz reference oscillator:

[image]

Noise model, when using the CDR 100MHz reference (same loop filter):
[image]

Comments:
- The thick blue shaded line shows the noise floor for an idealised "brick-wall" loop-filter. The red dashed curve shows the full noise model, including the transfer function of our proposed loop-filter. 
- All noise contributions have been normalised to the 2.4GHz DAC clock frequency
- ADCLK948 appears twice, because it is used both for REF_CLK (100MHz) distribution and for DAC_CLK distribution. In the 100MHz case, the broad-band noise is multiplied by the PLL.
- With a high-quality external reference, the HMC830 noise floor is below that of the AD9154 for almost all frequencies. This suggests that a clock mezzanine/external 2.4GHz oscillator will not generally be needed to generate the DAC clock (although they could still be useful, for example, for providing a LO)
- When using the CDR reference, the noise floor is limited by the SI5324. This leads to a ~20dB degradation of the DAC noise floor close to the 250kHz loop bandwidth. However, over much of the frequency spectrum, the DAC_CLK with CDR ref is within 10dB of the DAC noise floor, suggesting that CDR may be sufficient in many cases.
- The noise floor for both external and CDR references could be improved by switching to an active 3rd-order loop filter. We'll post a schematic + noise model for this next week.
- "chip" refers to the HMC830 PLL

Credit: @weidaZhang did all the work

gkasprow, 2016-11-23:
I've updated the schematics.

gkasprow, 2016-11-29:
I added some additional capacitor footprints to enable greater component value flexibility. Capacitors are available in E12 series, some values are also available in E24.

hartytp, 2016-11-29:
@gkasprow Looks good.

closed 2016-11-29 by hartytp

jordens, 2016-11-29:
@hartytp those are gigantic images you posted above (200 and 120 Megapixels respectively). Could you resize them? My machine starts thrashing every time I am hitting this page.

hartytp, 2016-11-29:
@jordens sorry, that is a bit sloppy. I'll fix that when I have time...

hartytp, 2016-11-29:
For completeness, here is the noise model for our HMC440 clock mezzanine PLL:

[image]

Comments:

- HMC440 is the lowest-noise PLL IC available AFAICT, particularly for flicker noise.
- PLL noise floor is expected to be well below the DAC noise floor at all frequencies. Thus, this PLL is really overkill for generating the DAC clock (the HMC830 should be fine for this).
- One use case for such a good PLL is generating a LO for upconversion. In this case, the DAC will run at ~300MHz output & LO will be at ~3GHz. For equal phase noise contributions from the DAC and LO, the LO phase noise should thus be ~20dB lower than the DAC clock noise.
- Below a few hundred Hz, phase noise is limited by the reference, even for a state of the art 100MHz source
- Above ~250kHz, noise is limited by the ADCLK948 mux (however, the noise should still be comparable to a top of the range Keysight synth)
- In between, it's limited by the PLL/VCO

Relevant to #43