sinara-hw/sinara #76: Metlino access to MTCA MLVDS

gkasprow, 2016-11-23:
Do we have any plans to use MLVDS bus?
In MTCA.4 standard it connects all AMC boards with 8 bidir MLVDS buses.
New Schroff backplanes will have  MLVDS bus routed to MCH, Tongue 2. There are unused pins.
I can just place ordinary goldpin connector on Metlino in such way that in future we can make Tongue2 board that supports such functionality.
Would such bus be useful for ARTIQ? Distribution of some triggers, resets, interlocks, time stamps...
Some ideas?

jordens, 2016-11-23:
I can't think of anything anything that would be generically useful without leading to a complexity nightmare.

I can see it being useful for bigger control loops (PIDs split between Sayma cards) without Metlino involvement but the point-to-point links on port 12-13 seem better suited for that.

gkasprow, 2016-11-24:
OK, so I will use these FPGA IOs for Fabric E routing

closed 2016-11-28 by gkasprow