sinara-hw/sinara #77: Metlino clocking

gkasprow, 2016-11-24:
On the front panel I installed 2 SMA connectors.
One would be optional 100MHz clock input.
What could be function of second SMA? In the email https://ssl.serverraum.org/lists-archive/artiq/2016-September/001007.html there was an option:

jordens, 2016-11-24:
I would DC couple it and put a big fan out after that SMA and feed the Metlino FPGA (clock capable) and -- if possible -- one of the telco clocks (DC coupled again) of the AMC backplane. On the root Metlino we could e.g. use it for PPS to align the RTIO clock to GPS clock.

The clocking scheme sounds good to me.

gkasprow, 2016-11-25:
The mux has 8 outputs so we can connect to:
- GTX clock sources - is one quad enough or it's better to route it to all quad clocks?
- Tongue 2 clock distribution connector
- RTM
- FPGA Global clock
- SMA output?

Or it is better to distribute the SI output to RTM, SMA and Tongue 2?

jordens, 2016-11-25:
I would use the second SMA input as described above (DC coupled, fan out to Metlino FPGA global clock and tounge 2 clock distribution).

The output from the SI5324 should go through a fan out and into as many GT quads as needed to drive everything. AFAICT we can clock quad N from the GTREFCLKs of the five quads N-2 to N+2. To use every transciever link for DRTIO there needs to be a driven clock among the four adjacent quads everywhere. @sbourdeauducq did i read that right?
Being able to drive a telecom clock (through tounge 2 I presume), and a global clock capable FPGA input as well are needed. RTM and a SMA output are much less important IMHO.

sbourdeauducq, 2016-11-26:
Yes, quoting the transceiver user guide:
"The maximum number of transceivers that can be sourced by a single clock pin pair is 20.
Designs with more than 20 transceivers require the use of multiple external clock pins to
ensure that the rules for controlling jitter are followed. When multiple clock pins are used,
an external buffer can be used to drive them from the same oscillator."

There needs to be a driven clock in the middle of blocks of up to five adjacent quads.

gkasprow, 2016-12-02:
the CDR clock is connected to CLK0 of 226 transceiver

closed 2016-12-02 by gkasprow